perfanalytics/pose2sim #2: About the name and order of 2d json files

ykk648, 2022-06-10:
as for dont know the names of 2d/2d-tracked json files, I used my own name format `N.json`, but the codes in `triangulate_3d.py` may not sort the json files right  , causing disorder 3d joints shown in OpenSim before I modified these codes.

https://github.com/perfanalytics/pose2sim/blob/2113aa5f4aad097cb40465ee79eaa8d48abe2955/Pose2Sim/triangulate_3d.py#L389

davidpagnon, 2022-06-10:
I'm not sure I understand your problem, can you be a little more specific?
- What are your file names like?
- What did you modify in the code?

These Try/Except lines try to triangulate from the pose2D-tracked folder, but if the tracking step has not been completed it reverts to triangulating from the pose2d folder.
Normally it should take your json files in order, regardless of their names.

Example: if you have weird json file names and two cameras like:
> pose2d

>>cam1_json
>>>1.json
>>>45.json
>>>blabla.json

>>cam2_json
>>>a.json
>>>8.json
>>>0005.json

It will triangulate 1 and a, 45 and 8, blabla and 0005.

The only thing is that your json folders must end with the string 'json' (but you can change this string in `Config.toml`).

davidpagnon, 2022-06-10:
An other possible cause of malfunction is if you just didn't use the right OpenPose model.
Did you use body_25 or body_25b ? You must specify it in the `Config.toml` file.

carlosedubarreto, 2022-06-10:
I had the same problem.
The body_25B skeleton is not the same as body_25.
And we can easily find just the body_25 skeleton description.

@ykk648 take a look at the file skeleton.py, you will find your answer there 😃

ykk648, 2022-06-13:
thanks for your reply, it may caused by different json name format.
here's codes I modified, I specifed the sort way for json files, if not, the json file sequence will be like `1.json`, 	`3.json`	,`2.json`, now it's `1.json` `2.json` `3.json`	

`   
json_file_sorted_probe = lambda x: int(x.split('.')[0])
    try:
        json_files_names = []
        for js_dir in json_dirs_names:
            before_sort = fnmatch.filter(os.listdir(os.path.join(poseTracked_dir, js_dir)), '*.json')
            before_sort.sort(key=json_file_sorted_probe)
            json_files_names.append(before_sort)
        # print(json_files_names)
        json_tracked_files = [[os.path.join(poseTracked_dir, j_dir, j_file) for j_file in json_files_names[j]] for j, j_dir in enumerate(json_dirs_names)]
    except:
        json_files_names = []
        for js_dir in json_dirs_names:
            before_sort = fnmatch.filter(os.listdir(os.path.join(pose_dir, js_dir)), '*.json')
            before_sort.sort(key=json_file_sorted_probe)
            json_files_names.append(before_sort)
        json_tracked_files = [[os.path.join(pose_dir, j_dir, j_file) for j_file in json_files_names[j]] for j, j_dir in enumerate(json_dirs_names)]
`

It may caused by rename of OpenPose results, I would like close this issue , thank you again.

closed 2022-06-13 by ykk648