perfanalytics/pose2sim #4: How to use two Cameras to calibration checkerboard #1

davidpagnon, 2022-06-10:
Copied from old directory:
https://github.com/davidpagnon/pose2sim/issues/1

carlosedubarreto, 2022-06-10:
I suffered also with calibration, so I did one tool in blender using easymocap.
The calibration part I tried to make as easy as possible. And I explain in one video.

Here are the intrinsic, extrinsic and manually fixing calibration using labelme

Instrinsic:
https://youtu.be/S8nSDZjcaSA?t=949

Extrinsic:
https://youtu.be/S8nSDZjcaSA?t=1243

Fixing extrinsic calibration, if needed:
https://youtu.be/S8nSDZjcaSA?t=1404

After that you can follow what I'm explaining here
https://youtu.be/Al-dj6uPP70?t=356

If when converting throws an error, possibly you will have to change the intri.yml and extri.yml.

their probably will be like this
[image]

and they should be like this, for the conversion to work:
[image]

in resume, remove the leading zero.

And that is it, it works great for me , and I found it less confusing.

I plan to make a better version, but that is what I have for now 😊 
Hope it helps.

davidpagnon, 2022-06-10:
Very nice full workflow with EasyMocap in Blender!
If it is of anyone else's interest, you can find the script you used for converting these intrinsic and extrinsic .yml files (as required in EasyMocap) to the .toml file (used by Anipose and Pose2Sim) here: 
`Utilities\calib_yml_to_toml.py`

Thank you for pointing out the issue with leading zeros, I need to fix it.

Also, I have a tool for displaying cameras with .toml calibration files. It is done in Maya, though. https://github.com/davidpagnon/Maya-Mocap
[image]

carlosedubarreto, 2022-06-10:
wow, this maya tool is very interesting...
And thanks for taking a look on the leading zeros. I was going to take a look on that, I'm just waiting for some decent free time LOL... And that never comes 😅

davidpagnon, 2022-06-10:
This is now fixed 😃

carlosedubarreto, 2022-06-10:
Thanks a lot for the fix, I'm downloading it to test 😃

carlosedubarreto, 2022-06-12:
Thanks a lot for your help @davidpagnon 
After a few days of coding I was able to release a version of an addon to do a full markerless mocap using your code a many other solutions.

Here is a link detailing it, with a tutorial showing and explaining the full workflow, and link to download.
https://www.patreon.com/posts/67676360

ykk648, 2022-07-12:
I'm using [multical](https://github.com/oliver-batchelor/multical)

davidpagnon, 2022-07-12:
I didn't know about this tool, thanks!

To sum it up, here are a few ways to calibrate:
- Converting a calibration file from Qualisys with Pose2Sim.calibrateCams(). The easiest way, providing that you have Qualisys cameras and that their calibration procedure has worked (harder in broad daylight and if cameras are far away) 
- Use a checkerboard with Pose2Sim.calibrateCams(). Easy, and you can have a visual feedback to check whether the pattern is correctly detected of not
- Use [Anipose](https://anipose.readthedocs.io/en/latest/tutorial.html#calibrating-the-cameras) calibration. I use the exact same folder structure and the same kind of output calibration file, so this should be seamless. This is probably much more robust than other solutions, so if accuracy is very important make sure to try it out
- Use Carlos' MPP2SOS calibration procedure, which is directly integrated in Blender. Helps you fix calibration if the checkerboard is not correctly detected
- Try [multical](https://github.com/oliver-batchelor/multical). I haven't tried it yet. There are probably lots of other tools over the internet, so if you find something that suits you, tell me about it!

Rather than making my own tool perfect, I'd rather embed the existing ones in Pose2Sim. It's on my to-do list -- like way too many other things :)

One further warning: whatever you do, make sure your checkerboard takes more than one 5th of the image (at the very least). Otherwise the checkerboard won't be correctly detected (and if it is, calibration will not be very accurate anyway).

carlosedubarreto, 2022-07-12:
Thanks for pointing my addon David.
Unfortunetely I had to make Mocap MPP2SOS paid to at least justify my dozens of coding time devoted to it. (I plan to make huge improvements.

But I dont like to put a barrier on stuff I create, so there is also another alternative to use the calibration workflow I've added to MP2SOS.

In case you are interested, there is another blender addon called CEB Easymocap (Its a interface in blender for the Easymocap tool)

It´s free and has most of the calibration process that MPP2SOS has.
https://carlosedubarreto.gumroad.com/l/ceb_easymocap_blender

On the site it says it costs $2, but there is a coupon on the description. I had to put it as paid because the files were too big and gumroad didnt allow the free tier.

Ming-Start, 2023-04-23:
Hello , I am currently using Pose2Sim tools but i have some problems that Cameras calibration uses to checkerboard. I have created two folders, and each folders have 230 pictures with checkboard. But when I run the code from the readme, it always show s this error:
Project directory: D:\Code\Human\pose2sim\Pose2Sim\Empty_project

Camera Camera01:
---------------------------------------------------------------------------
error                                     Traceback (most recent call last)
<ipython-input-4-4a5285842b1d> in <module>
----> 1 Pose2Sim.calibrateCams()

~\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py in calibrateCams(config)
    100     start = time.time()
    101
--> 102     calibrate_cams_all(config_dict)
    103
    104     end = time.time()

~\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\calibrate_cams.py in calibrate_cams_all(config)
    429
    430     # Calibrate
--> 431     ret, C, S, D, K, R, T = calibrate(config)
    432
    433     # Write calibration file

~\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\calibrate_cams.py in calibrate(config)
    372
    373     # Calibrate
--> 374     ret, C, S, D, K, R, T = calib_fun(config)
    375     return ret, C, S, D, K, R, T
    376

~\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\calibrate_cams.py in calib_checkerboard_fun(config)
    290         # Calibration
    291         r, mtx, dist, rvecs, tvecs = cv2.calibrateCamera(objpoints, imgpoints, img.shape[1::-1],
--> 292                                         None, None, flags=(cv2.CALIB_FIX_K3 + cv2.CALIB_FIX_PRINCIPAL_POINT))
    293         h, w = [np.float(i) for i in img.shape[:-1]]
    294         print(r, repr(mtx), repr(dist))

error: OpenCV(4.7.0) D:\a\opencv-python\opencv-python\opencv\modules\calib3d\src\calibration.cpp:3749: error: (-215:Assertion failed) nimages > 0 in function 'cv::calibrateCameraRO'
How can I solve this error ?

davidpagnon, 2023-04-27:
Hello, I suspect that your corners were not detected. Can you set the option "show_corner_detection" to true in Config.toml, and tell me it you see corners on your images?
I not, it may mean that the corners are not visible enough (with that many images, I doubt none worked), or that you have not specified the right amount of corners. Beware that the corners on the outside should not be counted.

piolho-eefe, 2023-07-17:
Hi... I am having issues with the checkerboard calibration. I have 21 images in each folder for each camera (three) inside calib2d and I am having the following issue:
[image]
If I take out the images from the folders, I have the following issue
[image]
What should I do?

davidpagnon, 2023-07-17:
Hi, I suspect that your checkerboards are not correctly detected, can you set `show_corner_detection` to true in Config.toml, and tell me what happens?
Also, if you can wait about a week or two, I'm hoping to be able to release a better calibration tool.

piolho-eefe, 2023-07-19:
You were right. It does not detect the corners... I'll wait for the new calibration tool. :)
Just a question, are there any rules on how to take pictures of the checkerboard? I believe my pictures are ok for corner identification. Is there anything wrong?
Examples:
[image]
[image]
Cheers.

davidpagnon, 2023-07-19:
The pictures look okay, although the white margin could be a little larger. Could you try to remove the yellow and grey bands? The pictures could be a little flatter and cleaner but it should not prevent the detection altogether.

Another issue could lie in the Config.toml file. Did you set `corners_nb = [5,4]`? Did you specify the right image extension? And lastly, are you sure you respected the file hierarchy as specified in the documentation?

piolho-eefe, 2023-07-19:
Hi, I corrected the corners to 5, 4 (it was 5, 6 for some mysterious reason). Now, it finds the corners for most pictures, but misses the last one for camera 2. I believe that, for this reason, when I try to triangulate, it gives an error. See:
[image]
[image]
[image]
[image]

davidpagnon, 2023-07-19:
Okay, that's the problem when you lay the checkerboard flat on the ground, it is harder to detect it. Setting the cameras a little higher could help solve this issue, but overall that makes calibration not very robust. 

The tool I'm currently developing will help take care of this issue. Sorry, I still need about a week or two to finish developing it and release it!

davidpagnon, 2023-08-21:
Hi, 

I am sorry I am so late, I had a few priorities to tackle first...
I finally released a tool for a [robust and simple calibration](https://github.com/perfanalytics/pose2sim#calculate-from-scratch). If you ever end up using it, please tell me if anything goes wrong or is not clear!

closed 2023-09-22 by davidpagnon