perfanalytics/pose2sim #8: OpenSim import in Blender and/or Maya

davidpagnon, 2022-06-10:
And to import .osim file in blender i used another addon
https://github.com/JonathanCamargo/BlendOsim

But my approach was using blender armatures, it was better in my opinion.

And I was able to import the mot file, but the mot file as basically angles in degrees, and making it work properly in blender was a pain in the but, and I gave up. I imported the data, but I felt I didnt know how to use the order of angles properly. I dont know how to explain better, but if I had to import just one angle per part of the body, it works right, but if I had to import 2 or 3 angles, things get messed up.

So my approach was using the location of the marker. and to do that I had to enable one optin in the inverse knematics xml file.

the change is explaned here
https://simtk.org/plugins/phpBB/viewtopicPhpbb.php?f=91&t=13422&p=38657&start=0&view=

basically is to add the option `<report_marker_locations>true</report_marker_locations>`
and when running the inverse knimatics, it will create a .sto file with the location of the markers (i guess its the markers LOL)

_Originally posted by @carlosedubarreto in https://github.com/perfanalytics/pose2sim/issues/3#issuecomment-1152176083_

davidpagnon, 2022-06-10:
Thanks @carlosedubarreto, this is great content and I'll try to check it out when (and if) I find time!

And yes, I would assume that this is due to their usage of Euler angles. They need to be imported in the right order, and they can lead to a few issues such as gimbal lock. Check this OpenSim page for more information about their angle conventions:
https://simtk-confluence.stanford.edu:8443/display/OpenSim/Basic+Geometry+and+Mechanics

carlosedubarreto, 2022-06-10:
@davidpagnon Thats amazing, thanks for sharing with me... 
I'll try to use it in the future. sometimes angles are better to use than the location position. (so I wont have to worry with the size of the armature, it will just work 😍

davidpagnon, 2023-11-04:
I started writing a Blender add-on to import OpenSim files. It would have been a huge pain to rewrite all the joint definitions, so instead I used the OpenSim API to get the positions and orientations of each limb I the global frame of reference. It works fine though! 
https://github.com/davidpagnon/Sim2Blend
I took my inspiration from another add-on, BlendOsim (but mine does not require Matlab). 

Next I need to rig the skeleton to be able to animate it. I do not expect it to be complicated but I am still a very beginner in Blender. 

I'm also going to let the user import cameras from a calibration file, and to add a few other features. It is online and the OpenSim feature is working, but I won't advertise it right know because I've still got buttons that don't do anything 😅

davidpagnon, 2024-02-27:
Completed with Pose2Sim_Blender add-on: https://github.com/davidpagnon/Pose2Sim_Blender

closed 2024-02-27 by davidpagnon