perfanalytics/pose2sim #11: Problem when running Pose2Sim.triangulate3d()

reubenlindroos, 2022-09-29:
Config and Calib:

[Calib.txt](https://github.com/perfanalytics/pose2sim/files/9674866/Calib.txt)
[Config.txt](https://github.com/perfanalytics/pose2sim/files/9674873/Config.txt)

Project directory structure:
```
├───calib-2d
├───opensim
│   └───Geometry
├───pose-2d
│   ├───pose_cam1_json
│   ├───pose_cam2_json
│   ├───pose_cam3_json
│   ├───pose_cam4_json
│   ├───pose_cam5_json
│   ├───pose_cam6_json
│   ├───pose_cam7_json
│   └───pose_cam8_json
├───pose-2d-tracked
│   ├───pose_cam1_json
│   ├───pose_cam2_json
│   ├───pose_cam3_json
│   ├───pose_cam4_json
│   ├───pose_cam5_json
│   ├───pose_cam6_json
│   ├───pose_cam7_json
│   └───pose_cam8_json
├───raw-2d
│   └───00_img
└───User
```

I think this might be an index-1 issue as if i change the config to go from frame 1-400 it seems to run fine.

davidpagnon, 2022-09-30:
Hi Reuben, thanks for your fix!
Before I merge it, I would like to clarify something. With your fix, if you specified a range [440,847] in the Config.toml file, f_range becomes [0,407] instead of [440,847]. Is it really what you want?

My guess is that your json files are starting from 440, as such: 
[cam1.440.json, cam1.441.json, .., cam1.847.json] 
instead of: 
[cam1.000.json, cam1.001.json, .., cam1.847.json]

The problem in my code is that it indexes it based on the number of files found by listdir, instead of by the numbers in their names. I'm afraid that if I merge the fix, the [cam1.000.json, cam1.001.json, .., cam1.847.json] configuration won't work anymore.

Do you want to provide a fix indexing files on their names? If not I'll definitely do it, but it could take a little longer.

reubenlindroos, 2022-09-30:
Hi David, 
Your right, ill close this PR as it wouldn't fix the issue for all cases. But at least we know where the problem is now.  A work around (for now) is simply to change indexing before running triangulate3d. I think you know best how you would prefer to have it but i might be able to fix it next week as ill be working with your code then as well.

davidpagnon, 2022-09-30:
I started writing something complicated like this, line 400

    for c in range(n_cams): 
        json_filtered = []
        for number in range(*f_range):
            for string in json_tracked_files[c]
                if str(number) in string.split('.')[1]:
                    json_filtered+=string
        json_tracked_files[c] = json_filtered

But it only works if the json files are named `XXXX.numbers.json`, and wouldn't work with the standard OpenPose naming convention `XXXX_000000000000_keypoints.json`. It is too complicated to parse if I want some flexibility in file names. So I suppose that unless you find a better way, one will have to keep in mind that the frame range is based on the number of files in the folder, rather than based on the filename...

closed 2022-09-30 by davidpagnon