perfanalytics/pose2sim #14: OpenSim GUI crashes after loading IK_Setup_Pose2Sim_Body25b.xml

rutuja1409, 2022-10-17:
While doing demonstration 2, the below step was mentioned. 
Load the provided IK_Setup_Pose2Sim_Body25b.xml scaling file from pose2sim/Demo/opensim. (Tools -> Inverse kinematics -> Load)

After loading the model and clicking on Run the system crashes and the OpenSim GUI shuts down suddenly.

@danielskatz , @davidpagnon can you please help me out here ?

davidpagnon, 2022-10-17:
Hi @rutuja1409, 
It workds fine on my end, with pose2sim v0.2.3, OpenSim 4.3.
I highly suspect that the issue is due to OpenSim. Did you try with another version, or on another computer?

If it still does not work, please give me the following information so that I can reproduce the error:
- Which version of pose2sim you use?
- Which version of OpenSim? (on Windows, MacOS, Linux?)
- If you loaded IK_Setup_Pose2Sim_Body25b.xml, it means that you managed to install body_25b in OpenPose? If not, use IK_Setup_Pose2Sim_Body25.xml instead (25 instead of 25b).
- Did the scaling step work?
- Did you try running other classic marker-based data through OpenSim?

P.S.: Daniel Skatz just edited the layout of the JOSS paper, I doubt he will have an idea.

davidpagnon, 2022-10-18:
I still can't reproduce the error, even with your settings. 
It has been working on several computers with many different configurations, so I'm a little puzzled.

Let's try to go step by step:
- I'm almost sure the problem is on OpenSim end, can you reinstall it?
- Can you send me a screenshot of your OpenSim GUI just before it crashes, when you have loaded the IK setup file?
- Can you try `pip install pose2sim --upgrade`?
- If it still does not work, can you redo the installation procedure from scratch, but instead of `conda create -n Pose2Sim python=3.8.8` do `conda create -n test python=3.8.8`  and then `conda activate test`

rutuja1409, 2022-10-19:
I did the steps. 
Pose2Sim - 0.2.0
Opensim - 4.4.
opencv-python - 4.6.0.66
python - 3.8.10
[image]

davidpagnon, 2022-10-19:
It does not really make sense to me, because your GUI is similar to mine, and because the trc file displayed in the GUI is the one I provide, not the one you just generated (which is stored in `Demo\pose-3d` rather than in `Demo\opensim`).

So the error definitely comes from OpenSim, and I am not sure I can help about it. 
Since you tried several OpenSim versions, I wonder if there is any configuration in you computer that makes it crash. Did I understand right, you told me you can run other inverse kinematic analysis (typically, marker-based) in OpenSim?

Just in case, can you send me the last few lines of `C:\OpenSim 4.4\opensim.log`, just in case a specific error that I should be aware of has been recorded?
Does it crash instantly, or does it "try" to solve kinematics for a minute or two?

PS: 
The calibration output is alright.
The filter3D() issue should not make a difference, however you can try solving it with `pip uninstall opencv-python`,
`pip install opencv-python-headless`.

rutuja1409, 2022-10-19:
> It does not really make sense to me, because your GUI is similar to mine, and because the trc file displayed in the GUI is the one I provide, not the one you just generated (which is stored in [...]

The last few line of opensim.log 
[2022-10-19 15:45:14.645] [info] Loaded marker file /home/lahoti/.local/lib/python3.8/site-packages/Pose2Sim/Demo/opensim/Balancing_for_IK.trc (21 markers, 100 frames)
[2022-10-19 15:45:16.437] [cout] [info] 
               MODEL: Pose2Sim_scaled
         coordinates: 62
              forces: 0
           actuators: 0
             muscles: 0
            analyses: 2
              probes: 0
              bodies: 30
              joints: 30
         constraints: 20
             markers: 21
         controllers: 0
  contact geometries: 0
misc modelcomponents: 0
[2022-10-19 15:45:16.437] [info] Running tool .

It crashes after 1-2 sceond.
I ran the below inverse kinematic model - 
https://simtk-confluence.stanford.edu:8443/display/OpenSim/Tutorial+3+-+Scaling%2C+Inverse+Kinematics%2C+and+Inverse+Dynamics#:~:text=were%20scaled%20manually%3F-,IV.%20Inverse%20Kinematics,-Kinematics%20is%20the

davidpagnon, 2022-10-19:
Okay, so we have the same output, but yours stops before processing the first frame.

I must say I'm running out of ideas... Maybe you should post [on the OpenSim forum](https://simtk.org/plugins/phpBB/indexPhpbb.php?f=91), asking them why would Inverse Kinematics make the GUI crash right before it processes the first frame? I'm really not sure the problem comes from Pose2Sim, as it has worked on every single computer I tried, including with your configuration...

Let's clarify a doubt: you ran inverse kinematics on the OpenSim tutorial (with their model) and it worked. However, did you use their `.osim` model with Pose2Sim, or the one I provide?

rutuja1409, 2022-10-20:
> Okay, so we have the same output, but yours stops before processing the first frame.
>
> I must say I'm running out of ideas... Maybe you should post [on the OpenSim [...]

With Pose2Sim I used the model which you have provided.
Does it require a lot of memory and GPU space to run the inverse kinematic model ? Is there a possibility that it crash because enough GPU is not present ?

davidpagnon, 2022-10-20:
I'm not sure you need a GPU, the system requirements for OpenSim seem rather reasonable to me...
https://simtk-confluence.stanford.edu:8443/display/OpenSim/Supported+Platforms

rutuja1409, 2022-10-20:
> I'm not sure you need a GPU, the system requirements for OpenSim seem rather reasonable to me... https://simtk-confluence.stanford.edu:8443/display/OpenSim/Supported+Platforms

Okay. I am reinstalling the opensim gui. These are the instructions I'm following to install on ubuntu. Is the correct or should I follow some other ?
https://github.com/opensim-org/opensim-gui#building-on-linux-beta

davidpagnon, 2022-10-20:
Oh okay, you are building it on Linux! That explains why this is so complicated. I must say I am not skilled enough to guide you through this...
However, I have heard that some people choose to install it in a virtual machine such as Wine or VirtualBox to simulate a Windows environment, maybe you could try this?

closed 2022-10-24 by davidpagnon