perfanalytics/pose2sim #17: Error when Pose2Sim.triangulate3D()

kristofpipeleers01, 2022-11-02:
Hi David!
I'm trying to use the Pose2Sim workflow with my own pictures. I think the camera calibration succeeded but when I'm now trying to triangulate, I get the error below.
[image]

Can you give more info about the error? Many thanks for considering my request.

davidpagnon, 2022-11-02:
Hi Kristof!

I definitely need to take the time to catch errors, in order to display clearer messages...

This error usually comes when you try to index an empty array. For example:
>import numpy as np
>a = np.array([])
>print(a[:,1])

It probably means that your x_files, y_files, and likelihood_files arrays are empty, at least for one specific frame or for one specific keypoint. More likely, Pose2Sim did not manage to find any of your 2D json files. 
- Did you first run OpenPose and correctly populate your pose-2d folder with the resulting json files? 
- What was the output of your tracking step?

_P.S.:_
_1. It should not have anything to do with potential calibration issues, since the error comes prior to the proper triangulation._
_2. The error also occurs if the arrays don't have a constant shape, typically if for a given frame, one of the keypoints was missing for some cameras. But I doubt it, I believe I fixed this a while ago._

davidpagnon, 2023-03-01:
This issue has been marked as stale. Don't hesitate to reopen it if you need further help!

closed 2023-03-01 by davidpagnon