perfanalytics/pose2sim #18: Is this method possible without calibration?

SiebrenM, 2022-11-23:
Hello

I am doing research on 3d pose estimation. I was wondering if the same method could be applied without calibration? I am not the creator of the videos that I want to use, so calibration isn’t possible. I only have the camera setup available that shows the position of each camera relative to each other.

Thanks in advance!

SiebrenM, 2022-11-23:
Thank you for the fast reply, that's really helpful!

closed 2022-11-23 by SiebrenM

davidpagnon, 2023-06-20:
Referencing this alternate method for better extrinsic calibration:
https://github.com/perfanalytics/pose2sim/issues/23#issuecomment-1493291952

davidpagnon, 2023-08-21:
Just in case, I finally released [a tool for calibrating cameras](https://github.com/perfanalytics/pose2sim#calculate-from-scratch), which should make it easier and more robust. If you ever end up using it, please tell me if anything goes wrong or is not clear!