perfanalytics/pose2sim #19: question about body model file

ghdgkskghdgksk, 2023-01-27:
I have a question about OpenSim kinematics in pose2sim.

- Is model  Model_Pose2Sim_Body25b.osim   slightly modified from model  [[lifting full-body model]](https://simtk.org/projects/lfbmodel)? It seems that only the markers part has been changed to the Openpose_Body25b format, is that correct?

- How  Scaling_Setup_Pose2Sim_Body25b.xml is made? As shown in the figure below, values related to scaling were inputted manually. For what reason is it decided as follows?.

[image]

davidpagnon, 2023-01-27:
Hi, thank you for you interest! 

- Yes, it is adapted from the "lifting full-body model" for the spine, and from the "Rajagopal 2015" model for the knee. There is more information about it in the <credits> of the osim file, and in the related publications (see section "How to cite" in Readme). 

- Some values were inputed manually, but this is mostly on smaller bones and it should not have any impact on the kinematics (such as the patella). Most of the other values are calculated as a ratio of the distance measured between markers. See <MeasurementSet> in the IK setup file.

davidpagnon, 2023-03-01:
This issue has been marked as stale. Don't hesitate to reopen it for further help!

P.S.: I realize I answered a little fast. Actually, none of the values are input manually, as you can see on the screenshot. These manual values won't be taken into account unless some markers are missing while scaling. Sorry about the confusion!

[image]

closed 2023-03-01 by davidpagnon