perfanalytics/pose2sim #20: Own videos not working

SiebrenM, 2023-01-31:
Hello

I am a engineering student from Belgium and am doing research on video fusion. The last months I have been trying to use Pose2Sim with my own recorded videos, but I can't seem to figure out what goes wrong. 
First I changed the config file to match with my settings. I have a project where I have a short video of one second of me standing in a static pose, recorded with two cameras. Both videos are the same resolution. In openpose everything seems fine. The calibration also looks fine when I check the corner detection  (I use the pose2sim calibration tool.) . But when I do the triangulation, the trc file stays empty. Or only a few columns are filled. 

I played around with the settings, tried multiple videos from different perspectives,... I also tried opencap, and used the same videos generated from there in pose2sim. I noticed the trc files are (almost) completely filled, but when I try it in openpose, the scaling is all wrong. The arms are 100 times bigger compared to the body. I believe this is caused by the calibration, or am I wrong here? 

Do you have any idea what I could be doing wrong? 

Thank you for your amazing work! 

Kind regards,
Siebren

davidpagnon, 2023-01-31:
Hi Siebren,

Thank you for your interest! It is hard to know what is wrong with that little information, do you think you could send me your data so that I can try and make it work?

I must say that calibration is the least accomplished and intuitive part of the soft, and this is one of the big things I want to improve shortly, probably within about two month or so.

For now, one piece of advice I could give is for you to position the cameras roughly in front of the person, rather than behind or too far on the side. This will not be a constraint anymore later on.

SiebrenM, 2023-02-01:
Hello,

Thank you for your fast response. I'm happy to share my data with you. I added two zip files of projects I tried. Trial9 is with videos I synchronized myself. Trial6 is with videos I recorded by using openCap. Thank you for your time already! 

Kind regards,
Siebren
[siebren_trial9_pose2sim.zip](https://github.com/perfanalytics/pose2sim/files/10561087/siebren_trial9_pose2sim.zip)
[siebren_trial6_opencap.zip](https://github.com/perfanalytics/pose2sim/files/10561115/siebren_trial6_opencap.zip)

davidpagnon, 2023-02-05:
Okay, so there is an issue with your calibration: pose2sim cannot find a good enough triangulation for any of your points, so it leaves them blank. The idea is to let them be interpolated between better points, but as nothing is good here, everything is there are no points between which to interpolate.

Two cameras may be too few, but I doubt you got results with OpenCap either, did you? Two things need to be changed:
- When you calibrate (for any application, not just Pose2Sim), you should try to cover the whole field of view of the camera, especially in the corners
- You should also decrease `vid_snapshot_every_N_frames` to about 20, in order to make sure there are enough frames to calibrate from.

SiebrenM, 2023-02-08:
Thank you. I'll try again with these two tips. I'll let you know if it worked. The results in OpenCap were good with the same 2 cameras. 
Thank you for your time!

davidpagnon, 2023-03-01:
This issue has been marked as stale. Don't hesitate to reopen it if you've need further help!

P.S.: As OpenCap works by infering over 40 markers from OpenPose keypoints by machine learning, and does not simply perform mathematical triangulation, I suppose it could get a probabilistic result even with poor calibration. I am not sure I would trust it much on a complex movement, though. Do you get any confidence result with your output?

closed 2023-03-01 by davidpagnon

SiebrenM, 2023-03-03:
Hi David,

My apologies for the late response. I tried again with your help yesterday, but unfortunately it didn’t improve the results. I also tried another calibration tool, now I am sure the calibration is good. I think the only thing left to try is use more cameras. Unfortunately, our research only allows us to use two cameras. 

Thank you for your help.

Kind regards,

Siebren

sk413025, 2023-03-03:
Hi SiebrenM, 

I would recommend that you increase the camera resolution to 1920x1080, either when shooting calibration plates or A-positions. I also used two low pixel cameras before, 640x480, which will often fail until I use 2 high resolution camera triangulation to steadily succeed.

You can use two phones to get the synchronized video, refer to the OBS and plugins mentioned [here](https://github.com/perfanalytics/pose2sim/issues/22#issuecomment-1438872965).

SiebrenM, 2023-03-03:
Hi

Unfortunately I already tried it with those resolutions, but it didn’t improve the results :(

But thank you for the tip!

davidpagnon, 2023-03-09:
Yes, two cameras are the absolute limit for any triangulation, and if one keypoint is ever occluded, it cannot be proceeded at all. @carlosedubarreto made it work with only 2 cameras, but it may have been on simple movements.

I can see a few solutions to robustly solve your problem, unfortunately none of them are workable for you (yet):
- Using more cameras
- Using OpenCap, which "guesses" real markers from OpenPose keypoints
- Triangulating the whole silhouette instead of sparse keypoints, in order to obtain more information per image

carlosedubarreto, 2023-03-09:
Hello @SiebrenM .
For the calibration you could use the CEB easymocap multicam I've made for blender to calibrate everything and with that result you can convert it to the way pose2sim needs.

I'm almost sure I did that using scripts already available in pose2sim.

If you want to try, you can get the blender addon for free at.
https://carlosedubarreto.gumroad.com/l/ceb_easymocap_multicam

There you can also find tutorials showing how to use it. to calibrate the cameras.

Hope it helps