perfanalytics/pose2sim #21: Warning during triangulate3D

felixtrinoma, 2023-02-09:
Hi, 

Could you help me understand the warning I get during "triangulate3D". It does seem to compute anyway but seing how bad are the results I am wondering if something could be wrong due to this warning. 

Please see below the output from function: 

> ---------------------------------------------------------------------
> Triangulation of 2D points for DLCto3D, for all frames.
> ---------------------------------------------------------------------
> 
> Project directory: [c:\Users\felix\Documents\ScapML\DLCto3D](file:///C:/Users/felix/Documents/ScapML/DLCto3D)
>   0%|          | 0/5 [00:00<?, ?it/s][c:\Users\felix\.conda\envs\DEEPLABCUTBeforeUpdate2-3\lib\site-packages\numpy\core\fromnumeric.py:3474](file:///C:/Users/felix/.conda/envs/DEEPLABCUTBeforeUpdate2-3/lib/site-packages/numpy/core/fromnumeric.py:3474): RuntimeWarning: Mean of empty slice.
>   return _methods._mean(a, axis=axis, dtype=dtype,
> [c:\Users\felix\.conda\envs\DEEPLABCUTBeforeUpdate2-3\lib\site-packages\numpy\core\_methods.py:189](file:///C:/Users/felix/.conda/envs/DEEPLABCUTBeforeUpdate2-3/lib/site-packages/numpy/core/_methods.py:189): RuntimeWarning: invalid value encountered in double_scalars
>   ret = ret.dtype.type(ret / rcount)
> 100%|██████████| 5/5 [00:00<00:00, 11.51it/s]
> 
> 
> 
> Mean reprojection error for STER is 118.6 px (~ 0.167 m), reached with 6.0 excluded cameras. 
> Mean reprojection error for XIPH is 127.7 px (~ 0.18 m), reached with 7.4 excluded cameras. 
> Mean reprojection error for C7 is 56.1 px (~ 0.079 m), reached with 7.2 excluded cameras. 
> Mean reprojection error for T8 is 128.7 px (~ 0.181 m), reached with 7.2 excluded cameras. 
> Mean reprojection error for SC is 42.0 px (~ 0.059 m), reached with 6.4 excluded cameras. 
> Mean reprojection error for AC is 86.8 px (~ 0.122 m), reached with 6.0 excluded cameras. 
> Mean reprojection error for CP is 50.2 px (~ 0.071 m), reached with 6.4 excluded cameras. 
> Mean reprojection error for ACp is 119.7 px (~ 0.169 m), reached with 7.2 excluded cameras. 
> Mean reprojection error for AA is 34.3 px (~ 0.048 m), reached with 6.0 excluded cameras. 
> Mean reprojection error for TS is 88.4 px (~ 0.125 m), reached with 6.4 excluded cameras. 
> Mean reprojection error for AI is 45.6 px (~ 0.064 m), reached with 6.4 excluded cameras. 
> ...
> In average, 7.02 cameras had to be excluded to reach these thresholds.
> 
> 3D coordinates are stored at [c:\Users\felix\Documents\ScapML\DLCto3D\pose-3d\DLCto3D_0-5.trc.](file:///C:/Users/felix/Documents/ScapML/DLCto3D/pose-3d/DLCto3D_0-5.trc.)
> Triangulation took 0.48 s.

My interrogation is about this part: 
>   0%|          | 0/5 [00:00<?, ?it/s][c:\Users\felix\.conda\envs\DEEPLABCUTBeforeUpdate2-3\lib\site-packages\numpy\core\fromnumeric.py:3474](file:///C:/Users/felix/.conda/envs/DEEPLABCUTBeforeUpdate2-3/lib/site-packages/numpy/core/fromnumeric.py:3474): RuntimeWarning: Mean of empty slice.
>   return _methods._mean(a, axis=axis, dtype=dtype,
> [c:\Users\felix\.conda\envs\DEEPLABCUTBeforeUpdate2-3\lib\site-packages\numpy\core\_methods.py:189](file:///C:/Users/felix/.conda/envs/DEEPLABCUTBeforeUpdate2-3/lib/site-packages/numpy/core/_methods.py:189): RuntimeWarning: invalid value encountered in double_scalars
>   ret = ret.dtype.type(ret / rcount)

Could you help me with this ? 

I can send data if that is needed. 

Thank you very much,
Félix Lefebvre

davidpagnon, 2023-02-11:
Hi Félix, I see you posted, and then closed the issue. Did you figure it out? Would you mind telling me what was the issue and how you fixed it?

felixtrinoma, 2023-02-13:
Hi David, 

I used DLC_to_OpenPose function to convert my h5 files from DLC, but since I wanted to check the 3D quality of my labeled data I used labelisation h5 that where missing the likelihood column. Thus I thought this was my issue and closed the topic. 

But it seems adding the likelihood value to my json files (with a value of 0.99 since positions are from manually labeled data) did not solve it..

Thus I re-open the issue. 

Thank you if you can help,

Félix Lefebvre

davidpagnon, 2023-02-13:
All h5 files I've treated until now had an x, y, and the likelihood column, maybe they changed the format with their last version?

Could you send me a minimal example that I could work with? I'll see if I can find time to figure it out.

davidpagnon, 2023-02-13:
Actually, I think I understand: you tried to triangulate the labeled data instead of their estimates. The problem is that, like you pointed it out, that h5 files for the labeled data only have x and y columns, since likelihood would not make sense there.

It would be easier to add a likelihood column with pandas in the h5 file, and then convert it to json files with DLC_to_OpenPose.py.

Then, you could try to increase the thresholds in the Config.toml file. It should work, unless there is an issue with your calibration.

felixtrinoma, 2023-02-13:
I added the likelihood directly in the json files. Does it make any difference ? 

Before trying to investigate why my results are poor, I would like to undestand why I get the warning message before the results, as it is still present now that I have added the likelihood value. 

```
Project directory: [c:\Users\felix\Documents\ScapML\DLCto3D](file:///C:/Users/felix/Documents/ScapML/DLCto3D)
  0%|          | 0/5 [00:00<?, ?it/s][c:\Users\felix\.conda\envs\DEEPLABCUTBeforeUpdate2-3\lib\site-packages\numpy\core\fromnumeric.py:3474](file:///C:/Users/felix/.conda/envs/DEEPLABCUTBeforeUpdate2-3/lib/site-packages/numpy/core/fromnumeric.py:3474): RuntimeWarning: Mean of empty slice.
  return _methods._mean(a, axis=axis, dtype=dtype,
[c:\Users\felix\.conda\envs\DEEPLABCUTBeforeUpdate2-3\lib\site-packages\numpy\core\_methods.py:189](file:///C:/Users/felix/.conda/envs/DEEPLABCUTBeforeUpdate2-3/lib/site-packages/numpy/core/_methods.py:189): RuntimeWarning: invalid value encountered in double_scalars
  ret = ret.dtype.type(ret / rcount)
```

davidpagnon, 2023-02-13:
It should not make a difference, it is just a lot of work 😅

The warning means that you are trying to calculate the mean of no values. So I guess that the results were so poor that all cameras had to be excluded, and then there were not enough good frames to interpolate the missing ones.

That's probably an error I should catch to give a clearer message, but it would be easier if you sente your data (if they are not confidential, of course).

felixtrinoma, 2023-02-14:
Hi David, 

I can surely send some data I am testing with, I would just prefer to keep it private if possible. Could you send me an email so I can send you the data ? 

felix.lefebvre@univ-lyon1.fr or felix@trinoma.fr

Thank you, 
Félix

davidpagnon, 2023-03-01:
This issue has been addressed by email. @felixtrinoma, don't hesitate to post here if you've got any findings that you think could interest other users! The issues are also my primary source of inspiration for improving the code :)

closed 2023-03-01 by davidpagnon