perfanalytics/pose2sim #22: How to capture videos from 8 cameras simultaneously?

sk413025, 2023-02-18:
Hello everyone,

I'm wondering what kind of equipment you use to capture videos from multiple cameras? I'm using a Windows 10 desktop with four USB 3.0 ports, and I'm using Python with OpenCV to capture high-definition video from cameras with an image size of 1920x1080.

However, when I try to read from four cameras at once, I get a USB controller error message. I'm considering purchasing a PCI-e USB expansion card. I've tested a PCI-e 2.0 1x expansion card with four USB ports, but it still failed. Should I use a higher-end expansion card, such as a PCI-e 3.0 4x expansion card?

I'm still a novice in this area, so I would appreciate it if anyone with experience could share their experience or equipment configuration. Thank you very much.

* I was able to successfully perform camera calibration and triangulation using pose2sim with two cameras.

davidpagnon, 2023-02-19:
Hi,

I've used different versions of Windows and I know of people having used Linux, and MacOS should not be a problem either. The main difficulty would be compiling OpenSim on non-Windows OS, but it is feasible.

I've tried Qualisys Miqus video and underwater video cameras, Sony RX0-2, Gopro 7 and 8, and iPhones. I believe @carlosedubarreto has used webcams, but I have never tried anything plugged to the USB, I'm sorry I can't help more :/ Maybe he'll answer here, or lead you to his discord chat (CEB productions or something like that).

carlosedubarreto, 2023-02-19:
Hello @sk413025 
Most of the tests I've done were using alternative cheaper cameras similar to go pro, and also tested with cellphones.
Using cell phones I was able to record multiple (4 cameras) at the same time using usb, OBS(that free screen recorder) and an android software (https://play.google.com/store/apps/details?id=com.dev47apps.obsdroidcam&hl=pt_BR&gl=US)

The problem you are saying was a on I didnt have.
If you allow me to suggest, go first with fewer cameras in a way that you dont need expeansion, just plug then on your pc to try, if it doesnt work like that, I think an expansion card wont help.

I dot have much to give as example, but if in any way you have problems with the webcam, why not try to use cellphone cameras?

I had a great experience using them. You record each one on its own (just take care to record the cameras using the same resolution) and use a clapboard before you start doing the motion you want to record, so you will be able to sync the videos manually to feed the software.

if you dont have a claperboard, just drop an object to the floor, I use this last solution to sync my videos in blender VSE.

Hope it helps.

sk413025, 2023-02-20:
> Hi,
>
> I've used different versions of Windows and I know of people having used Linux, and MacOS should not be a problem either. The main difficulty would be compiling OpenSim on non-Windows OS, but [...]

> 
@davidpagnon  
Thank you for your reply. I'll follow your suggestion and test using Gopro and iPhones to verify if it's my camera's problem.
I also have a question, you mentioned that you haven't used anything plugged to the USB, so how do you sync the videos from Gopro and iPhones? Could you please share your experience?

sk413025, 2023-02-20:
> Hello @sk413025 Most of the tests I've done were using alternative cheaper cameras similar to go pro, and also tested with cellphones. Using cell phones I was able to record multiple (4 cameras) at the same time using usb, OBS(that free screen recorder) and an android software (https://play.google.com/store/apps/details?id=com.dev47apps.obsdroidcam&hl=pt_BR&gl=US)
> 
> The problem you are saying was one I didnt have. If you allow me to suggest, go first with fewer cameras in a way that you dont need expansion, just plug then on your pc to try, if it doesnt work like that, I think an expansion card wont help.
> 
> I dot have much to give as example, but if in any way you have problems with the webcam, why not try to use cellphone cameras?
> 
> I had a great experience using them. You record each one on its own (just take care to record the cameras using the same resolution) and use a clapboard before you start doing the motion you want to record, so you will be able to sync the videos manually to feed the software.
> 
> if you dont have a claperboard, just drop an object to the floor, I use this last solution to sync my videos in blender VSE.
> 
> Hope it helps.

@carlosedubarreto 

Thank you for your suggestion. I plan to follow your advice and test using my smartphone first.

As I haven't used `DroidCam OBS` before, I'm not entirely sure about the whole process. As far as I understand, are you saying that we can each record videos from different phones using `DroidCam OBS` and then **manually adjust the synchronization of the videos through video editing software?** Also, the clapboard you mentioned is used to help adjust the time synchronization manually, as long as all cameras capture it. Please correct me if my understanding is incorrect.

In addition, you mentioned `Blender VSE`, and I will find your articles or videos to learn how to use it. In fact, when I encountered problems with camera calibration and triangulation, I referred to a lot of your tutorials on Easymocap, which was really helpful to me. Thank you very much.

carlosedubarreto, 2023-02-20:
hello @sk413025 
Let me explain better.
The use of `droidcam` was to record all the cameras at the same time.
Using `droidcam` I recorded making each camera to be in each corner of the image in `OBSStudio`. 

I told about that, But I'm not sure if this would work well, I just tested the recording, but didnt test after recording it. So maybe its not a valid solution, sorry.

The `Clapboard `would be to use in case you record each camera at its own (when you set a camera in a place and then press record on each one of them individually). Because if you use `OBSStudio` with `DroidCam`, they will be already in sync.

And about the video to sync, I did a Tutorial explaining that.
Here is the link at the moment I explain how to do it in `Blender VSE`

https://youtu.be/S8nSDZjcaSA?t=271

Low I'm sorry you had to watch my videos, the old ones are the worst 😅(even the audio was a crap)

davidpagnon, 2023-02-20:
I see different ways for synchronizing videos:
- Either with a flash, a clapperboard, or by dropping an object like Carlos did. This can be done either in Blender, and then you export the video files ; or using ffmpeg. Using ffmpeg, you could extract all frames from the videos, check how many frames are offset, and delete them. Then you can run your 2D pose estimation directly on the images.
- Another way is to first run your 2D pose estimation on your videos, and then check when the correlation in the vertical speed of the wrist (for example) is highest, and then delete the offsetted json coordinate files. This method also works pretty well. I plan to integrate in to Pose2Sim soon, but I'm reaching the end of my PhD so it may not be before next month. In the mean time, here is the script I used, which should be usable enough :)

davidpagnon, 2023-02-20:
```
import numpy as np
import pandas as pd
import matplotlib.pyplot as plt
from scipy import signal
from scipy import interpolate
import json
import os
import fnmatch
import pickle as pk

'''
0. On transforme les json en pandas
1. On calcule les vitesses (verticales ou 2D)
2. On plot les correlations des vues 2 à 2 (sur un point particulier, ou sur tous les points, ou sur une sélection de points pondérés)
3. Dans l'idéal, on fait ça automatiqueement pour toutes les vues, en coisissant les paires 2 à 2 avec le plus haut coefficient de corrélation, 
et on demande confirmation avant de "supprimer" les frames en question (en réalité, renommées *.json.del -> option reset_sync dans le Config.toml)
'''

############
# FUNCTIONS#
############

def convert_json2csv(json_dir):
    json_files_names = fnmatch.filter(os.listdir(os.path.join(json_dir)), '*.json')
    json_files_path = [os.path.join(json_dir, j_f) for j_f in json_files_names]
    json_coords = []
    for i, j_p in enumerate(json_files_path):
        if i in range(*frames):
            with open(j_p) as j_f:
                try: 
                    json_data = json.load(j_f)['people'][0]['pose_keypoints_2d']
                except:
                    print(f'No person found in {os.path.basename(json_dir)}, frame {i}')
                    json_data = [0]*75
            json_coords.append(json_data)
    df_json_coords = pd.DataFrame(json_coords)
    return df_json_coords

def drop_col(df,col_nb):
    idx_col = list(range(col_nb-1, df.shape[1], col_nb)) 
    df_dropped = df.drop(idx_col, axis=1)
    df_dropped.columns = range(df_dropped.columns.size)
    return df_dropped

def speed_y(df):
    df_diff = df.diff()
    df_diff = df_diff.fillna(df_diff.iloc[1]/2)
    df_yspeed = pd.DataFrame([df_diff.loc[:,2*k+1] for k in range(int(df_diff.shape[1]/2))]).T
    df_yspeed.columns = np.arange(len(df_yspeed.columns))
    return df_yspeed

def speed_2D(df):
    df_diff = df.diff()
    df_diff = df_diff.fillna(df_diff.iloc[1]/2)
    df_2Dspeed = pd.DataFrame([np.sqrt(df_diff.loc[:,2*k]**2 + df_diff.loc[:,2*k+1]**2) for k in range(int(df_diff.shape[1]/2))]).T
    return df_2Dspeed

def interpolate_nans(col, *kind):
    '''
    Interpolate missing points (of value nan)

    INPUTS:
    - col: pandas column of coordinates
    - kind: 'linear', 'slinear', 'quadratic', 'cubic'. Default: 'cubic'

    OUTPUT:
    - col_interp: interpolated pandas column
    '''

[... 90 lines cut ...]
    camy = df_speed[cam2_nb-1].loc[range(*np.array(frames)),id_kpt[0]]
elif id_kpt == ['all']:
    camx = df_speed[cam1_nb-1].loc[range(*np.array(frames)),:].sum(axis=1)
    camy = df_speed[cam2_nb-1].loc[range(*np.array(frames)),:].sum(axis=1)
elif len(id_kpt)>1 and len(id_kpt)==len(weights_kpt): # ex: id_kpt1=9 set to 10, id_kpt2=10 to 15
    dict_id_weights = {i:w for i, w in zip(id_kpt, weights_kpt)}
    camx = df_speed[cam1_nb-1].dot(pd.Series(dict_id_weights).reindex(df_speed[cam1_nb-1].columns, fill_value=0))
    camy = df_speed[cam2_nb-1].dot(pd.Series(dict_id_weights).reindex(df_speed[cam2_nb-1].columns, fill_value=0))
else:
    raise ValueError('wrong values for id_kpt or weights_kpt')

f, ax = plt.subplots(2,1)
# speed
camx.plot(ax=ax[0], label = f'cam {cam1_nb}')
camy.plot(ax=ax[0], label = f'cam {cam2_nb}')
ax[0].set(xlabel='Frame',ylabel='Speed (px/frame)')
ax[0].legend()

# time lagged cross-correlation
offset, max_corr = plot_time_lagged_cross_corr(camx, camy, ax[1])

f.tight_layout()
plt.show()

```

sk413025, 2023-02-21:
@carlosedubarreto
Thank you for the explanation. I have tried the `DroidCam` + `OBS` solution and it is very convenient :)
It allows me to transmit 1980x1080 pixel images from three iPhones with only slight delay in a dedicated `wifi` environment, and triangulate A-pose and simple movements.
However, when I add a fourth phone, `OBS` encounters problems such as screen transmission or image storage crash.
Also, using `DroidCam` to connect to `USB` can only read videos from three phones at the same time, and `OBS` crashes when the fourth phone is added.

I will continue to try to identify and solve any possible technical issues and learn from the tutorial video you provided :)

@davidpagnon
Based on my previous experience with DroidCam + OBS over `wifi` and `USB`, I think the most feasible way to record multiple camera videos simultaneously is to perform 2D pose estimation and check the correlation of each camera separately, as you mentioned.
Therefore, I really appreciate the script you provided, it is a great reference.

In addition, we have decided to purchase an advanced expansion card PCI-e 8x for testing. I will update with new results. Thank you very much for your assistance, I think we can close this issue now.

closed 2023-02-24 by sk413025