perfanalytics/pose2sim #23: Errror with markertrajectory

rlpjanssen, 2023-04-02:
Hi!

I was converting video images from 3 cameras and following the steps:

from Pose2Sim import Pose2Sim
Pose2Sim.calibrateCams()
Pose2Sim.track2D()
Pose2Sim.triangulate3D()
Pose2Sim.filter3D()

The .trc files i get after filtering are no good to use in OpenSim. I added images of the failed markers:

[image]
[image]
[image]
[image]
[image]

Has anyone also had this issue?
I could really use some help.
Thanks in advance!
Kind regards,
RLP Janssen

davidpagnon, 2023-04-02:
Well there is definitely a problem at some point, since you've got no output. It's difficult to tell with that little information, but the usual issue is related to calibration. I really plan to make a tool for making it easier (there is one in the "draft" branch of the repository, but it is not ready for production yet).

First question, how do you calibrate?

rlpjanssen, 2023-04-02:
i calibrate with a checkerboard. indeed the calibration is not really good:
[image]

davidpagnon, 2023-04-02:
Yes there is definitely a problem with calibration, I have to admit that the tool I provided is not as robust and flexible as I initially thought, which is a pain since you cannot do anything without a good calibration.

I plan to start working on this relatively soon (in about 2 weeks), and to provide a few robust and easy methods. In the meantime, you can you this script:

https://github.com/perfanalytics/pose2sim/blob/draft/Pose2Sim/Utilities/calibrate_extrinsics.py

Here is how it works:
- First, keep aside the calibration file that you obtained with calibrateCams(). The extrinsic parameters are probably wrong, but the intrinsics should be fine. At least, provided that you used enough views of your checkerboard (about 20) from different angles, and that corners are correctly detected (you can verify this by setting show_corner_detection to true in the Config.toml file).
- Then, put a large checkerboard down, or just measure out any object in the scene, large of possible. This can be lines on the floor, a frame, a table, anything that you can get 3D coordinates of. You will write those coordinates in the objp matrix. 
- After that, extract an image from each viewpoint, for example with this script:
import cv2
vidcap = cv2.VideoCapture(video_path)
success, image = vidcap.read()
if success:
    cv2.imwrite("first_frame.jpg", image)  
- And then you can follow the instructions in the file: Click the chosen points on the images: the script will match these clicked image points with the measured object 3D points, take into account the intrinsic parameters, and give you extrinsic parameters that you can copy-paste in the calibration file. 
- The rest of the script is dedicated to verify that the reprojection error is low enough.

(I'm writing this from my phone on the way back from a long trip, so I apologize if this is not perfectly clear 😅)

rlpjanssen, 2023-04-02:
Hi! Thanks a lot!

How do i exaclty obtain the coordinates of on object in the scene? Can i take the checkerboard from the calibration videos?

And do i then need to take this object from each camera view? I have 3 cameras.

This was my calibration by the way:
[image]
I thought it looked quiet allright here but furtheron with the triangulate i thus had problems.

davidpagnon, 2023-04-02:
You can use any object, so a checkerboard would work. Ideally, it should be as large as possible, and contain at least 10-15 points. Don't forget that you need to write down its 3D coordinates, so in the case of a flat object such as a checkerboard, don't forget to add a 3rd coordinate that you will set to zero. 

For example, if you have a checkerboard with 3 x 4 20 cm squares, your objp would be:
objp = np.float32([[0,0,0], [0,0.2,0], [0,0.4,0], [0,0.6,0],
    [0.2,0,0], [0.2,0.2,0], [0.2,0.4,0], [0.2,0.6,0],
    [0.4,0,0], [0.4,0.2,0], [0.4,0.4,0], [0.4,0.6,0]]) 

And then you would use the script to click and retrieve the points on each camera view and obtain extrinsic parameters, one by one.

I am not sure I see any obvious mistakes in your calibration values, but I have a few questions:
- Did you check whether the corner detection was good?
- I can see that one camera is in landscape, and two in portrait. Was it actually the case, or could the mistake lay there?
- Your ret value is quite high in the 2nd camera (it should be below 1), and fu and fv are very different. Can you check this one in particular?
- Also, I don't remember what exactly but I made some changes in the calibrateCams() function that I may have not reflected in the script that you used in the standalone utility script. I would advise you to try again with the regular Pose2Sim.calibrateCams() procedure and the Config.toml file.

rlpjanssen, 2023-04-02:
The corner detection is indeed not good at all... i didn't notice this before. I get a load of 'Corners not found.'
It is in fact the case that 2 cameras were vertical and one horizontal (2 iphones and one iPad). I don't know why the corners aren't detected tho? The checkerboard is in fully visable in all frames. 

my config file looks like this:
[image]
Our calibration checkerboard in the videos look like this:
[image]

Thanks a lot already for helping me out!

davidpagnon, 2023-04-02:
You're welcome! I did realize that the opencv function I used to detect corners is not very robust. I was actually thinking of replacing it with a simple neural network, but again, that's a simple project that still requires a little bit of time. 
However, it works quite a lot better if there is a visible white margin around the checkerboard (such as on this page: https://calib.io/pages/camera-calibration-pattern-generator). I would advise to tape the paper on the other side, so that the limit between white and black is even clearer. 
And you're welcome :)

rlpjanssen, 2023-04-08:
Hi @davidpagnon,
Thanks for all the help! The calibration is going way better now, all the corners are found!
Does this look like a normal calibration to you? 
[image]
We have some problems at the triangulation now: not all marker trajectories are scaled the same way.
This is for example the knee and the hip:
[image]
This can't be right?
Thanks in advance!

davidpagnon, 2023-04-08:
Hello, your calibration does not stand out as problematic, but it is hard to say without the file. I can see that now, all cameras are in landscape mode?

For me to be able to better help you, do you think you could send me a minimal working example, with your calibration file, a few json files for each camera, and potentially your 3D file too? 
Also, are you sure that your cameras are synchronized? How do you handle this?

rlpjanssen, 2023-04-09:
Hi,

I edited the videos again: all the cameras are vertical now and not horizontal. 
The calibration looks like:
[image]

I synchronized the cameras in Davinci Resolve based on waveforms.
I will send you an e-mail with some footage and files!

rlpjanssen, 2023-04-09:
Hi!

Hereby the files that I used. I will share a link from two folders in
Google Drive (it was too large). It is the A-pose folder in which I tried
to scale myself. And the Move folder is a fraction of a movement.
The link:
https://drive.google.com/drive/folders/1dd_HkW1Y7ocsz7i5klcknEh5cLpqYN7g?usp=sharing
I've also shared it with your e-mail: ***@***.***

Thanks in advance!
Greetings, Robine

davidpagnon, 2023-04-11:
Thank you for sending me the documents!

So the problem is definitely the calibration step. 
- Intrinsics: your results should be consistent, but it would be more exact if you tried to go more in the corners, and to position the checkerboard at an angle more often than frontally to the camera plane.
- Extrinsics: this is where I think the main issue lies. What you need to remember is that the last image is used for setting the coordinate system. Consequently, the same image needs to be seen by all cameras. From the videos you sent me, I don't think this is the case.

Hang in there, calibration is complicated, but then everything should come together smoothly! (and again, I plan to code an easier process, it's just hard to find the time :'( )

rlpjanssen, 2023-04-11:
Hi David!

Thanks for all the help! We got it working now! The calibration is indeed a
bit tricky, so thanks for all the tips!

davidpagnon, 2023-04-12:
Awesome, I'm glad it helped! Did it all work all the way to the inverse kinematic step?
If so, do you want to close the issue?

rlpjanssen, 2023-04-12:
Yes, it worked all the way! Thank you! It can be closed.

closed 2023-04-12 by davidpagnon

davidpagnon, 2023-08-21:
Just in case, I finally released a tool for a [robust and simple calibration](https://github.com/perfanalytics/pose2sim#calculate-from-scratch). If you ever end up using it, please tell me if anything goes wrong or is not clear!