perfanalytics/pose2sim #24: Error with IK - Number of Work Units cannot be negative

rishivamshi, 2023-04-05:
Hi Team, 

First of all I would like to thank all the contributors of this wonderful repo. 

I am trying to follow the steps mentioned in the README file, After generating the filtered TRC files, when I tried to utilized the filtered TRC file in inverse kinematics, by replacing Balancing_for_IK.trc file, OpenSim throws the error - number of work units cannot be negative. 

Can you please help me understand my error? Is my approach right?

Thanks, 
Rishi.

davidpagnon, 2023-04-05:
Hello, you're very welcome!
It's hard to say, I've never had this issue. Can you visualize your trc file, and does it look coherent ? (file - > preview experimental data).
Did you check that the times in the IK tool corresponded to the times of the trc file? (open it with any text editor to check)

rishivamshi, 2023-04-05:
Hi, 

Thanks for the prompt reply, 
So far I am just following the steps mentioned in the README file. 

This is where I am currently, 
In the Installation, I was successfully able to do all the steps. 

After which, I moved to demonstration Part 1: Build 3D TRC file on Python, 
At this step, I was able to generate Demo_0-100.trc and Demo_filt_0-100.trc files. 

Now, I moved to Demonstration Part 2: Obtain 3D joint angles with OpenSim, 
and after following steps for scaling and IK, I was able to see skeletal model move in the visualizer window. 

However, my doubt is where are we using the TRC files (Demo_0-100/Demo_filt_0-100.trc files) in Inverse Kinematics. Because upon closer look, for IK, the xml settings file is using Balancing_for_IK.trc file. 

Can you please let me how can I use the generated TRC files in Demonstration 1 for IK. 

Thanks, 
Rishi.

davidpagnon, 2023-04-05:
Okay, I see. Yes, you're right, it uses a file that I put in the OpenSim folder, but it should be the same as the one you obtain after triangulation.

Are you telling me it's not working? If so, I'll have to check this (but tomorrow, it's half past midnight here 😅).

rishivamshi, 2023-04-05:
Yes, to the best of my understanding when the file that you have put in the opensim folder (Balancing_for_IK.trc) is replaced with the file after triangulation (Demo_filt_0-100.trc), OpenSim throws an error saying that number of work units cannot be negative. 

I highly appreciate your sincere efforts in helping us out, I am a student who is trying to learn and implement your work. It would be of great help if you could help me with my doubt at your earliest convenience. 

Have a Goodnight, 

Thanks, 
Rishi.

davidpagnon, 2023-04-06:
Okay I found the issue! This is actually what I was thinking when I asked you this: 
> Did you check that the times in the IK tool corresponds to the times of the trc file? (open it with any text editor to check)

There is a little issue with OpenSim when it reads the time range in the trc file: it reads the last frame correctly, but not the first one, which makes the IK tool search between 2.516666 and 1.666667 seconds, instead of 0.01666667 to 1.666667. Hence the negative work units error. If you replace with the right values, it should work.

I made sure both the generated file and the provided one are identical so that there is less risk of error, it will be valid at the next release.

rishivamshi, 2023-04-06:
Hi @davidpagnon ,

Thanks for the reply , I have checked if changing time as you mentioned in filtered trc will fix the issue or not , your analysis is precise to the core, it worked smoothly as desired. thank you so much for your wonderful contributions. 

Kindly let me know if I can help you in writing a script to help normalize the time values from (0.016667 to 1.6667), to (2.5166 to 4.1667), so that once the Demo_filt-0-100.trc is generated, this script can be executed for smooth transition to Inverse kinematics part.

As a student and as someone with inquisitive mind I am curious to learn and grow , I will be more than happy to contribute.

Thanks,
Rishi Vamshi

closed 2023-04-06 by rishivamshi

davidpagnon, 2023-04-06:
I'm glad it worked out! 
I actually made the change a little earlier, but I do appreciate the offer! I'm always open for help, and in case you want to participate there is a to-do list at the end of the Readme file. If you're interested in one of the items or if you have an idea for improving any other aspect, be sure to reach out!