perfanalytics/pose2sim #25: Require Help in understanding how to get .osim files for our model

anudeep2804, 2023-04-12:
Hi Team, 

Firstly thank you for the wonderful repo, it's been of great use and help to me. 

I am trying to perform inverse kinematics and I am interested in the COCO dataset, I see that the Empty_project/OpenSim doesn't have .osim and scaling setup files for COCO. 
**I am just curious about how to generate these files, are these machine-generated or is there any steps or tutorial to create my own file?** 

To give you a little background, I changed the config.toml, pose_model to COCO, and generated the TRC file (it has 14 key points). 

I am trying to perform scaling and inverse kinematics, but I do not see the required files. 

Any help from the community will be greatly appreciated. 

Thanks in advance,
Anudeep

anudeep2804, 2023-04-14:
Thanks for the reply David, will be waiting to hear back from you.

davidpagnon, 2023-04-16:
Hello,
You are right, I'm realizing I omitted to add the COCO opensim files!
I uploaded them [there ](https://github.com/perfanalytics/pose2sim/tree/main/Pose2Sim/Empty_project/opensim) but I haven't found time to test them, can you tell me if it works before I add them to the next release? I'd appreciate it :)

You can choose the COCO model in your Config.toml file, which will give the correct correspondence between keypoint numbers and keypoint names (see `skeleton.py`).
They you can proceed with tracking, triangulation, and filtering.
And then you can use the OpenSim model and setup files (with your own .trc results and time ranges). As a side remark, I'd like to point our that there is no need for marker adjustments in the scaling step, and it is greyed out on purpose. 

To be more explicit, all models are the same, only the markerset (and locations of markers) change. Also, I locked the feet and hand angles since they cannot be calculated with the COCO dataset. Concerning the scaling and setup files, I removed the foot segment and the associated markers from the body_25 files.

davidpagnon, 2023-04-16:
Also, your TRC only has 14 keypoints instead of 18 because I chose not to triangulate all the eye and ear keypoints, which are redundant with the nose one in the (simplistic) motion of the head.

anudeep2804, 2023-04-19:
Hi David,

Thank you for your assistance. Based on your explanation, I understand that scaling is not a compulsory step, and we have opted to skip it. However, we are facing challenges in executing the inverse kinematics with the .trc files generated using the provided demo data and the coco model. Despite having 14 markers in the .trc file and making necessary adjustments to the setup files (including modifying the .trc filenames and time ranges), we are still unable to successfully run the inverse kinematics.

davidpagnon, 2023-04-22:
Hi,
Actually you do need to scale the model. To make it simple, the length of each limb is adjusted according to the distance between triangulated keypoints. For example, the length of the shank must be about equal to the distance between the ankle and the knee keypoints. Without scaling, the joints on the model cannot follow the corresponding triangulated keypoints during inverse kinematics.

That may be your only problem, but maybe there are other issues early on? Have you tried to check whether your TRC coordinates were consistent? (with Mokka, or in OpenSim with file -> preview experimental data)

anudeep2804, 2023-04-22:
Hi David,

Thanks for your reply, really appreciate your interest in solving my doubts,

I understand your take on scaling now, however, I am looking for demo files for the coco model ( you have provided standing_for_scaling. trc for Body_25b ) similarly can you provide demo files for the coco model so that we could run scaling and inverse kinematics, our objective is just to run activities in OpenSim section with the .trc file generated using demo data and coco model. can you please guide me on this.

davidpagnon, 2023-04-22:
Oh okay I see. The problem is that the keypoint order in COCO does not correspond to the one in BODY_25, so I would have to run again the pose estimation. And unfortunately I don't have the original videos anymore. 

Can you tell me why you want to run COCO instead of any other model? It is rather outdated, and does not provide any foot keypoints. If you can have a correct calibration, you can still run the tutorial on "Use on you own data".

anudeep2804, 2023-05-03:
Hi David , 

Our Model that we would like to work on is more closer to COCO so we just want to run the entire pipeline using coco , we are missing scaling files in that process , 

Howver I have a follow up question , 
1) is there a way to generate our own .osim file for my model 
2) how can we generate the scaling and Ik setup files 

your help would be  greatly appreciated if you could provide us some guidelines on how to proceed further

davidpagnon, 2023-05-04:
Okay, noted!

1. The .osim file does not change much when you change the detection algorithm. 
The main difference is the MarkerSet (open you file with any text editor and do a search). Keypoints may not be detected in the same place when changing detection model. To modify the marker set, you can open the .osim file in OpenSim, and move the pink markers by hand. It may not be as accurate as with a proper optimization technique for minimizing distance pink markers and detected keypoints, but it should not be bad anyway. Then, still in OpenSim, you can choose to lock some degrees of freedom. Finally, save your .osim model.

2. The scaling and setup files can also be adapted either the the .xml files or in the OpenSim GUI. 
    For scaling, you have to change the "marker data for measurements" and the time range. You may also want to change the measurement set (2nd tab) or the  weight of each keypoint (3rd tab).
   For IK, idem (simpler, actually).

However, these questions are more relevant to OpenSim itself. I would advise you to follow their tutorials (and least number 1 and 3), which are pretty well made: https://simtk-confluence.stanford.edu:8443/display/OpenSim/Examples+and+Tutorials 

Also see this previous answer:
> To be more explicit, all models are the same, only the markerset (and locations of markers) change. Also, I locked the feet and hand angles since they cannot be calculated with the COCO dataset. [...]

anudeep2804, 2023-05-11:
Hi David,

Thanks for your support throughout , after following your instructions I was able to run the inverse kinematics part on demo data provided using coco model , however the movement looks very weird. 

and also I have noticed few discrepencies between the markers in  filtered trc file generated after (Pose2Sim.filter3D()) method and the markers in setup and Osim files for coco model 

for instance , marker name in Demo_filt_0-100.trc is LHipJ whereas in Osim and setup files its LHip for coco. is this a valid case ? 

these are steps I followed to complete the inverse Kinematics part 

1) updated the pose_model='COCO' in Config.toml file and completed the process till Pose2Sim.filter3D() and obtained the filtered trc file 
2) performed scaling upon this 
3) updated the <marker_file> </marker_file> tag in IK_Setup_Pose2Sim_Coco18 to <marker_file>Demo_filt_0-100.trc</marker_file> and performed Inverse kinematics the movement looks very different 

for the sake of knowledge and testing I updated the Demo_filt_0-100.trc with data from Balancing_for_IK.trc in Demo folder accordingly for COCO model  and then performed Inverse kinematics again , it worked fine now . 

Just wanted to understand why is it different when I followed the traditional process  mentioned . your help or thoughts would be much appreciated as always. thanks being so active and helping the community.

davidpagnon, 2023-05-12:
Hi,

- You are right, LHipJ should be LHip: I have never used the Coco and Mpi models, so I have never been able to spot the mistake! It is now corrected on skeletons.py
- Balancing_for_IK.trc is the result you would obtain by using the Body_25b model, so it makes sense that it works, as long as the osim markers are placed correctly. 

Overall, you would be better off using your own (calibrated) videos and running Coco than using my body_25b results and trying to map Coco analysis on them.

However, you would not have very accurate results, but they should not be weird.
- Can you do in OpenSim: file -> preview experimental data, and tell me if the movement of the trc file looks okay? If not, please send me your project at contact@david-pagnon.com 
- If so, can you check whether the right wrist point is located at the right wrist, etc for all of points? If not, there may be an error in indexing that I made in the skeletons.py file (right switched with left for example). If it is the case, please go to skeleton.py, and invert right with left. If not, please send me your project at contact@david-pagnon.com.

davidpagnon, 2023-06-06:
This issue has been marked as stale because it has not had any activity in the last 3 weeks. Feel free to reopen it if you need it.

closed 2023-06-06 by davidpagnon