perfanalytics/pose2sim #26: Wrong Calibration file

ClarkGrif, 2023-05-16:
Hello,

First of all, thanks for sharing your work with us!

I am trying to follow the steps mentioned in the README file, and I have reached the camera calibration. However, my calibration file is a .cal because I am using Optitrack PrimeColor cameras, which can be calibrated and synchronized through Optitrack Motive. 
How can I make it work? 

Thanks,
Clara

davidpagnon, 2023-05-16:
Hello, I don't know these files, can you send me a sample so that I can see if I can translate it to the OpenCV format?

davidpagnon, 2023-05-16:
Sure, you can send it to contact@david-pagnon.com, or preferably if it is not a binary file, you can simply copy-paste its content here.

ClarkGrif, 2023-05-16:
Thank you for your answer, I have sent it by email.

davidpagnon, 2023-05-16:
I received it, thanks!
Unfortunately, this is a binary file and I cannot read it. Is there an option to export it as a txt file?
You may need to contact your support center if this is not natively possible, they might be able to do something about it.

ClarkGrif, 2023-05-22:
I contacted the support center to see if they can do something about it.
Thank you !

claraaudap, 2023-06-05:
Hello,

I gace up with the calibration file .cal because they cannot or don't want to give me an other type of calibration file  so I want to do the calibration with a chessboard. 

I followed the steps mentioned in the README file, and I have reached the camera calibration. I'm trying to do the calibration with the chessboard, so I took videos of a checkerboard with the 8 cameras I have and put them in a different folder for each camera but when I run the following script : Pose2Sim.calibrateCams() an error appears : 

[image]

I don't understand what I'm doing wrong 
Can someone help me 

Thank you, 
Clara

PS : Attached, you will see an example of the chessboard video

claraaudap, 2023-06-05:
Attached, you will find my config file 

[Config1.txt](https://github.com/perfanalytics/pose2sim/files/11652474/Config1.txt)

davidpagnon, 2023-06-06:
Hi, this is a shame because converting a calibration would have been much easier than doing the calibration yourself...

I cannot see the attached example of the chessboard video. From the error you sent, it seems like no images are found, which could be caused by
- A wrong video extension name (beware of capital letters)
- A wrong folder name
- Not enough frames in the video

Not sure this is connected, but if I were you I would name my projet folder something different from "Pose2Sim" since it is the exact name of the package: it could confuse the system and make it look for files in the wrong folder.

If you want me to test it, you can send me you whole project folder (my email address: contact@david-pagnon.com)

Also, and although this should not have anything to do with the error, I would recommend calibrating with images rather than with videos. This way, you can choose the best ones for calibration. You can extract frames from a video with [ffmpeg](https://ffmpeg.org) for example.

ClarkGrif, 2023-06-06:
Thank you for your answer. I have sent you my file by email. As for me, I am trying what you advised me.

davidpagnon, 2023-06-06:
Thank you Clara, I got the files.
There are 2 minor issues with your Config.toml file:
- First, vid_extension should be 'avi' rather than 'AVI' (at least on Linux and Mac, it is case insensitive on Windows)
- Second, corners_nb should be the internal corners: in your case, [6,8] rather than [7,9]

Also, the way you do it, intrinsic parameters will be pretty well estimated. These correspond to the camera properties, and usually need to be calculated only once.
However, you also need to obtain extrinsic parameters, i.e. the position of all cameras as regards to your reference frame. This frame of reference can be set by putting the checkerboard down, in a way that it is visible to all cameras. In Config.toml, you can choose frame_for_origin so that it specifies whether it is the first or the last image. 

It should work with your checkerboard since it is relatively large, but I had more accurate results with using other more spaced reference points in the room. See [this issue](https://github.com/perfanalytics/pose2sim/issues/23#issuecomment-1493291952) if you want to try out this method.

claraaudap, 2023-06-07:
When I run calibrateCams(), this message appears : 

[image]

And this picture appears : 

[image]

davidpagnon, 2023-06-07:
Okay, perfect! To keep detecting corners on the next images, you can just close them one by one.
This setting is mostly set to make sure corners are correctly detected, but it is annoying to close windows one by one. To disable it, you can set "show_corner_detection" to false in Config.toml

ClarkGrif, 2023-06-07:
Will it pass by camera 2 on its own?

davidpagnon, 2023-06-07:
> But I just didn't understang how to obtain extrinsic parameters. Do I need to take an other video with the checkerboard down in a way that it is visible to all cameras ? Or, am I just laying the checkerboard in a way that it is visible to all cameras and run calibrateCams() ?

Hi Clark,
No, only one video per camera is enough for both intrinsic and extrinsic calibration.

I would recommend filming with all cameras together, showing the checkerboard to each of them carefully (exactly the way you already did), and then stepping back and laying it down to set your reference frame (i.e., your extrinisic parameters).
Alternatively, you can do the opposite: first lay the checkerboard down, and then show it to each of your cameras. In this case, `frame_for_origin` should be set to 0 instead of -1.

It will do all cameras on its own.

claraaudap, 2023-06-07:
Great, thank you I get it ! 
And last thing, what does it means : 

[image]

I'm sorry for all this questions !

davidpagnon, 2023-06-07:
Oh, so this is a different problem! It seems like a numpy function I was using just got deprecated.
I just solved it, could you type `pip install pose2sim --upgrade` to implement the change?

claraaudap, 2023-06-07:
Sorry ahah, you are answering to one people with two accounts !

So I upgrade pose2sim and the error disappears!

davidpagnon, 2023-06-07:
Awesome!
Tell me if you experience any other errors, especially with extrinsic parameters.

claraaudap, 2023-06-20:
Hello,

I tried to film with all cameras together, showing the checkerboard to each of them carefully and then stepping back and laying it down but I think it's not working well. I sent you an email with an example of my video. Can you check what I am doing wrong please. 
And is it normal that the calibration takes a lot of time for a 3minutes video ?

Thanks 
Clara

davidpagnon, 2023-06-20:
Hi Clara,

First of all, your coverage of the camera field by the checkerboard is good. 

Then, it takes long because you optimize calibration over a ton of detected points. Considering the number of points in your checkerboard, I would say you need between 10, and maximum 30 images of the checkerboard. There are two solutions to it: 
- either you increase the `vid_snapshot_every_N_frames` parameter, 
- or better, you extract all images from each video (with ffmpeg, for example) and carefully choose a few of them on which to do the corner detection. 

This is tedious, but this has to be done only once for each kind of camera. Actually, are they all the same kind, with the same lenses? If so, check the Calib.toml file you obtained: are all intrinsic parameters (especially matrix) roughly the same? If so, you don't need to do it again, intrinsic calibration has worked. 

Then there is extrinsic calibration. What I suspect is that the corner #1 in each video does not correspond to the same one on the checkerboard, when it is laid on the floor. Can you check this by setting `show_corner_detection` to true? This is the problem when cameras are all around the subject, and not in the same semi-circle (and this is why I should implement the possibility to work with a Charuco board, but that's another subject). 

There is another option for extrinsics though, which is to use the temporary script I wrote and explained there: https://github.com/perfanalytics/pose2sim/issues/23#issuecomment-1493291952
The idea is to point by hand each point you want to use for calibrating extrinsic (whether it is the checkerboard or anything in the environment), in the right order.

claraaudap, 2023-06-20:
I finished the calibration and this is the file I've got
[image]

claraaudap, 2023-06-20:
Do you think the data are good ?

davidpagnon, 2023-06-20:
Okay, so there are several things: 

- Intrinsic parameters don't jump to me as inconsistent: the "matrix" results are pretty close from camera to camera. I would tend to trust them

- Extrinsic parameters can't be good if corners are not detected when the checkerboard is on the ground, so the rest of Pose2Sim cannot work. I agree this is really a pain, however the good news is that once calibration is fixed, everything else is straightforward. 

I would really advise you to use the script I linked [there](https://github.com/perfanalytics/pose2sim/issues/23#issuecomment-1493291952). The main difference is that in the original function, extrinsic parameters are calculated from the detected corners on the last image; with the proposed script, they would be calculated from points you would point by hand (see image for example points you could label). Have a look, I edited the post to give more details about how to use the script.

[image]

ClarkGrif, 2023-06-21:
The above calibration file was generated for another video. On the other video, it detects the corners well. So do you think the data of the calibration are good?
I'm gonna try with the script.

And for the 2D tracking of person,  is it normal to have 'nan px' ?

ClarkGrif, 2023-06-21:
Thnk you for your time ! 
I'm gonna try what you gave me

claraaudap, 2023-06-21:
I finally using your script for the extrinsic parameters. 

This is what I changed on the script : 

[image]

Here you can see the content of my folder calib-2d,  Calib_checkerboard is the file of my previous calibration that I show you above : 

[image]

And this is the error I've got :

[image]

Do you know what am I doing wrong ?
Sorry for all this questions, it must be annoying for you.

claraaudap, 2023-06-21:
And I'm using a checkerboard 6x9  80 cm squares

claraaudap, 2023-06-22:
Hello, I'm using the script for the extrinsic parameters but an error appears : 

[image]

And the script looks like this : 

[image]

Do you know why I get this issue ?
Thank you

claraaudap, 2023-06-22:
I moved the cameras in line just to see if it works all the way through. And during the 3D triangulation, this error appears : 

[image]

I think it's may be because of my calibration, my calibration file looks like that :

[image]

ClarkGrif, 2023-06-22:
I don't know why, me neither, I cannot see my previous messages. 
If you want I can send you by mail my folder of th project, it can be easier.

davidpagnon, 2023-06-22:
Okay why not, send me your folder. I may not be able to work on it before next week, though.

There should be an output in the terminal that tells you about the calibration residual error (you can also find it in the log file that should be in your project folder). A good error is below a centimeter (2 or 3 cm should still give you coherent results)

ClarkGrif, 2023-06-22:
Thank you!

I just checked and the résidual calibration errors are [3.357 , 4.672 , 4.069] mm. So the results should be coherent ! 
But for the Tracking of person, the mean reprojection error is 28.5px. So maybe it comes from there. I don't know.
And for the triangulation, i've got this : 

[image]

ClarkGrif, 2023-06-26:
Hello, 
I just tried to calibrate using the script, and an error occurs when I do the 2D tracking of the person. 

[image]

I have several questions regarding the script. When I write down its 3D coordinates of my checkerboard, 6x9 80 mm squares, should I write them in mm or m? 
And when I click point by point, should I do it in rows or columns? Does it matter?

Thank you

davidpagnon, 2023-06-26:
Hello Clara, 

I am sorry I am not very available these days. Among other projects, I am actually trying to make a better calibration tool that I would really like to release around mid-July (no promise, unfortunately). 

Results are not completely incoherent (open your resulting TRC file with OpenSim: File -> Preview experimental results). But they are definitely not good enough for biomechanics. It is not normal that mean reprojection error is above the thresholds.

- Setting the cameras higher is a good idea to make sure the checkerboard is correctly detected when laid on the ground, however then the keypoint detection is not as good (you can use one of my Utility scripts [there](https://github.com/perfanalytics/pose2sim#utilities) to verify this). But I lowered the likelihood_threshold and results were not much better, so this is not the main problem

- The main issue is that one or more cameras have to be excluded every frame to meet the conditions specified in Config.toml. When you only use 3, you should expect to not have to exclude any. Of course, using more is recommended and would lead to better results, but still, it shows that there is a calibration issue. I tried tweaking parameters such as likelihood_threshold and reprojection_error_threshold, however it did not make a large difference

- I would tend to trust the intrinsic calibration results, although they are probably not perfect since the "matrix" is not the same despite calibrating cameras from the same model

- So the issue is extrinsic parameters. There are two potential explanations: 
  - either the checkerboard is still not accurately detected when laid on the ground, 
  - or this represents a too small area, which leads to good results just where it is laid, but not as good further from it

claraaudap, 2023-06-27:
Thank you for your response. It'q fine if you cannot answers right away. It's already really helpful and nice to take time to read and reply all the time.

I just share with you my result. If you want to comment you can.

When I use the script. I get this calibration result : 

[image]

I don't know if it looks fine.

But when I'm doing the 2D tracking of person; I still have "nan px" and a warning, I don't know where it comes from and I'm pretty sure it's not what I need to have.

[image]

For the triangulation, I have the same warning then for the tracking. And got this result 

[image]

[image]

And for the filter3D, I have this kind of graph : 

[image]

All of this, give me something really not workable on Opensim

davidpagnon, 2023-06-27:
If track2D does not work, nothing else will. And in this case, it is most likely a calibration issue usually on extrinsic parameters.

- Are you sure your objp coordinates are good?
- Are you sure you clicked them in the right order in the image?
- Are all objp visible in image? If not, remove the ones that cannot be seen for the concerned images.
- Do they take a large enough space on the camera frame?

I have the feeling that it would be much easier to talk through it by visioconference. I am not going to be available this week, but next week I'll be rather flexible.

ClarkGrif, 2023-06-28:
I mean I'm sure I'm doing something wrong but I don't know why !
It would be great to do a visio to talk about it. Next week is perfect for me !

claraaudap, 2023-06-28:
I can explain you how I use the script. 

First, I write the coordinates in the objp matrix of my checkerboard. As I said it's a 6x9 80 mm squares : 
[image]

I extract from all my cameras and click on every point like this : 
[image]

After I replace the matrix and distortion found in my calib.toml file which give me : 
[image]

Finally, I put this in the terninal and I've got this result : 
[image]

This result looks really weird because all the data are kinf of the same but the cameras are in different position.

ClarkGrif, 2023-06-28:
I found my mistake !! 
The result is still not workable but it's better.

closed 2023-07-04 by davidpagnon

davidpagnon, 2023-07-06:
Actually @claraaudap @ClarkGrif , if you manage to make it work with Optitrack, would you mind posting a quick tutorial in a new issue so that I can add it to the documentation? 🙏

ClarkGrif, 2023-07-06:
If it's working, I'll do that ! 
Thank you for your help with all my issues

ClarkGrif, 2023-07-07:
Hello, just a last question, what is the difference between OpenPose and the track 2D ?

davidpagnon, 2023-07-07:
- OpenPose detects the pose on videos,
- track2D chooses the right person to triangulate in case several are detected by openpose (otherwise person0 can correspond to different persons across cameras, which leads to non consistent triangulation)