perfanalytics/pose2sim #28: Extrinsic parameter retrieval using Optitrack

claraaudap, 2023-07-25:
Hello, 

In this post, I will tell you how to retrieve extrinsic parameters using Optitrack's Motive softwar.

First, you need to calibrate your cameras using Optitrack's Motive software to obtain the .cal calibration file generated by the software. Next, create a project in Visual Studio using C++ and write the following code to read the calibration file and retrieve the position and orientation matrix of your cameras. 

[extrinsics.txt](https://github.com/perfanalytics/pose2sim/files/12158638/extrinsics.txt)

Then, you must transform the 3x3 orientation matrix into an orientation vector. For this purpose, I have created the following code as I was unable to convert it using the void cv::Rodrigues function.

[transformée.txt](https://github.com/perfanalytics/pose2sim/files/12158654/transformee.txt)

I hope that's clear enough

Clara

davidpagnon, 2023-07-25:
Thanks Clara! I'll add it to the doc and tag you as soon as I upgrade Pose2Sim :)

davidpagnon, 2023-08-21:
Hi Clara, 

Just in case you need to move away from the Optitrack calibration, I finally released a tool for a [robust and simple calibration](https://github.com/perfanalytics/pose2sim#calculate-from-scratch). If you ever end up using it, please tell me if anything goes wrong!

closed 2024-07-17 by davidpagnon