perfanalytics/pose2sim #35: Rotate the video calibration files :

ANaaim, 2023-09-20:
Hello, 

I would like in order to be able to use openpose more precisely to rotate the video we are using (sorry for the illustration using paint): 

[image]

Does some code like this  would be able to modify correctly the calibration file ? I am not very aware on the different assumption on the frame definition of the camera in the final .toml file. I supposed currently that the z of the camera was pointing forward so for the above example we would need to rotate the camera of pi/2 to correct the image 

```python
   # toml_path is the path with the original calibration 
   C, S, D, K, R, T = read_toml(toml_path)
    print(C)
    all_rotation = []
    for c in C:
        print(c)
        print(rotation_to_do[c])
        all_rotation.append(rotation_to_do[c])
    R = [np.array(cv2.Rodrigues(r)[0]) for r in R]
    T = np.array(T) * 1000
    # In the illustration above we would need rot to be equal to np.pi/2 (and all the adapted rotation would be in the all_rotation list for the other camera)
    RT = [rotate_cam(r, t, ang_x=0, ang_y=0, ang_z=rot) for r, t, rot in zip(R, T, all_rotation)]
    R = [rt[0] for rt in RT]
    T = [rt[1] for rt in RT]

    R = [np.array(cv2.Rodrigues(r)[0]).flatten() for r in R]
    T = np.array(T) / 1000

    toml_write(toml_path_export, C, S, D, K, R, T)
```

Thanks in advance.

davidpagnon, 2023-09-20:
It is actually a complicated issue, which explains why I haven't released a fix yet. There are two possible cases:
- Your videos are filmed in portrait mode with a phone, a GoPro, or with most consumer devices: then OpenPose successfully detects the orientation of the video, and everything runs smoothly. I assume you are not in that case.
- Your videos may be filmed in portrait mode with cameras that do not encode their orientation (Qualisys cameras for example). Then you need to go through a two-step process:
  - Rotate your openpose videos: `bin\OpenPoseDemo.exe --video 'video_path' --frame_rotate 90`
    (you can also use ffmpeg or moviepy if your openpose version is below 1.6.1)
  - Perform calibration on your original videos, then "rotate" your calibration file
    See [this answer](https://github.com/perfanalytics/pose2sim/issues/31#issuecomment-1710288817) for a walkthrough

Or you could do all your cameras in a loop, exactly the way you suggested :)

ANaaim, 2023-09-20:
I am doing automatically the rotation in openpose as suggested here, and transforming the rotation in the toml file. In your example in the other thread you are also applying a rotation around the x axis to correct the data and I do not understand why . 

And i do not understand why i shoud do the following  : 

```
switch values in size = [sizeW, sizeH]

also switch cu and CV in matrix = [[..,.., cu], [..,.., cv], [..,..,..]]
```

As the other parameters should not be change by a simple rotation. But i might be missing something.

davidpagnon, 2023-09-20:
I'm not sure I understand, where do I apply a rotation around X?

- Do you agree that when you rotate your video (90° or -90°), you basically switch W and H?
- cu and cv are the coordinates of the center of the image. Hence if you rotate your image, you switch cu and cv
- you should also rotate fu and fv, which is equal to f/pu and f/pv (focal length divided by the x and y sizes of the pixel). But with modern cameras, they are virtually equal so I would not bother

ANaaim, 2023-09-20:
In the end i understood it, thanks ^^'

For the correction around X as in your code ang_x is a default argument with a value of np.pi : 

```python
def rotate_cam(r, t, ang_x=np.pi, ang_y=0, ang_z=0):
```

In your answer in #31  you suggested to do 
```python
rotate_cam(r, t, ang_z=np.pi)
```
I supposed that the rotation around X was supposed to be Pi. 

In the end, the rotation correction seems to work with the following (dirty) code : 

```python 
  C, S, D, K, R, T = read_toml(toml_path)

    all_rotation = []
    for c in C:
        all_rotation.append(rotation_to_do[c])
    R = [np.array(cv2.Rodrigues(r)[0]) for r in R]
    T = np.array(T) * 1000

    RT = [rotate_cam(r, t, ang_x=0, ang_y=0, ang_z=rot) for r, t, rot in zip(R, T, all_rotation)]
    R = [rt[0] for rt in RT]
    T = [rt[1] for rt in RT]

    R = [np.array(cv2.Rodrigues(r)[0]).flatten() for r in R]
    T = np.array(T) / 1000

    S_rot = np.zeros(np.array(S).shape)
    K_rot = np.zeros(np.array(K).shape)

    for ind_rot,rotation in enumerate(all_rotation):

        if rotation == 0:
            S_rot[ind_rot] = S[ind_rot]
            K_rot[ind_rot] = K[ind_rot]
        elif rotation == np.pi/2:
            S_rot[ind_rot][0] = S[ind_rot][1]
            S_rot[ind_rot][1] = S[ind_rot][0]

            K_rot[ind_rot][0,0] = K[ind_rot][1,1]
            K_rot[ind_rot][1,1] = K[ind_rot][0,0]
            K_rot[ind_rot][0,2] = K[ind_rot][1,2]
            K_rot[ind_rot][1,2] = K[ind_rot][0,2]

        elif rotation == np.pi:
            S_rot[ind_rot] = S[ind_rot]
            K_rot[ind_rot][0,0] = K[ind_rot][1,1]

        elif rotation == 3*np.pi/2:
            S_rot[ind_rot][0] = S[ind_rot][1]
            S_rot[ind_rot][1] = S[ind_rot][0]

            K_rot[ind_rot][0, 0] = K[ind_rot][1, 1]
            K_rot[ind_rot][1, 1] = K[ind_rot][0, 0]
            K_rot[ind_rot][0, 2] = K[ind_rot][1, 2]
            K_rot[ind_rot][1, 2] = K[ind_rot][0, 2]
        else:
            raise ValueError('The rotation value is not correct')

    toml_write(toml_path_export, C, S_rot, D, K_rot, R, T)

    print('Calibration file generated.\n')
```

I will clean it later, but at least a full pipeline work :) Thanks again

closed 2023-09-20 by davidpagnon