perfanalytics/pose2sim #36: Extrinsic calibration fail

Pose2Sim-fan, 2023-10-09:
Greetings.

Extrinsic calibration fails at first camera with this:
[image]

Pose2Sim-fan, 2023-10-09:
calibration works for other camera, but results are incorrect:
[image]

davidpagnon, 2023-10-10:
Hi, 

Long answer ahead!

**Bad calibration for second camera:**

I can see several potential explanations for the bad calibration:
- Is intrinsic calibration correct? The checkerboard must cover a large part of the image, and be shown under different orientations (ie, not always parallel to the camera plane). But you are right, I should add a way to get a more objective feedback to the next release. If you want to get it right now, just add this line right after line 617 in calibration.py (type `pip show pose2sim` to find the location of your package):
  ```
       logging.info(f'Intrinsics error: {np.around(ret_cam, decimals=3)} px.')
  ```
  It should be below 0.5 px.
- I don't think it would explain such large errors, but currently, large distortions are not taken into account in the point reprojection. You can see that if you use a wide angle, that parallel lines won't be actually parallel around the corners, and perpendicular ones won't be perpendicular either (look at your piece of furniture for example). It is on the to-do list, but I am not sure when I'll be able to put my hands on it yet... It could be this week, or much later. 
- How did you obtain your images for extrinsic calibration? If you use a phone for example, you cannot take a picture, because the format won't be the same as in the video (and likewise for most non research-grade cameras). You need to take a short video, and Pose2Sim will extract the first frame for you.
 
After you've checked all that, if it still does not work, you can send me your Config.toml file and the images you use for calibration.

Also, do you use only 2 cameras? This is the extreme lower limit, and may not lead to accurate results. Any phone, GoPro, etc can be used.

davidpagnon, 2023-10-10:
*Failure for first camera:*
Did you by any chance close the window instead of typing 'Y' when you were done clicking? 
Try to change line 691 with this and tell me if it solves your problem:
```_, r, t = cv2.solvePnP(np.array(objp), imgp, mtx, dist)```

Pose2Sim-fan, 2023-10-14:
>   ```
>        logging.info(f'Intrinsics error: {np.around(ret_cam, decimals=3)} px.')
>   ```
After    ```if show:            cv2.drawChessboardCorners(img, corner_nb, imgp, ret)```?

> * How did you obtain your images for extrinsic calibration? If you use a phone for example, you cannot take a picture, because the format won't be the same as in the video (and likewise for most non [...]
Extracted a frame myself from video.

>do you use only 2 cameras? This is the extreme lower limit, and may not lead to accurate results. Any phone, GoPro, etc can be used.
Yes, I put them in right-angle to each other.

> Did you by any chance close the window instead of typing 'Y' when you were done clicking?
No, I typed Y

> ``` _, r, t = cv2.solvePnP(np.array(objp), imgp, mtx, dist)```
After ```mtx, dist = np.array(K[i]), np.array(D[i])```?

I keep getting indentation-error.

```    _, r, t = cv2.solvePnP(np.array(objp), imgp, mtx, dist)
                                                           ^
TabError: inconsistent use of tabs and spaces in indentation```
![image](https://github.com/perfanalytics/pose2sim/assets/147420351/a08bfdc2-911a-4e87-8ec4-0c632f851eb3)

davidpagnon, 2023-10-14:
Okay, I just upgraded the package with the changes, so more simply you can just type `pip install pose2sim --upgrade` in powershell instead, and try again from the beginning :)

Pose2Sim-fan, 2023-10-15:
Now it gives me this: 
[image]

Is this the right version? https://visualstudio.microsoft.com/downloads/#build-tools-for-visual-studio-2022

[image]

davidpagnon, 2023-10-16:
Hi, I'm a little surprised, as Visual Studio is not required for Pose2Sim. At what stage does this happen? is it during install? Did you first create a conda environment?

Pose2Sim-fan, 2023-10-16:
Both installation and upgrade.

```
conda create -n Pose2Sim python=3.7
conda activate Pose2Sim
pip install pose2sim
```

then the error.

Or if upgrade with `pip install pose2sim --upgrade`

davidpagnon, 2023-10-16:
But does it happen when you install Pose2Sim, or when you run the python functions?

Pose2Sim-fan, 2023-10-16:
...yes?

davidpagnon, 2023-10-17:
My question is at what stage does it happen? 
When installing the package? 
Or can you install it properly, and this is when you run calibration that there is a bug?

This is really weird since Visual Studio is not required, and the changes I made are very minor, so I am trying to see where this behavior could come from.

Pose2Sim-fan, 2023-10-17:
Happens when attempting to install Pose2Sim, and happened when attempting upgrade of Pose2Sim with `pip install pose2sim --upgrade`.
When the update/upgrade failed the first time (same error) I deleted everything and attempted re-install.

Error comes up during installation with `pip install pose2sim`.

davidpagnon, 2023-10-17:
I'm not sure what's going on, it all works fine on my end.

Maybe try to completely remove your conda environment:
`conda env remove -n Pose2Sim`

Reinstall the building tools:
```
pip install wheel --upgrade
pip install setuptools --upgrade
pip install pip --upgrade
```

And then restart the installation procedure from the beginning. Not that you don't have to specify the Python version, they are all supported until Python 3.11.

Pose2Sim-fan, 2023-10-18:
`pip install pip --upgrade` results in this:
[image]
command is `-m pip install pip --upgrade`

Odd error, but doing as it said then re-did `pip install pip --upgrade` and it worked.

attempting to re-do installation:
```
conda create -n Pose2Sim
pip install pose2sim
```

[image]

I try `pip install wheel --upgrade` and it says "requirement already satisfied".

davidpagnon, 2023-10-18:
My problem here is that I can't reproduce the issue, so it makes it hard to test solutions.
Since it seems like pandas is involved, I'm trying something else. I just drafted a pose2sim release that let pandas upgrade to its last version (which also means that Python < 3.8 won't be supported).

Can you try again the whole process?

```
conda deactivate
conda env remove -n Pose2Sim
conda create -n Pose2Sim python=3.8 -y 
conda activate Pose2Sim
pip install pose2sim
```

If it still does not work, can you send me the whole console output?

Pose2Sim-fan, 2023-10-18:
it worked; different error message.

triangulation with `Pose2Sim.triangulation()` fails.

[image]

davidpagnon, 2023-11-07:
This issue has been closed because it is stale. Feel free to reopen it if needed.

closed 2023-11-07 by davidpagnon