perfanalytics/pose2sim #38: Corners not found. Fail when I try to calculate intrinsic parameters

smwscth, 2023-10-20:
I have tried to run demo  without any problems, but when I try to calculate my photo the corners can't be found. 
I tried to put the pictures in demo into my folder. The pictures in demo could still be recognized when I ran them, but the corners of my pictures could not be found.
I have examined config.toml and I'm sure 'show_detection_intrinsics = true'. 
I need help! Thanks for answering my question. I don't know what the problem is.
This is one of my picture
[image]

davidpagnon, 2023-10-20:
Hi there,

Here are the instructions, which seem to be followed well :

> Make sure that the board:
is filmed from different angles, covers a large part of the video frame, and is in focus, is flat, without reflections, surrounded by a white border, and is not rotationally invariant (Nrows ≠ Ncols, and Nrows odd if Ncols even).

The image may be a little blurry on the upper left corner but that should not be overly problematic.

The other thing I can think about is the number of corners, which is a little tricky : this is the number of internal corner. I your case, 8x11 (you lay have specified 9x12, didn't you?) 

Also,you could have a few less corners and a smaller checkerboard without a problem, just in case it is easier for you :)

smwscth, 2023-10-21:
hello,
Thank you very much for your answer!
I try 8*11 and it can work now! 
[image]

closed 2023-10-21 by smwscth