perfanalytics/pose2sim #40: HELP NEEDED - WANT TO CONTRIBUTE?

davidpagnon, 2023-10-27:
Dear all, 

I am now doing a relatively unrelated post-doc, and am working on Pose2Sim only on the side. There are tons of features that I think could be useful, and that I cannot find time to implement. If anyone wants to contribute, I'll be more than happy to oblige!

The list is at the end of the Readme file, but I'll put it here for the sake of clarity.

**MAIN PROJECTS:**
- Graphical User Interface
- Multiple person triangulation
- Synchronization
- Calibration of moving cameras
- Self-calibration based on keypoint detection

I have a clear idea of how to do each of these, so if you are not sure of the details or are not sure whether you can use or develop the skills, I can definitely tell you more about it!

davidpagnon, 2023-10-27:
**DETAILED TO-DO LIST:**

**GUI: **
▢ Simple interface [tkinter](https://realpython.com/python-gui-tkinter) (or [Kivy](https://kivy.org/) if we want someting nice and portable, or Google Colab, maybe Streamlit). See [Sports2D Demo](https://bit.ly/Sports2D_Colab) for OpenPose and Python package installation on Google Drive).
▢ GUI: Blender add-on (cf [MPP2SOS](https://blendermarket.com/products/mocap-mpp2soss)), or webapp (e.g., with [Napari](https://napari.org/stable). See my draft project [Maya-Mocap](https://github.com/davidpagnon/Maya-Mocap) and [BlendOsim](https://github.com/JonathanCamargo/BlendOsim).

**Demo: **
▢ Add videos for users to experiment with other pose detection frameworks
▢ Demo: Time shift videos and json to demonstrate synchronization
▢ Demo: Add another virtual person to demonstrate personAssociation
▢ Tutorials: Make video tutorials.
▢ Doc: Use [Sphinx](https://www.sphinx-doc.org/en/master) or [MkDocs](https://www.mkdocs.org/) for clearer documentation.

**Pose:**
▢ Support [MMPose](https://github.com/open-mmlab/mmpose), [SLEAP](https://sleap.ai/), etc.

**Calibration:**
▢ Once object points have been detected or clicked once, track them for live calibration of moving cameras. Propose to click again when they are lost.
▢ Calibrate cameras by pairs and compute average extrinsic calibration with [aniposelib](https://github.com/lambdaloop/aniposelib/blob/d03b485c4e178d7cff076e9fe1ac36837db49158/aniposelib/utils.py#L167).
▢ Support ChArUco board detection (see [there](https://mecaruco2.readthedocs.io/en/latest/notebooks_rst/Aruco/sandbox/ludovic/aruco_calibration_rotation.html)).
▢ Alternatively, self-calibrate with [OpenPose keypoints](https://ietresearch.onlinelibrary.wiley.com/doi/full/10.1049/cvi2.12130). Set world reference frame in the end.
▢ Calculate calibration with points rather than board. (1) SBA calibration with wand (cf [Argus](https://argus.web.unc.edu/), see converter [here](https://github.com/backyardbiomech/DLCconverterDLT/blob/master/DLTcameraPosition.py)). Set world reference frame in the end.

**Synchronization: **
▢ Synchronize cameras on 2D keypoint speeds. Cf [this draft script](https://github.com/perfanalytics/pose2sim/blob/draft/Pose2Sim/Utilities/synchronize_cams.py).

**Multi-person:**
▢ Person Association: Multiple persons association. 1. Triangulate all the persons whose reprojection error is below a certain threshold (instead of only the one with minimum error), and then track in time with speed cf [Slembrouck 2020](https://link.springer.com/chapter/10.1007/978-3-030-40605-9_15)? or 2. Based on affinity matrices [Dong 2021](https://arxiv.org/pdf/1901.04111.pdf)? or 3. Based on occupancy maps [Yildiz 2012](https://link.springer.com/chapter/10.1007/978-3-642-35749-7_10)? or 4. With a neural network [Huang 2023](https://arxiv.org/pdf/2304.09471.pdf)?

**Triangulation:** 
▢ [Undistort](https://docs.opencv.org/3.4/da/d54/group__imgproc__transform.html#ga887960ea1bde84784e7f1710a922b93c) 2D points before triangulating (and [distort](https://github.com/lambdaloop/aniposelib/blob/d03b485c4e178d7cff076e9fe1ac36837db49158/aniposelib/cameras.py#L301) them before computing reprojection error).
▢ Multiple person kinematics (output multiple .trc coordinates files). Triangulate all persons with reprojection error above threshold, and identify them by minimizing their displacement across frames.
▢ Pre-compile weighted_traingulation and reprojection with @jit(nopython=True, parallel=True) for faster execution.
▢ Solve limb swapping (although not really an issue with Body_25b). Try triangulating with opposite side if reprojection error too large. Alternatively, ignore right and left sides, use RANSAC or SDS triangulation, and then choose right or left by majority voting. More confidence can be given to cameras whose plane is the most coplanar to the right/left line.

**Filtering: **
▢ Implement [SmoothNet](https://github.com/perfanalytics/pose2sim/issues/29)

**OpenSim: **
▢ Add model with [ISB shoulder](https://github.com/stanfordnmbl/opencap-core/blob/main/opensimPipeline/Models/LaiUhlrich2022_shoulder.osim).
▢ OpenSim: Implement optimal fixed-interval Kalman smoothing for inverse kinematics ([this OpenSim fork](https://github.com/antoinefalisse/opensim-core/blob/kalman_smoother/OpenSim/Tools/InverseKinematicsKSTool.cpp)), or [Biorbd](https://github.com/pyomeca/biorbd/blob/f776fe02e1472aebe94a5c89f0309360b52e2cbc/src/RigidBody/KalmanReconsMarkers.cpp))

**Others:**
▢ Catch errors: try to make the code bug, and give a clear error message
▢ Conda package
▢ Docker image
▢ Run pose estimation and OpenSim from within Pose2Sim
▢ Run from command line via click or typer
▢ Utilities: Export other data from c3d files into .mot or .sto files (angles, powers, forces, moments, GRF, EMG...)

**Bugs: **
▢ common.py, class plotWindow(). Python crashes after a few runs of Pose2Sim.filtering() when display_figures=true. See https://github.com/superjax/plotWindow/issues/7.
▢ calibration.py. FFMPEG error message when calibration files are images. See https://github.com/perfanalytics/pose2sim/issues/33#:~:text=In%20order%20to%20check,filter%20this%20message%20yet..

rlagnsals, 2023-10-29:
Hi, I'm an avid user of pose2sim.
I look forward to new updates every day thanks to your amazing work.
I would like to be of help to you, but I am not a very good developer, so I think I can help you with things like making tutorial videos.
I'm really excited to see your to-do list!

carlosedubarreto, 2023-10-29:
thats a great todo list.
Pos2Sim was essential to make MPP2SOS to work, and I would like to give something back, helping to improve the tool.
The main problem I have now is also lack of time.

Some huge things happened in my life (like I had to move 4 times in less than 2 years and because of that and some other things I have basically no emergency funds, so I'm having to work my ass off to pay the debts)

the problem is that getting rid of those problems will take months.
but I'll keep following this topic, to be able to do something as soon as the moment comes.

thanks a lot for the project and for trying to make it even better.

I'm sure, with the help of others, it will go a long way and also help a lot of people!!!

ANaaim, 2023-10-30:
Might be able to help :) As suggested above, your pipeline help our lab setting our marker-less pipeline so would be good to help you too. Not sure if i would be the best for the core of the code, however as a good an end user I might to some stuff on tutorial or element for easing the use of the tool-box.

davidpagnon, 2023-11-01:
Hi everyone, I'm pleasantly surprised that anyone responded at all, thank you!

@rlagnsals, A tutorial would be great! 
Other things that do not require extensive coding skills would be to try and make the code bug, and tell me when the error message is not clear so that I can raise a specific message.
I haven't looked much into it, but hosting the documentation to [github.io](https://docs.github.com/fr/pages/quickstart) may not be too complicated. Not sure about this though.

@carlosedubarreto, I don't know the details but your life does sound crazy, and I'm amazed at the way you're involved in so many other projects already!  Good luck with everything, and let me know if things calm down a little.

@ANaaim That would be awesome! I feel like a relatively important and easy step would be to make a GUI. It does not have to be beautiful, I'm just thinking of a tkinter window with the exact same info as in the Config.toml file, but presented with tabs for each steps, and drop-down menus for each parameter. I guess a good addition to this would be to load a Config.toml file, or conversely, to save the parameters to a Config.toml file.
And I also see that you suggested something for accessing more easily the skeletons, if you are up for converting the python file to a toml one, I would also greatly appreciate this!
drop-down menus that would write a Config.toml file and

rlagnsals, 2023-11-04:
I'm eager to contribute to your project and bring value to Pose2Sim! 
I'm currently in the process of creating a tutorial video. Additionally, I've been dedicating my efforts to code camera calibration using OpenPose keypoints for contribute. However, I've faced challenges during the joint optimization process. Initially, it diverged instead of converging and subsequently got stuck in a local optimum. This was unexpected, especially since I adhered to the methodology presented in the paper.😱

BY the way, could you guide me on how I can specifically communicate with you and collaboratively work on various projects to make meaningful contributions?

davidpagnon, 2023-11-04:
This would be awesome!
I put "calibration on OpenPose keypoints" low on the list of priorities because it would take me too long to dive in the subject, but if you already started, it could go back up on the bucket list!

Here is a good place for people to talk about how they would like to contribute, but you're right that once that step is passed we should use another platform to exchange more in depth. I don't know, I'm on all platforms (WhatsApp, Signal, Telegram, and others), but Teams or Discord may be best. I'm afraid I don't have the skills or time to set up a Discord server, so unless someone wants to do it, Teams may be the most convenient. 

Here is an invite for people who may be interested in contributing: https://teams.live.com/l/community/FEAWpdmXJDNT5TA3AI

davidpagnon, 2023-12-17:
@carlosedubarreto @rlagnsals @ANaaim 
I told you I would keep you informed when it is done, there is now a new version that allows for automatic batch processing :)
More information about it here: https://github.com/perfanalytics/pose2sim?tab=readme-ov-file#batch-processing

Don't hesitate to tell me if there is anything you think should be done differently!
Next project will be (whenever I find time) to handle limb swapping, and to look more in depth into the calibration based on keypoints that @rlagnsals has been working on. And then all the rest :D

davidpagnon, 2024-02-06:
[Blender add-on](https://github.com/davidpagnon/Pose2Sim_Blender) now released!

https://github.com/perfanalytics/pose2sim/assets/54667644/5d7c858f-7e46-40c1-928c-571a5679633a

davidpagnon, 2024-02-07:
Hi Tim, 
Thank you for your appreciation and your help! 
Some people from the Université de Valenciennes are also into improving the user-experience: they are starting with conducting interviews with non-specialists (clinicians and coaches), and planned to build a GUI accordingly. It would be best not to make both of your works redundant, but there is probably space for other people working on it! 
If you desire, here is a link to the Discord discussion about it: https://discord.com/channels/1183750225471492206/1183754095782264914/1196455548795756685

davidpagnon, 2024-02-07:
It is working on my end, but maybe you need to first create a discord account?
Here is the link to the general discord, and the conversation is in the "user-experience" channel: https://discord.gg/RpUeuhFN

timungereth, 2024-02-07:
I tried with my old and a new account to get into the general discrod but it says the link is invalid or outdated, so not really sure whats the issue here.

carlosedubarreto, 2024-02-07:
Yes, @davidpagnon it says its an invalid invite.

Probably you've created the invite and didnt change the setting so it wont expire (happened a lot with me ) 😁
[image]

davidpagnon, 2024-02-07:
Sorry you were right, here is a link that won't expire now!
https://discord.com/invite/4mXUdSFjmt

davidpagnon, 2024-02-29:
Multi-person analysis is now supported!

> Sorting people across view, and then across frames.
>  Set [project] multi_person = true for each trial that contains multiple persons.
>  Set [triangulation] reorder_trc = true if you need to run OpenSim and match the generated .trc files with the static trials.
>  Make sure that the order of [markerAugmentation] participant_height and participant_mass matches the order of the static trials.

Feel free to test it and tell me if anything does not go as expected.

davidpagnon, 2024-04-02:
Multiperson analysis much faster now: as fast as easymocap, and more robust results.
A challenging test scene used to take 1h30 to be processed, it is now under 2 minutes.

davidpagnon, 2024-04-17:
Thanks to all the people who have boosted the project lately!

- Automatic synchronization is now available (thanks @hunminkim98 for working on it with me)
- Non-batch processing is possible for quick analysis

davidpagnon, 2024-09-20:
OpenSim scaling and inverse kinematics are now fully automatic and integrated into Pose2Sim!
Bonus: no need for a static trial anymore.
Thanks to @peterlololsss, we worked on it together :)

divan, 2025-01-09:
Hi, I was keeping an eye on your project for a while, as I need to develop a solution for single-camera motion capture for specific sport with export to Blender. I've been using Wonder Dynamics pipeline so far, with impressive results, but it's a closed proprietary solution, with a lot of extra functionaility that I don't need and don't control. So exploring open-source options.

While the documentation for this repo seems to be great, I'm still not sure if the tools used in this project will allow a **single camera motion estimation, including the moving camera** (so camera motion estimation is a must in my case). In case it's doable, my first thought was that I would love to have a GUI desktop app for that, and that's where I think I can potentially contribute as I have pretty good experience making cross-platform Flutter apps, including desktop ones.

Kent-McPhail, 2025-01-10:
Hi I would like to contribute and currently have quite a bit of time on my hands so point me in a direction and I could definitely get started.

davidpagnon, 2025-01-15:
Hi @divan and  @Kent-McPhail!

Thanks for your interest, that's awesome! And sorry for the late answer, I'm on paternity leave since last night so I won't be very available for a couple weeks 😊

@divan, right now we have two packages, Sports2D and Pose2Sim, although the first will probably be integrated in the latter eventually. Sports2D works with a single camera, but only for 2D motion. I still need to integrate kinematic optimization. Pose2Sim works with multiple cameras, for any motion. But monocular 3D pose estimation models are getting better, so we will probably offer the possibility to skip calibration and triangulation to go straight to the kinematic optimization part. 

I have never used WonderDymanics but I just had a look [at this video](https://youtu.be/0OFioe_Fo8w?si=gDXG1-kFPW0bG2Zr) and it does seem like they are doing a great job! I have never done camera pose estimation but I can see a lot of github projects around it, hopefully not too difficult to integrate. Of course, the company has a lot more human and financial means than an open-source solution, but I think it is achievable :)

Note that for now, the skeletons imported in Blender are not rigged, they are just individual bones. I wonder if this is something you could be interested in?

A flutter app would be awesome, too! @AYLARDJ has been working on a tkinter app lately, on [this branch](https://github.com/perfanalytics/pose2sim/commit/055d8a4fb9106f8908cfacacdb9769842f299b5d#diff-1857ddb43608d9025ba0fe7743856f8ea3fbe713a0fad04493535a73dd80e057R1). Unfortunately, I did not find time to review it yet, but he made a demo video of how it could work. Maybe you should coordinate and see how you can work together? Is there a chance for you join the discord server? 

https://github.com/user-attachments/assets/8302a9b1-f572-4866-8d18-a1c5ffa482ac

https://github.com/user-attachments/assets/0db8f1e6-84e4-4b9a-a5c4-b6f3a8b43acf

@Kent-McPhail, when you look at the to-do-list, is there anything that looks close ebough to your skillset and your interest? Or anything else that is not there but that you think could be useful?
We can probavly find a mission from this point on :)

Kent-McPhail, 2025-01-15:
Congratulations on the new kid first off. 

A couple areas I am interested in honestly.
One is a rigged skeleton in blender as I am currently trying to work with data from openbiomechanics.org where I want to be able to visualize the data in blender.  Not much direction in there other than I would like to be able

Utilities: Export other data from c3d files into .mot or .sto files (angles, powers, forces, moments, GRF, EMG...)
This seems like something I could work on as well and a good starting off point too

davidpagnon, 2025-01-20:
Great indeed! I'll answer more on the Discord channel

davidpagnon, 2025-01-21:
New: It is now possible to use custom detection and pose models:
- Other than 'lightweight', 'balanced', or 'performance' modes, you can use any other pose estimation models through [RTMLib](https://github.com/Tau-J/rtmlib) (hand, face, animal, or any custom trained models).
- The (optional) detection model, pose model, and input sizes can be written in a dictionary (**within triple quotes**) as shown below. Models can be local paths or URLs, with .onnx or .zip extensions. Make sure the input_sizes are **within square brackets**. 

  ```
  # Equivalent to mode='balanced', with body_with_feet pose model
  mode = """{'det_class':'YOLOX',
         'det_model':'https://download.openmmlab.com/mmpose/v1/projects/rtmposev1/onnx_sdk/yolox_m_8xb8-300e_humanart-c2c7a14a.zip',
         'det_input_size':[640, 640],
         'pose_class':'RTMPose',
         'pose_model':'https://download.openmmlab.com/mmpose/v1/projects/rtmposev1/onnx_sdk/rtmpose-m_simcc-body7_pt-body7-halpe26_700e-256x192-4d3e73dd_20230605.zip',
         'pose_input_size':[192,256]}"""

  # With one-stage RTMO model 
  # Requires pose_model = 'Body'. Marker augmentation won't work, Kinematic analysis will
  mode = """{'pose_class':'RTMO', 
         'pose_model':'https://download.openmmlab.com/mmpose/v1/projects/rtmo/onnx_sdk/rtmo-m_16xb16-600e_body7-640x640-39e78cc4_20231211.zip', 
         'pose_input_size':[640, 640]}"""

  # With animal pose estimation:
  # Marker augmentation won't work, and you will need to create your own OpenSim skeleton for kinematic analysis.
  mode = """{'pose_class':'RTMPose',
         'pose_model':'https://download.openmmlab.com/mmpose/v1/projects/rtmposev1/onnx_sdk/rtmpose-m_simcc-ap10k_pt-aic-coco_210e-256x256-7a041aa1_20230206.zip',
         'pose_input_size':[256,256]}"""

  # Same approach for hand or face pose estimation, check the RTMLib documentation for more information.
  ```

divan, 2025-01-22:
Thanks @davidpagnon. I'm trying to make sense of the models/tools landscape myself and so far it's hard to tell whether it's my lack of knowledge/terms, or if it's just messy :) Speaking of monocular videos, I found this example from [EasyMocap](https://chingswy.github.io/easymocap-public-doc/develop/03_fitsmpl_monocular.html), which seems to be doing exactly what I need (in fact, my main sport is figure skating):

[image]

I tried to use this repo it was absolute mess, requiring to download manually files from dropbox/onedrive/other filesharing platforms and some how cook together, and some files aren't available anymore. Not usable at all, but at least it gives hope that the open-source models are already good enough.

I also tried Meta's Sapeins models and they're worked well, but I'm struggling to understand how to use their output into anything I can use in Blender...

davidpagnon, 2025-01-23:
Hi @divan, is there any chance you could join the Discord server so that we can keep talking on a more appropriate medium? If not, don't hesitate to open a dedicated issue (even though there is no real "issue")

davidpagnon, 2025-02-18:
New: 
Just released an update of the Blender add-on to automatically create a skeleton (any from skeletons.py) upon trc imports.
https://github.com/davidpagnon/Pose2Sim_Blender

[image]

Note that the c3d and OpenSim files are not rigged yet

davidpagnon, 2025-03-17:
We are now lucky to have multiple contributors working at the same time on Pose2Sim and I'm so happy about that!
However, it means that we need to be careful when we do so, otherwise we'll end up with impossible merge conflicts and an unstable toolbox.

Here are some updated instructions about how to submit pull requests on the experimental branch, that will be merged to the main branch once everything is stable.
Note that there are other perfectly acceptable ways to do it (via [Git Desktop](https://desktop.github.com/download/) or the [gh command](https://cli.github.com/), for example), just make sure that your pull requests are targeting the experimental branch.
Please provide as much useful information as possible about how you modified the code, and a rationale for why you're making this pull request.
Up-to-date instructions on how to contribute. Remember you can be helped on the [Discord server!](https://discord.com/invite/4mXUdSFjmt): [image]

## 1. Fork the repository
Go to https://github.com/perfanalytics/pose2sim
Click on the Fork button and create your fork

<br>

## 2. Set yourself on the experimental branch
### Option 1: With git command line
Open a command line where you'd like the Pose2Sim files to be copied
```
git clone https://github.com/YOUR-USERNAME/pose2sim.git
cd pose2sim
git checkout experimental
git pull
```

### Option 2: With [Git Desktop](https://desktop.github.com/download/)
Open Github Desktop
Click File -> Clone repository
In the URL tab, set the URL of your fork to https://github.com/YOUR-USERNAME/pose2sim.git, the local path where you'd like the Pose2Sim files to be copied, and click "Clone"
Now click on "Current Branch" and switch to "Experimental"
Click "Fetch origin"

<br>

## 3. Edit the code
That's the fun part (sometimes)

<br>

## 4. Propose your edits
### Option 1: With git command line
```
git add .
git commit -m "description of your edits"
git push origin experimental
```
### Option 2: With GitHub Desktop
Describe your changes and click on the "Commit to experimental" button
Click "Push origin"

<br>

## 5. Create a pull request
### Option 1: On the Github repository website
Go to https://github.com/perfanalytics/pose2sim
Click on the "Pull request" tab, click "New pull request", click "Compare across forks"
Set "base repository" to "perfanalytics/pose2sim" and "base" to "experimental"
Set "head repository" to your fork and "compare" to "experimental"
Click "Create pull request"
Add a title and description, Click "Create pull request"
### Option 2: with [Github CLI (gh)](https://cli.github.com/)
```
gh pr create --base experimental --head YOUR-USERNAME:experimental --title "Your PR Title" --body "Description of your changes"
```

davidpagnon, 2025-05-26:
Thanks @chengyuZou! 
I'm not sure where to put it yet, but for the moment I could create a section in the Discord server, along with both blog posts @hunminkim98 wrote a while ago. No obligation but did you join? If not I'll post it myself.
[image]