perfanalytics/pose2sim #41: Might be interesting to be able to add a skeleton model without having to modify the skeleton.py function

ANaaim, 2023-10-30:
Hello, 

Just some thoughts on the fact that when using DeepLabCut model we have to ''dive'' quite far in the code to be able to modify the skeleton function to add a skeleton model to be able to perfoms triangulation. This is even more of a problem when using virtual environement because you have to go modify the function in each environment (quite a pain to have to say to each user where there are supposed to modify a built-in function when they are not very used to do programming)

I am perfectly aware that such use is quite rare, but being able to change easily the skeleton would ease the use of this with deep-lab-cut I think. 

Would it be possible you think to give a dictionary or even a calibration file with all the needed information to modify such information ? Is there a reason for having this in a function and not in calibration file ?  

Best regards,

davidpagnon, 2023-11-01:
There is actually no good reason for it! 

This file could perfectly be named "skeletons.toml" and interpreted during execution. 
However, one needs to make sure to find a way that the hierarchy and the labels (numbers) are provided in a clear way.
Added in the to-do list.

davidpagnon, 2023-11-01:
Actually, when specifying (for example) `pose_model = 'BODY_25B'` in Config.toml, the code currently looks for 'BODY_25B' in skeletons.py, and reads the hierarchy with anytree (**block L427 in triangulation.py, and L260 in personAssociation.py**). 

I suppose that if the string is not in skeletons.py, it could simply look for it in Config.toml.
That would do something like: 
```
if pose_model in locals():
  model = eval(pose_model)
elif pose_model in config.keys():
  from anytree.importer import DictImporter
  importer = DictImporter()
  model = importer.import_(config.get(pose_model))
else:
  raise NameError('Model not found in skeletons.py nor in Config.toml')
```

And the skeleton would be added at the end of the Config.toml file, kind of like that:
```
# CUSTOM skeleton (others are in skeletons.py)
# In this example, CUSTOM reproduces the BODY_25B skeleton.
[CUSTOM]
name = "CHip"
  [[CUSTOM.children]]
  id = 12
  name = "RHip"
    [[CUSTOM.children.children]]
    id = 14
    name = "RKnee"
      [[CUSTOM.children.children.children]]
      id = 16
      name = "RAnkle"
        [[CUSTOM.children.children.children.children]]
        id = 22
        name = "RBigToe"
          [[CUSTOM.children.children.children.children.children]]
          id = 23
          name = "RSmallToe"
        [[CUSTOM.children.children.children.children]]
        id = 24
        name = "RHeel"
  [[CUSTOM.children]]
  id = 11
  name = "LHip"
    [[CUSTOM.children.children]]
    id = 13
    name = "LKnee"
      [[CUSTOM.children.children.children]]
      id = 15
      name = "LAnkle"
        [[CUSTOM.children.children.children.children]]
        id = 19
        name = "LBigToe"
          [[CUSTOM.children.children.children.children.children]]
          id = 20
          name = "LSmallToe"
        [[CUSTOM.children.children.children.children]]
        id = 21
        name = "LHeel"
  [[CUSTOM.children]]
  id = 17
  name = "Neck"
    [[CUSTOM.children.children]]
    id = 18
    name = "Head"
      [[CUSTOM.children.children.children]]
      id = 0
      name = "Nose"
    [[CUSTOM.children.children]]
    id = 6
    name = "RShoulder"
      [[CUSTOM.children.children.children]]
      id = 8
      name = "RElbow"
        [[CUSTOM.children.children.children.children]]
        id = 10
        name = "RWrist"
    [[CUSTOM.children.children]]
    id = 5
    name = "LShoulder"
      [[CUSTOM.children.children.children]]
      id = 7
      name = "LElbow"
        [[CUSTOM.children.children.children.children]]
        id = 9
        name = "LWrist"
```

This is ugly and would require some testing, but it should work rather smoothly.

ANaaim, 2023-11-02:
Will try it next week :)

closed 2023-11-07 by davidpagnon