perfanalytics/pose2sim #42: Can't run scale tool when I try body25B model

smwscth, 2023-11-03:
Hello,
I try body25 and it work good, but I think it's not accurate enough because the motion would  mismatch sometimes.
So I try body25B today and the previous steps work good, but when I try to scale the model, I'm sure I followed the correct procedure but the run button is not work. If I load body25 scale file, the scale tool can run, but when I load body25b scale file, it can't work.
Thank you for your answer very much.
 
[image]

smwscth, 2023-11-03:
I found the error, I forget to change the config.toml, I change body25 to body25b and it work good.
Body25B is better than Body25 indeed but I don't think the movement is as smooth as in the demo.
How can I improve my movements? Thank you for your excellent work, I would appreciate it if you could give me some advice.
[25b.webm](https://github.com/perfanalytics/pose2sim/assets/107971536/a133ebbc-fddd-4618-b9d5-90c87072ef17)

davidpagnon, 2023-11-03:
I'm glad you found your issue!

There are a lot of reasons that would exp'aint that the movement is not smooth.
- best is to use more cameras. How many do you use?
- if you don't have many, their placement is also important : front view rather than side views, about 30 degrees from each other is better (never close to 0 or 180°, 90° is not good either). 2 cameras can work but I do not advise it.
- There are a few other parameters that you can play with at each stage. What is the result of personIdentification and triangulation? What filter did you use?

smwscth, 2023-11-06:
Hello,
I buy four new videos on the Internet, I wish they can work better! I will try more times after the videos arrive.
And I have some questions about the cameras position, do I need to shoot behind it? May I ask which seat will have a better effect?
Thank you for your answers.
[image]
[image]

davidpagnon, 2023-11-06:
With 4 cameras, I am not actually sure what would work better!
I would be interested if you could try both and tell me about it :)

smwscth, 2023-11-17:
Hello,
I received the camera and tried to continue, but the results are still less than satisfactory.
Is it possible that my camera calibration error is too large?
I noticed an abnormal twitching of the bones in my left hand, I'm sure I have three cameras that can capture my left hand,Is this because one of the cameras can't capture the movement of my left hand?
Thank you for your answer, I will try more soon.
This is the caibration error and the motion.
 
[image]

[help.webm](https://github.com/perfanalytics/pose2sim/assets/107971536/039ded7b-f772-4906-9e47-183a169527b4)

davidpagnon, 2023-11-17:
Hi, 
- The calibration error for the first camera is definitely high. Are you sure of your intrinsic and extrinsic calibration? 
- The other camera calibration errors are not great either. Overall, intrinsic error should be bellow 0.5 px, and extrinsic error below 2 cm. Can you redo the process after:
  - upgrading Pose2Sim to make sure you have the last version (which gives you intrinsic errors)? `pip install Pose2im --upgrade`
  - setting `overwrite_intrinsics` to true in Config.toml (to make sure your intrinsic parameters are recalculated)?
- Did you filter your data?

Out of curiosity, how did you position your cameras after all? And which kind of cameras are they?

rlagnsals, 2023-11-18:
@smwscth 
I'm also using the 4 cameras, and I've done a lot of demo tests, and the second photo in your post gives the best 3D reconstruction results.

davidpagnon, 2023-11-18:
Could you by any chance send me your data? I am still a bit surprised of the results 🤔

smwscth, 2023-11-21:
> Could you by any chance send me your data? I am still a bit surprised of the results 🤔
Hello,
I'm sorry for not replying in time. Something happened these two days. I made a new test this morning and the result is much better.
But my calibration result is not good still, I think maybe my calibration of camera intrinsic parameters are not good, I will try to another set of images.
I'd be happy to send you my data, thank you for your answer very much, I will send to your email today.
This is my new result and calibration error.
[good_result.webm](https://github.com/perfanalytics/pose2sim/assets/107971536/a5d7c8df-d732-4639-9b91-5bbc03f56307)
[image]

smwscth, 2023-11-21:
This is my new data, I think my intrinsic calibration is not good and I will try again soon, thank you for your answer! 

天霜雪
***@***.***

 

 

从QQ邮箱发来的超大附件

test_Pose2Sim.rar (112.24M, 2023年12月21日 10:38 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=753833328672c59a7ca9171b40650b4d4b4c0a06525d5c501e5c045353480c03055b1e0602560c4f010a005751065a5701000a0766743916564b476d360a4a07016b5a5f481758103305&code=3832fe9b

davidpagnon, 2023-11-21:
Thank you for sharing your data.

I see two issues: 
- Your checkerboard is too far from your cameras: it needs to cover as much of the picture as possible. Your intrinsic calibration error is above 1 px, it should be below 0.5 px. You can redo the intrinsic calibration without having to start over from the beginning.
- But the main issue is in your intrinsic folder: you may have made a mistake as the images of `int_cam1_img` are the ones of camera 2.

Could you address it and tell me how it goes?

smwscth, 2023-11-23:
Hello,
Thank you for correcting my mistake, I re-calibrated the internal parameters and made a longer and more distinguishable box for the external parameters, I believe the results will be better. I will update the new result soon.

smwscth, 2023-11-28:
Hello,
Now I can get a smooth motion, thank you for your help!
But I still can't get the calibration error down to 30mm, actually I get one camera error down to 30mm and others are near 90mm, I can get a smooth motion but I think I can do better future. 
I think when I put the checkerboard near the camera, the focal length of the camera is different from when it was shot, maybe I need a much bigger checkerboard and stay away from the camera?
[good_resut.webm](https://github.com/perfanalytics/pose2sim/assets/107971536/b2819ffc-034f-4a09-b734-3bc0559d75bf)

davidpagnon, 2023-12-13:
This issue has not gotten any contribution in a long time and has been marked as stale. Please open it again if you need it!

closed 2023-12-13 by davidpagnon