perfanalytics/pose2sim #43: img = cv2.imread(img_path) seems to not return error when a path to a video is given

ANaaim, 2023-11-06:
In the following code in Pose2Sim\calibration it seems that img=cv2.imread(img_path) does not return a error when a path to a video is given and return a None as a result there is an error later on ==> I think we should test if img is None as well. 

If you agree, I can do a pull request.

```py

def findCorners(img_path, corner_nb, objp=[], show=True):
.
.
.
    try:
        img = cv2.imread(img_path)
    except:
        with suppress_stdout_stderr():
        # with warnings.catch_warnings():
        #     warnings.simplefilter("ignore")
            cap = cv2.VideoCapture(img_path)
            ret, img = cap.read()
    gray = cv2.cvtColor(img, cv2.COLOR_BGR2GRAY)
    img = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)
```

ANaaim, 2023-11-06:
After some reading, i am pretty sure of  the behavior of cv2.imread(img_path)

The OpenCV cv2.imread
function then returns either of two values:

    A NumPy array representing the image with the shape (num_rows, num_cols, num_channels)
    , which we’ll discuss later in this tutorial
    A NoneType
    object, implying that the image could not be loaded

davidpagnon, 2023-11-07:
Awesome, thank you!

I propagated this change to the other occurrences of it in extrinsic calibration, and I also deleted the function suppress_stdout_stderr which is not needed anymore.

As always, to get the new version you can do `pip install Pose2Sim --upgrade`.

closed 2023-11-07 by davidpagnon