perfanalytics/pose2sim #45: When redoing the automatic checkerboard Calibration extrinsics parameters the automatic point should be remove

ANaaim, 2023-11-06:
I don't know if it is the expected behavior but when trying to redo the automatic extrinsics parameters calibration the labeled point are kept on the image making the process of pointing the checkerboard manually quite ... impossible. 

Just to check is it the expected behavior ? 

The point were clicked to show you that both are on the same picture. 
[image]

best regards,

davidpagnon, 2023-11-07:
You're right, this is unreadable. I think I will leave only the 4 outer corner numbers in order to help the user know in which order to click them, and remove the others. I won't have time to do it right now though, maybe next week-end (so feel free if you want to fix it).

However I think your checkerboard is quite small (or the image has a rather low definition), I would suggest using a larger one or points on the scene.

davidpagnon, 2023-11-07:
Actually wait a minute, I'm going to do it :D

davidpagnon, 2023-11-07:
Done!

closed 2023-11-07 by davidpagnon

ANaaim, 2023-11-07:
^^' ok I was already trying to do it but thanks

davidpagnon, 2023-11-07:
Oh sorry!
I guess next time I will consider that when you are suggesting a change, you are also trying to implement it :)

ANaaim, 2023-11-07:
^^ it is for you, I love when thing are not done by me (one of my teacher always said : be lazy but intelligent) but not when I know that someone is doing the job for me...  

I tested it, it make it more readable, but i would remove everything in the case that the user would use the key c. But as you said i should have used a bigger checkerboard.