perfanalytics/pose2sim #50: Question about COCO_133 Skeleton

gokcergun, 2023-11-08:
Hi, 

Thank you for developing the Pose2Sim project; it appears to be incredibly useful. I have a couple of queries I'm hoping you can help clarify:

1. I noticed a potential indexing discrepancy in the skeleton.py file related to the Coco_133 skeleton model. According to the COCO dataset documentation, indexing starts from 1, as seen in the annotation illustration [here](https://github.com/jin-s13/COCO-WholeBody/blob/master/imgs/Fig2_anno.png). However, in the skeleton.py file, it seems the indices follow zero-based indexing, but still COCO_133 model has the same indices as in the documentation. Should the indices -1 or is this adjusted elsewhere within the codebase?

2. Can I utilize an alternative pose estimation model, such as HRNet_w48, which also returns COCO_133 keypoints, and incorporate the COCO_133 skeleton model into my configuration file? 

Thank you!

davidpagnon, 2023-11-08:
Hi there,

And thanks for your feedback! I'm sorry I cannot run tests right now because I am swamped, so I will try to answer by memory. 

1. It is quite possible that I shifted the indices! Please tell me if this works better after shifting, I'll do the change in the next release. 

2. I haven't tried it but it should work, as long as your pose estimation files are separated in camera folders, and separated frame by frame. Check Readme.md and the Utilities folder, you may find the tools you need for converting the files. Don't hesitate to reach out if this does not work! 

 However, be aware that I did not implement all Coco_133 points in skeletons.py. This is because the default OpenSim model I provide does not allow for finger (nor face) motion. So only overall hand motion is supported. Depending on you goal, if you want more keypoints you can do two different things:
- If you are only interested in the triangulated points, either edit skeletons.py, or add a Coco_133_full model at the end of Config.py with all the points you are interested in. You can read the resulting trc in OpenSim: File - > Preview experimental data. 
- If you want inverse kinematics, you also need a more detailed OpenSim skeleton. Once this is done, you will need to use the OpenSim Gui to add these markers to the model.

gokcergun, 2023-11-08:
Thank you for the comprehensive response. For now, using only the body keypoints suffices for my needs, but later I'll consider incorporating additional hand and face keypoints as you've described. It would be nice to see the results. 
I guess, in my case, it doesn't matter which pose_framework I choose in the config (except openpose), is it true?

Thank you!

davidpagnon, 2023-11-08:
No, it does not matter as long as you convert the files to the openpose format :)

davidpagnon, 2023-12-13:
Hi @gokcergun, could you confirm that you need to shift the keypoint indices to 1 to get them zero-based? (like point #1 in skeleton.py should become point #0?)
If so, I'll write the change (or else you can just submit a pull request and I will merge it)

gokcergun, 2023-12-13:
Hi @davidpagnon Sorry, I could not test it in Pose2Sim yet. But, if you don't correct COCO indexing somewhere else, then I think it should be -1 in skeleton.py. Please compare, i.e. nose index in the attached image with the coco [doc](https://github.com/jin-s13/COCO-WholeBody/blob/master/imgs/Fig2_anno.png)
[image]

If you want, I check it this weekend.

davidpagnon, 2023-12-14:
I fixed it [there](https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/skeletons.py), could you tell me if I shifted numbers the right way?

gokcergun, 2023-12-14:
Yep, it seems correct to me. I'll double check with data in a couple of days.

closed 2023-12-26 by davidpagnon