perfanalytics/pose2sim #51: Why the name of the camera in intrinsic calibration is set to cam_01 instead of the true name of the video ?

ANaaim, 2023-11-10:
Hello, 

In calibration.py the function calibrate_intrinsics ''change the name of the camera'' : 
C.append(f'cam_{str(i+1).zfill(2)}')   

Should not it be like in the conversion files (example in qualysis) where the name of the camera is the name of the camera in the folder ? 
I need these information later on and having the camera in the calibration file having the same name as in the ''real camera'' would be easy. 

Best regard,

davidpagnon, 2023-11-10:
I don't see any reason for it, so if you feel like proposing the change, feel absolutely free about it! 😊

ANaaim, 2023-11-10:
Great, I will do the PR ==> for me would be more logical to have the name of the cam to be sure to associate the information to the correct camera.

closed 2023-11-11 by davidpagnon