perfanalytics/pose2sim #53: Problem with Triangulation - ValueError: cannot reshape array of size 0 into shape (0,newaxis)

LempereurMat, 2023-11-17:
Dear David,
Thank you for your pipeline. I tried to use it with synchronized video from Vicon.
I have no problem with OpenPose, Calibration and Tracking the person. When I tried the triangulation, I have this error : ValueError: cannot reshape array of size 0 into shape (0,newaxis).
When I change interpolation parameter to 'none', it works but the trajectories are set to 0.
Do you have any idea ? I can send my project directory if you need.
Best regards,
Mathieu

LempereurMat, 2023-11-17:
You will find a zip file with the config file, the vicon calibration file and the first json coordinates from the 2 cameras.
[Pose2Sim from Vicon.zip](https://github.com/perfanalytics/pose2sim/files/13391820/Pose2Sim.from.Vicon.zip)

davidpagnon, 2023-11-17:
Do you only have 2 video cameras? You sent me only two json files, although I can see much more in your calibration file. Are all the others opto-electronic (marker) cameras?

LempereurMat, 2023-11-17:
We have 2 video cameras, the others are opto-electronic cameras.

davidpagnon, 2023-11-17:
This is what I was thinking: the conversion tool converts all cameras to my .toml file, not only videos but also opto-electronic (markers). I need to have a better way to filter out all non-video cameras, but I do not currently have an extensive list of them.

I just contacted Vicon again to try and get an answer, in the meantime the fix I just implemented will hopefully make it work on most Vicon hardware. Just replace your calibration.py (see where it is installed with `pip show pose2sim`) [with the new one here](https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/calibration.py).

However, I could not test it because no person was detected on the second camera you sent me, and of course I cannot triangulate with only one camera. Also, do you only use 2 video cameras? This is the absolute minimum for triangulation to work, and you will not benefit from some of the features that make the process particularly robust with more cameras.

davidpagnon, 2023-11-20:
Awesome then!

closed 2023-11-20 by davidpagnon