perfanalytics/pose2sim #54: Struggling with Camera Calibration

UdolfSeelenfrost, 2023-12-10:
Hello David,

I am trying to test your pose2sim program on the fit3d dataset.
Triangulation starts like this:
[image]
and ends like this:
[image]

You have previously pointed out that this may point towards mistakes in camera calibration. If I dont use interpolation, the triangulation does finish but the results are terrible.

The fit3d dataset provides camera parameters in JSON format and I have done my best to manually translate them to your camera calibration format, but I may have made a mistake of course. For example I have used the cv2.Rodrigues method to generate a rodrigues vector from the rotation matrix.
I have attached my calibration files, I would super appreciate it if you could take a look as I'm a bit stuck :)

[ProblemImages.zip](https://github.com/perfanalytics/pose2sim/files/13629579/ProblemImages.zip)

UdolfSeelenfrost, 2023-12-11:
All good, thanks for the reply. It's not my real name, no :)

davidpagnon, 2023-12-11:
Hi again, 
I could not find time to really check it by myself, but:
- intrinsic parameters seem okay. You chose intrinsics without distortion, but it should not alter results much. Just in case you want to take them into account, the distortion coefficients that I use are k1, k2, p1, p2
- for extrinsic parameters, can you try: 
   ```python
   RT = [[R11, R12, R13], [R21, R22, R23], [R31, R32, R33], [T1, T2, T3]]
   R = cv2.Rodrigues(RT[:,:3])[0].squeeze()
   T = RT[:,3]
   ```
   and tell me if triangulation gets any better?

If not, try transposing R (lines become columns).
Anf if still not, I'll try to debug it by myself.

UdolfSeelenfrost, 2023-12-11:
Your first suggestion for the extrinsics is how I have done it so far.
I have tried your second suggestion as well, unfortunately didn't change the result.
I have sent you an E-mail with my current project as I can't share the data publicly

davidpagnon, 2023-12-13:
Hi there, sorry I'm kind of in a rush right now but I'll check it by the end of this week or the following!

RobinBaumann, 2023-12-13:
Hi, it looks like the format in which the Fit3D Dataset provides the camera "extrinsics" is a bit messed up. The provided translation is the camera's translation in world coordinates (like you would use it in the camera pose), but the rotation is the true extrinsic rotation (or world-to-camera transform if you wish). I can successfully project the keypoints into the image by transforming `t` according to:

```python
frame_id = 0 #annotations[action_name]['fr_id']
j3d = j3ds[frame_id]
R, t = cam_params['extrinsics']['R'], cam_params['extrinsics']['T']
t_q = (-R @ t.T).T
j3d_in_camera = (R @j3d.T).T + t_q
frame = frames[frame_id]
j2d_camera = project_3d_to_2d(j3d_in_camera, cam_params['intrinsics_w_distortion'], 'w_distortion')
plot_over_image(frame, j2d_camera)
```
[image]

you can try this out with the notebook provided by the Fit3D authors [here](https://github.com/sminchisescu-research/imar_vision_datasets_tools).

davidpagnon, 2023-12-13:
Nice, thank you @RobinBaumann!
@UdolfSeelenfrost, can you try it and tell me how it goes? Just in case, I found the functions Robin mentioned [at this address](https://github.com/sminchisescu-research/imar_vision_datasets_tools/blob/main/util/dataset_util.py).
Keep me updated if you need further help!

UdolfSeelenfrost, 2023-12-16:
Okay, that was is @RobinBaumann, thank you. Following function worked in the end:

`#2. R is world, T is camera
def rodrigues2(R, T):
    R_rod = cv2.Rodrigues(R)[0].squeeze()
    T_q = (-R @ T.T).T #same as T = - R.dot(T)

    return R_rod,T_q`

The result is a bit janky but not too bad:
https://github.com/perfanalytics/pose2sim/assets/71120700/911eb586-2921-4f4a-987b-bd9703787058

I would love to try it with the (apparently better) body_25b model but the download links from openpose aren't working atm...

closed 2023-12-16 by UdolfSeelenfrost

davidpagnon, 2023-12-16:
How many cameras did you use? Are any of them filming the sagittal plane? I am soon going to release an upgrade where triangulation first tries to swap right and left sides before removing a camera in order to get the best reprojection error. 

In the meantime, you can try to filter the data a bit more, and to remove the set a weight of 0 for the nose keypoint in IK (since you are probably not tracking the rotation of the head).

UdolfSeelenfrost, 2023-12-17:
There are four cameras, positioned at 45°, 135°, 225°, 315° from the eyes of the actor. So there is no sagittal plane being filmed I think. A stronger filter and removing the nose sounds like good ideas!