perfanalytics/pose2sim #56: .qca.txt

lishuaifu, 2023-12-13:
Hello, I want to know how to export the .qca.txt file of Quaanalysis, I think I can only get .qca file

davidpagnon, 2023-12-13:
Hi, I'm not sure as I do not use QTM myself, but could you try to just add a ".txt" at the end of the filename?

lishuaifu, 2023-12-13:
Hello, I'm sorry. I tried this method, and after opening it, there was garbled code

davidpagnon, 2023-12-13:
I am not sure then, sorry... There must be an option to export calibration in the Qualisys software since many others have done it, but right now I cannot check it since I don't have access to the software...
When you find it, can you answer here so that I add the precise explanation to the tutorial?

lishuaifu, 2023-12-13:
No problem. I'll check again

ANaaim, 2023-12-14:
I think that when you export the calibration you can add the txt at the end : 
[image]
then : 
[image]

davidpagnon, 2023-12-14:
Thanks Alexandre! 
@lishuaifu here is the solution :)

lishuaifu, 2023-12-14:
Thank you very much. I'll try it

 

 image.png (view on web)
 image.png (view on web)

lishuaifu, 2023-12-14:
Thank you very much. I also export from this interface before, I did not want to add .txt suffix at the end, I will try again！

closed 2023-12-26 by davidpagnon