perfanalytics/pose2sim #57: Body_25b not available anymore?

davidpagnon, 2023-12-16:
It seems like the link to download the Openpose experimental models does not work anymore.
I mentioned it on their Github. In the meantime, the other models work fine!

https://github.com/CMU-Perceptual-Computing-Lab/openpose_train/issues/61#issuecomment-1858904069

davidpagnon, 2023-12-17:
[Here is a link](https://computingservices-my.sharepoint.com/:f:/r/personal/dp2032_bath_ac_uk/Documents/Shared/Pose2Sim/[backup_models](https://computingservices-my.sharepoint.com/:f:/r/personal/dp2032_bath_ac_uk/Documents/Shared/Pose2Sim/backup_models?csf=1&web=1&e=Nk3sTf)?csf=1&web=1&e=Nk3sTf) to download the body_25b and body_135 models while the OpenPose train website is out.

You first need to require access before you can download them though.

davidpagnon, 2024-01-08:
Better since you don't have to send me an email: someone uploaded the models there: https://github.com/aarav2you/openpose-models/tree/main/pose/1_25BBkg/body_25b

Just follow [the instructions there](https://github.com/CMU-Perceptual-Computing-Lab/openpose_train/tree/master/experimental_models#body_25b-model---option-2-recommended), and but download the caffemodel above instead.

hmejbri, 2024-03-28:
> Better since you don't have to send me an email: someone uploaded the models there: https://github.com/aarav2you/openpose-models/tree/main/pose/1_25BBkg/body_25b
>
> Just follow [the instructions [...]

This link does not work anymore. Is there any other link I can use to download body_25b

hmejbri, 2024-03-28:
> Better since you don't have to send me an email: someone uploaded the models there: https://github.com/aarav2you/openpose-models/tree/main/pose/1_25BBkg/body_25b
>
> Just follow [the instructions [...]

I cant seem to find any link to download the caffe model for the body_25b

davidpagnon, 2024-03-28:
The link I first sent is dead apparently, and the lab I work for does not let me give download access to everyone. If for one reason or another, you can't download via @hunminkim98's drive, send me an email and I'll send you a direct link!

hmejbri, 2024-04-16:
Is there any link to download only the body_25b model ? I'm working on an aws machine and I dont have 12GB to download all the models

hmejbri, 2024-04-17:
> Send you an email ([dp2032@bath.ac.uk](mailto:dp2032@bath.ac.uk)) and I will share a link for downloading only the files you need (roughly 100 MB for one model). My university policy won't let me publicly share a link, I need an email address.

Hi, I sent you an email yesterday and still no response, here is my email in case you didnt receive it (hmejbri431@gmail.com)

davidpagnon, 2024-07-11:
I'm closing this as we don't use OpenPose anymore (still supported though). Feel free to contact me if you need the models anyway!

closed 2024-07-11 by davidpagnon