perfanalytics/pose2sim #59: Questions about the bone points of the body25b

smwscth, 2023-12-25:
Hello,
I've been collecting a lot of data with Pose2Sim recently, and it works pretty well with proper calibration, but I've had a few problems collecting the arm movement data in the data:

I want to extract the coordinates of the arm from the json file to collect data, but I can't find the skeletal point arrangement order of the body25b on the network, I don't know which data should be extracted.

Could you please tell me what are the bone points of the arm in json files of the body25b? Thank you for your answer.

davidpagnon, 2023-12-26:
Hi, here are the points for body_25b: 
[image]

You can also find it [in the skeleton.py file](https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/skeletons.py), or at the end of any Config.toml file.

Does it answer the question?

smwscth, 2023-12-27:
Thank you! That would certainly solve the problem

closed 2023-12-27 by davidpagnon