perfanalytics/pose2sim #60: Latest release is broken?

UdolfSeelenfrost, 2024-01-04:
I just pulled the repository and upgraded Pose2Sim to the latest version 0.5.3. Unfortunately this new version isn't working for me. When trying to do the tutorial (Demonstration Part-1) i get the following error:
[image]

I thought i may be because of the recently added limb swapping, but when I disable it in config.toml I just get another error:
[image]

I get the same behavior with data that has previously worked. I will just work with an older version until this is fixed.

davidpagnon, 2024-01-04:
Sorry yes, it is broken due to the last push I did this afternoon in order to better handle lens distortions, I should have used the draft branch but I got confused. I'll fix it in the evening!

UdolfSeelenfrost, 2024-01-04:
Nice, thank you. The new folder structure looks good, I want to use it.

davidpagnon, 2024-01-08:
Hi there, it should be fixed but please leave the parameter `undistort_points` to false in Config.toml, I still have a few issues when handling distortions.

UdolfSeelenfrost, 2024-01-09:
Haven't tested it but I trust you

closed 2024-01-09 by UdolfSeelenfrost