perfanalytics/pose2sim #65: Compatibility issue with the latest version of lxml

fabiocat93, 2024-01-26:
### Issue Description
I have encountered a compatibility issue with the latest version of lxml (v5.1.0). The current codebase of this repository seems to only work with lxml v4.9.3, which was released in July 2023.

### Suggested Solutions
1. **Code Update:** It would be beneficial if the code could be updated to support lxml v5.1.0. This update will ensure compatibility with the latest features and security patches of lxml.

2. **Specify Dependency Version in `setup.py`:** If updating the code isn't immediately feasible, I suggest specifying the exact version of lxml required (`lxml==4.9.3`) in the `setup.py` file. This would prevent installation issues for new users who might unknowingly install the latest version of lxml.

I hope this issue can be addressed soon. Thank you for your attention to this matter!

Best,
@fabiocat93

davidpagnon, 2024-01-26:
Hi @fabiocat93,

Thank you for pointing this out. Can you tell me a bit more specifically what does not work, along with the error message? I just upgraded lxml to the 5.1.0 version and ran most Pose2Sim functions, and it seems to run without any issues.

fabiocat93, 2024-01-26:
Sure. I should specify that I am using a MacBook Pro M2 with macOS Sonoma 14.3. Here is the error message is see:

```
Traceback (most recent call last):
  File "calibration.py", line 15, in <module>
    calibrate(workspace, config['calibration_configs'])
  File "/motor_behaviors_analysis/data/utility/calibration.py", line 51, in calibrate
    Pose2Sim.calibration(subproject_config_dict)
  File "/mba/lib/python3.8/site-packages/Pose2Sim/Pose2Sim.py", line 185, in calibration
    from Pose2Sim.calibration import calibrate_cams_all
  File "/mba/lib/python3.8/site-packages/Pose2Sim/calibration.py", line 45, in <module>
    from lxml import etree
ImportError: dlopen(/mba/lib/python3.8/site-packages/lxml/etree.cpython-38-darwin.so, 0x0002): symbol not found in flat namespace '_exsltDateXpathCtxtRegister'
```

And by installing `lxml==4.9.3`, I see no error.

davidpagnon, 2024-01-26:
It's a bit weird, [I test the package](https://github.com/perfanalytics/pose2sim/blob/main/.github/workflows/continuous-integration.yml) on all platforms including MacOS, and I did not get any error...

Could you try to upgrade lxml again with `pip install lxml -U` and see if it wildly solves the problem?
If it is not solved, try then `pip install --no-binary :all: lxml`

You may find good ideas on this thread: https://stackoverflow.com/questions/35006614/what-does-symbol-not-found-expected-in-flat-namespace-actually-mean

Keep me updated!

fabiocat93, 2024-01-29:
thank you @davidpagnon  for your hints. it worked out with `pip install --no-binary :all: lxm`!

closed 2024-01-29 by fabiocat93