perfanalytics/pose2sim #66: Session_dir should not always be a default behavior

ANaaim, 2024-01-31:
Hey 

We tested the new version. I may have a small comment one comment on the following code : 
For session_dir they should not be a default behavior but the code should look if there is the information in the config file and if it is not in there use the default behavior

```python 
def track_2d_all(config):
....
    # Read config
    project_dir = config.get('project').get('project_dir')
    session_dir = os.path.realpath(os.path.join(project_dir, '..', '..'))
    pose_model = config.get('pose').get('pose_model')
    tracked_keypoint = config.get('personAssociation').get('tracked_keypoint')
    frame_range = config.get('project').get('frame_range')
    undistort_points = config.get('triangulation').get('undistort_points')

```

if you are ok i can do a PR looking like that (i am not sure of the syntax but it give you the idea)

```python 
def track_2d_all(config):
....
    # Read config
    project_dir = config.get('project').get('project_dir')
    if config.get('project').get('session_dir') exist:
          session_dir = config.get('project').get('project_dir')
    else:
          session_dir = os.path.realpath(os.path.join(project_dir, '..', '..'))
    pose_model = config.get('pose').get('pose_model')
    tracked_keypoint = config.get('personAssociation').get('tracked_keypoint')
    frame_range = config.get('project').get('frame_range')
    undistort_points = config.get('triangulation').get('undistort_points')

```

Thanks :)

ANaaim, 2024-01-31:
maybe not relevant

davidpagnon, 2024-03-05:
Related: Batch processing is nice, but creating the folder architecture every time is a pain: it should work in non-batch mode (with a root folder rather than a session, participant, and trial folder)

davidpagnon, 2024-04-15:
This has been addressed in the last release: One can create a Session->Participant->Trial folder structure for batch processing, or just have a Config.toml file along with a pose and a calibration folder.

closed 2024-04-15 by davidpagnon