perfanalytics/pose2sim #67: Calibration - Rotation

gokcergun, 2024-02-04:
Hi, 

I have extended the code by writing a function to convert CalibIo calibration files. 
In the end config file, you are writing the rotations as rotation vector (rvec), is it true? 

Thank you.

davidpagnon, 2024-02-04:
Hi, 

It could be useful indeed! Once you have integrated it in the code, do you think you could make a pull request so that it will be available for other users?
I am writing the rotations as rotation vectors indeed, via the OpenCV function [cv2.Rodrgigues](https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga61585db663d9da06b68e70cfbf6a1eac).

If you want to make sure your calibration is correct, you can use [the visualization tool I made for Blender](https://github.com/davidpagnon/Pose2Sim_Blender). The tool is ready and should be relatively intuitive to use, I just haven't had time to update the documentation with the new functions yet. I'll try to finish it tomorrow.

gokcergun, 2024-02-04:
Thank you! Yeah sure, I will send you a merge request after I check its accuracy. Blender Tool seems cool.

closed 2024-02-04 by gokcergun