perfanalytics/pose2sim #68: Question related to Static Pose

gokcergun, 2024-02-04:
I am wondering if the Sim2 system would be compatible with sitting postures. I have an old recording captured with 12 cameras, featuring a sitting posture. I do not have any T-Pose recordings, but I can identify a frame that serves as a T-Pose while in a sitting posture. Would this work? Or should the static pose be standing?

Thank you.

davidpagnon, 2024-02-04:
It should not be a person blem as long as keypoints are correctly detected, the person just needs to be static for a few frames. I've actually done the scaling on one frame with a moving person, it's not ideal but results don't look bad.

gokcergun, 2024-02-04:
Great, thank you. Yes, I thought the same, one frame can be replicated.

closed 2024-02-04 by gokcergun