perfanalytics/pose2sim #70: [Extrinsic Calibration] Broadcast Error

mattorga, 2024-02-19:
Good day!

I have been trying to calibrate my cameras, and the intrinsic calibration works perfectly well after a few custom configurations. However, when I finish identifying the points in the calibration of the extrinsics (scene), I get an error `ValueError: operands could not be broadcast together with shapes (8,1,2) (0,)`.  But looking at the errors, I can't trace where the dimensionality error comes from. 

Full error:
[image]

Calibration configuration:
[image]

davidpagnon, 2024-02-19:
Hi Matt, 
The error says it all: you probably did not click any points on the image.
``` 
'No points clicked (or fewer than 6). Press 'C' when the image is displayed, and then click on the image points corresponding to the 'object_coords_3d' you measured and wrote down in the Config.toml file.'
```

You first need to manually measure the 3D coordinates of 10 or more points in the scene (tiles, lines on the wall, boxes, treadmill dimensions, etc). These points should be as spread out as possible. Replace `object_coords_3d` by these coordinates in Config.toml.
Then film the scene, proceed to calibration, and type `C` to click the points on the image. Just follow the instructions written on the image. Try it first with the Demo to understand how it works.

More information on the Readme: https://github.com/perfanalytics/pose2sim?tab=readme-ov-file#calculate-from-scratch

Keep me updated if you manage to make it work!

davidpagnon, 2024-02-27:
Closed as stale. 
Please feel free to reopen the issue if needed!

closed 2024-02-27 by davidpagnon