perfanalytics/pose2sim #71: Can I import output of Sports2D into pose2sim and then into OpenSim

gabriead, 2024-02-25:
Hey guys,
I am pretty new to this domain so I need some help to get going. Can I use the output of Sports2D input that into pose2sim and then get an OpenSim model? What output data fomr Sports2D would I have to use (and in what format) and how would the workflow look like to use this in pose2sim? Thank's a lot in advance!

davidpagnon, 2024-02-26:
Hi, why do you want to use the output of Sports2D?

- if it's for the multiperson analysis, I plan to release that feature for Pose2Sim by the end of this week
- if it's for the easier interface, there is a university that is working on it, but it will probably bably not be available before a few months
- if it is because you can use only one camera, this could be possible with an OpenSim 2D model, but since Sports2D was meant to be a minimal solution, I did not implement it. This would also be very buggy if the person were not perfectly in the frontal or sagittal plane, too.

In any case, Sports2D outputs json files which can be used in Pose2Sim, but you also need several cameras and a calibration file.
The CSV files for positions can be turned to trc files. This is easy (open the CSV file and a tec file by file, just the header changes) but I haven't done it.

I hope it answered your question!

gabriead, 2024-02-26:
Hi @davidpagnon thank's for your response! Exactly I only have one camera therefore it will be third use case you describe. Is there a model/workflow out there to transform Sports2D 2D data into 3D data to get an OpenSim 3D model? The authros of this paper (https://www.frontiersin.org/articles/10.3389/fspor.2022.994221/full) seem to have done something similar.
If not I would be happy to start with an Open Sim 2D model. What would I have to do to implement that? Can we collaborate on that, would love to make such a solution available in your package for everyone then!

davidpagnon, 2024-02-26:
I did not know about this paper, thanks!

The main difference is that they used a Kinect, which has a depth sensor that can retrieve 3D. Some alternative approaches can lift 3D from a single video (BlazePose being the easiest to use). I've been thinking of implementing it in Pose2Sim, but I could not find time for it. Also, I highly suspect it would not be very accurate. 

But I don't know, it may be worth trying! It would still be more accurate than fitting an OpenSim model just on 2D points from Sports2D (although Sports2D supports multi-person detection, unlike BlazePose that you would use otherwise). In both cases, you could have good angles and timing, but positions will be in pixels. To get them in meters, you would need to calibrate the camera.

The workflow would go this way:
- **BlazePose detection:** The script [I have here](https://github.com/perfanalytics/pose2sim/blob/5b402111ecabe70fd6e009790f8b0e597132850b/Pose2Sim/Utilities/Blazepose_runsave.py#L178) does not take the depth information into account. Retrieving `p.z*W` would let you obtain depth ([More info on their page](https://developers.google.com/mediapipe/solutions/vision/pose_landmarker/python)). However, keep in mind that (1) depth is approximate, and (2) BlazePose takes the mid-hip point as the depth origin: thus you could not retrieve the CoM displacement, unless you implement a conversion of the frame of reference. 
- Then you would need to convert the output to a trc file. If you only need timings and angles and the person does not go further or closer to the camera, this could be enough. Filtering, scaling, and IK would be unchanged.

You could scale the pixel coordinates to obtain meter coordinates, by letting the user select two points on the image and specifying their relative distance. This would only work if the person stays in the frontal or sagittal plane though. Just imagine that you face the camera, and that your shoulder width is 100 px in the image, and 50 cm in reality. You could infer that 1 m <-> 200 px. However, if you move sideways, your shoulder width would be like 10 px on the image, and still 50 cm in reality.

All of it is pretty doable, but unfortunately, I am not sure when or if I'll find time for it. If you feel like doing it, feel absolutely free though, I'll be happy to review and accept pull requests! Also, just in case, [here is the discord server](https://discord.gg/Ye9jmBzBEF) where other users participate in the development of Pose2Sim, it is easier than discussing via Github issues.

gabriead, 2024-02-26:
Thank's for the invitation, happy to discuss the topic further there!

davidpagnon, 2024-02-26:
The other approach would use this model ([2D_model.zip](https://github.com/perfanalytics/pose2sim/files/14405208/2D_model.zip)), with the OpenPose MarkerSet (see [there](https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/OpenSim_Setup/Model_Pose2Sim_Body25.osim)). 
This is actually a 3D model, but it has limited degrees of freedom (only in the sagittal plane). 

With this approach you could convert the json or csv coordinates to a trc file, with depth being set to zero.
Scaling would be done only in X and Y direction, not Z. And IK would be mostly unchanged.

Same method (and same limitations) as the other approach if you want meter coordinates.

davidpagnon, 2024-02-27:
Sorry I still haven't read the article in depth but I just see that they did not use the depth sensor of their Kinect. So they used an approach similar to BlazePose to estimate depth.

Note that they set themselves up in a pretty simplified situation, where the person stays in the frontal plane, and does not move forwards, backwards, nor to the side. This removes most of the constraints I was telling you about.

gabriead, 2024-02-27:
No worries! I have contacted one of the authors to get more information on the exact details of their approach. The thing I am still struggling the most currently is to understand how I transform the data into a format that can be used in OpenSim. So currently with the RealSense I can obtain the following data (small example) : [ left_ellbow[x_coord, y_coord, z_coord], right_ellbow[.....,]]. Is that data sufficient to use your transformation scripts to get a first OpenSim model? As soon as I understand how to map the RealSense data into OpenSim I can start to understand more details and follow you up on the constraints.

davidpagnon, 2024-02-27:
You will need to convert it yourself, but it should not be too complicated: just have a look at the trc file resulting from the Demo, that should be a bit clearer.

davidpagnon, 2024-03-12:
This is issue has been closed as stale. Feel free to reopen it if needed!

closed 2024-03-12 by davidpagnon