perfanalytics/pose2sim #74: Large number of Json are annoying to copy. Would it be possible to imagine that P2S would accept an other file format such as hdf5 or any other ?

ANaaim, 2024-02-29:
I know that json is the base export format file for open pose but when you are manipulating large amount of individual frame in a video and each is file is in a different json file it make folder with large number of file making it quite "difficult" ( I means extremely long) to copy. How could it be handle in a elegant manner ? We might concatenate all the json in a single file ? Any thought on this issue ?

ANaaim, 2024-02-29:
I saw that it was in your to do list at the end of the read me. I might try to have a look.

davidpagnon, 2024-02-29:
I totally agree, it would be nice if Pose2Sim could natively accept both json and h5 format (such a the DeepLabCut one).
You may have a look at this piece of code: https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/Utilities/Blazepose_runsave.py#L51