perfanalytics/pose2sim #79: Bug when using P2S 0.7 with marker augmentation

ANaaim, 2024-03-12:
When using the marker augmentation with P2S v0.7.1 i have the following error message : 

```
Traceback (most recent call last):
  File "C:/Program Files/JetBrains/PyCharm 2023.1.4/plugins/python/helpers/pydev/pydevd.py", line 1534, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm 2023.1.4\plugins\python\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "LBMC_marker_less\full_test_pipeline.py", line 109, in <module>
    triangulation_data(final_data_path)
  File "LBMC_marker_less\triangulation_data.py", line 118, in triangulation_data
    Pose2Sim.markerAugmentation(config_dict)
  File "\Miniconda3\envs\Pose2Sim0_70\lib\site-packages\Pose2Sim\Pose2Sim.py", line 482, in markerAugmentation
    augmentTRC(config_dict)
  File "\Miniconda3\envs\Pose2Sim0_70\lib\site-packages\Pose2Sim\markerAugmentation.py", line 195, in augmentTRC
    outputs = np.reshape(outputs, (outputs.shape[1], outputs.shape[2]))
  File "<__array_function__ internals>", line 200, in reshape
  File "Miniconda3\envs\Pose2Sim0_70\lib\site-packages\numpy\core\fromnumeric.py", line 298, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "Miniconda3\envs\Pose2Sim0_70\lib\site-packages\numpy\core\fromnumeric.py", line 54, in _wrapfunc
    return _wrapit(obj, method, *args, **kwds)
  File "\Miniconda3\envs\Pose2Sim0_70\lib\site-packages\numpy\core\fromnumeric.py", line 43, in _wrapit
    result = getattr(asarray(obj), method)(*args, **kwds)
  File "\Miniconda3\envs\Pose2Sim0_70\lib\site-packages\tensorflow\python\framework\ops.py", line 924, in __array__
    raise NotImplementedError(
NotImplementedError: Cannot convert a symbolic tf.Tensor (sequential_2/time_distributed_2/Reshape_1:0) to a numpy array. This error may indicate that you're trying to pass a Tensor to a NumPy call, which is not supported.
python-BaseException
```

Did you encounter such error ? or any idea where it would come from ==> I am using directly your code for this part

==> After some investigation it seems that this is due to the fact  that outputs cannot be converted into a numpy... I am still watching

ANaaim, 2024-03-12:
In V0.55:
```py
        json_file = open(os.path.join(augmenterModelDir, "model.json"), 'r')
        pretrainedModel_json = json_file.read()
        json_file.close()
        model = tf.keras.models.model_from_json(pretrainedModel_json)
        model.load_weights(os.path.join(augmenterModelDir, "weights.h5"))
        outputs = model.predict(inputs)
        keras.backend.clear_session()
```
outputs is an nd array
but in V0.7 outputs is a tensorflow object

ANaaim, 2024-03-12:
Why when using subject hieght you are using the following code : 

```py
            # Step 3: Normalize with subject's height.
            norm2_trc_data_data = copy.deepcopy(norm_trc_data_data)
            norm2_trc_data_data = norm2_trc_data_data / subject_height[p]
```
There is normally only one height per subject ?

ANaaim, 2024-03-12:
Linked to PR #80

davidpagnon, 2024-03-12:
Thanks, I see the issue!
Since I wanted Pose2Sim to support multi-person analysis, I made `participant_height` accept lists of floats. This is why I added these lines: 
```
    if not type(subject_height) == list:
        subject_height = [subject_height]
        subject_mass = [subject_mass]
```

There was a reason for doing `outputs = model(inputs)` instead of `outputs = model.predict(inputs)` but I can't remember it. Let's revert it like you did and see if it works again.

ANaaim, 2024-03-13:
In that case, I should investigate more ==> it means that in my case it was using the indices 1 which means that it considered that they were two subject when they were only one so not the desired behavior. 

What was your idea for detecting that they were multiple subject ?

ANaaim, 2024-03-13:
I suppose that's the way it is done : 
```py
trc_files = [f for f in glob.glob(os.path.join(pathInputTRCFile, '*.trc')) if 'filt' in f and '_LSTM' not in f]
```
Some thought : 
1) Does not work if data were not filetered (which could be wanted).
2) We could add a test after to be sure that if there is multiple file that should be augmented that there is the number of height parameters that corresponds 
```py
if len(trc_files)!=len( subject_height) or len(trc_files)!=len( subject_mass)
    raise ValueError('The number of height and mass parameters given does not correspond to the number of participants found'')
```
3. Maybe add an option in marker augmentation to the number of subject that should be analysed. 

I tried on my code and i do not manage to reproduce the bug where I add a second indices...

davidpagnon, 2024-03-19:
1. I must say I was unsure of myself when excluding the unfiltered data from marker augmentation, I just did not want to do marker augmentation on both filtered and unfiltered data. I think your idea of simply adding a parameter in the Config.toml file to specify whether marker augmentation should be done on filtered or unfiltered data is good!
2. Checking that the number of height and weight parameters is a good idea indeed.
3. Do you think with point 1. we still need to add an option to specify the number of subjects to be analyzed?

If you already did all that, do you want to just submit a pull request?

The names of the files with multiple subjects look like that:
`S00_P01_T02_Participants1-2_P1_0-100_filt_butterworth_LSTM.trc` and `S00_P01_T02_Participants1-2_P2_0-100_filt_butterworth_LSTM.trc`
I agree it is overkill, but basically it is <trial folder name> + <ID participant> + <frame_range> + <filter and LSTM if relevant> .trc

ANaaim, 2024-03-22:
I will do it. 

For the marker augmentation on multiple person maybe we should use the information from other fields... 

I will try some stuff and do the PR

ANaaim, 2024-03-27:
I might have found a way to maintain both beahvior : 

```py
    all_trc_files = [f for f in glob.glob(os.path.join(pathInputTRCFile, '*.trc')) if '_LSTM' not in f]
    trc_no_filtering = [f for f in glob.glob(os.path.join(pathInputTRCFile, '*.trc')) if '_LSTM' not in f and 'filt' not in f]
    trc_filtering = [f for f in glob.glob(os.path.join(pathInputTRCFile, '*.trc')) if '_LSTM' not in f and 'filt' in f]
    
    if len(all_trc_files) == 0:
        raise ValueError('No trc files found.')
    if len(trc_filtering)>0:
       trc_files = trc_filtering
    else:
        trc_files = trc_no_filtering
```
If there is no trc file ==> error , if there is at least one filtered file we consider that we should use the filetered file else we use the non filered one ? 

Any though ?.

davidpagnon, 2024-03-30:
Okay, that's a good idea, thanks!

ANaaim, 2024-04-02:
PR done