perfanalytics/pose2sim #84: More detail information using COCO133.osim and scaling and IK

maurice1128, 2024-03-30:
Hi, I'm using COCO133 whole body and try to integrate into opensim
but the marker in Model_Pose2Sim_Coco133.osim on hand has only few marker, and the IK value is not enough

is there a more comprehensive version of 
Model_Pose2Sim_Coco133.osim,
Scaling_Setup_Pose2Sim_Coco133.xml,
IK_Setup_Pose2Sim_Coco133.xml

especially on hand analysis

thankyou so much

davidpagnon, 2024-03-30:
Hi, 

The issue with the OpenSim models is that they are usually highly specialized and focusing on a special kind of movement: for example, you have OpenSim models that only analyze the neck joints while a jockey falls from a horse.

What is possible though, is to combine the default .osim model with a hand model. I do not have time to do it now, but if you do it I will be happy to add it to Pose2Sim!

You have already pretty well understood what needs to be done: 
- Add the hand points on Coco133 in skeletons.py
- Edit the Scaling and IK setup files so that they take these points into account
- Combine the default OpenSim model with a hand model, and add hand markers to it

Best regards

maurice1128, 2024-03-31:
okok, no problem, I'll try my best.
I look up at the internet, and the combine of opensim model  is using the opensim creator right? or is there any other recommendation?
because I've tried the GUI of Opensim only, and it doesn't seems to have the combination function inside
Best regards

davidpagnon, 2024-03-31:
Well you can do it different ways, including with the OpenSim or Matlab API, but the way I did it was by copy pasting lines in .osim model files.

It can be a bit tricky, but basically you need to copy the bodies of a hand model into the default Pose2Sim model. You also need to copy-paste the joints, and make sure they parent to the right thing in the default model. The main challenge maybe to make the wrist joint compatible across both models.

And then you need to add markers in the right position that correspond to the hand keypoints, but this can be done from the GUI.

davidpagnon, 2024-03-31:
OpenSim creator may be a good way to do it but I've never tried it. If you find it useful, please tell me, I'd be interested!

davidpagnon, 2024-04-12:
This issue has been closed because it has not had any activity in two weeks. Feel free to reopen it if needed!

closed 2024-04-12 by davidpagnon