perfanalytics/pose2sim #87: Prb in calibratioin

ANaaim, 2024-04-05:
Hello, 

I think there is an error in calibration.py at line 691

Why in extrinsic method  we have 
```python
 imgp, objp_not_used = findCorners(img_vid_files[0], extrinsics_corners_nb, objp=object_coords_3d, show=show_reprojection_error)
```
and not 
```python
 imgp, objp = findCorners(img_vid_files[0], extrinsics_corners_nb, objp=object_coords_3d, show=show_reprojection_error)
```
obj_not_used should be objp no ? 

```python

            # Find corners or label by hand
            if extrinsics_method == 'board':
                imgp, objp_not_used = findCorners(img_vid_files[0], extrinsics_corners_nb, objp=object_coords_3d, show=show_reprojection_error)
                if imgp == []:
                    logging.exception('No corners found. Set "show_detection_extrinsics" to true to click corners by hand, or change extrinsic_board_type to "scene"')
                    raise ValueError('No corners found. Set "show_detection_extrinsics" to true to click corners by hand, or change extrinsic_board_type to "scene"')

            elif extrinsics_method == 'scene':
                imgp, objp = imgp_objp_visualizer_clicker(img, imgp=[], objp=object_coords_3d, img_path=img_vid_files[0])
                if imgp == []:
                    logging.exception('No points clicked (or fewer than 6). Press \'C\' when the image is displayed, and then click on the image points corresponding to the \'object_coords_3d\' you measured and wrote down in the Config.toml file.')
                    raise ValueError('No points clicked (or fewer than 6). Press \'C\' when the image is displayed, and then click on the image points corresponding to the \'object_coords_3d\' you measured and wrote down in the Config.toml file.')
                if len(objp) < 10:
                    logging.info(f'Only {len(objp)} reference points for camera {cam}. Calibration of extrinsic parameters may not be accurate with fewer than 10 reference points, as spread out as possible.')
            
            elif extrinsics_method == 'keypoints':
                logging.info('Calibration based on keypoints is not available yet.')
            
            # Calculate extrinsics
            mtx, dist = np.array(K[i]), np.array(D[i])
            _, r, t = cv2.solvePnP(np.array(objp), imgp, mtx, dist)
            r, t = r.flatten(), t.flatten()

```

davidpagnon, 2024-04-08:
Hi, I haven't tested it but it reasonably seems like you are right! Are you using a checkerboard to calibrate the extrinsic parameters? I do not generally recommend this as I feel like you would get suboptimal results, but since I have not tested it often, it may be a bug that I have missed.
I supposed you have tried with objp instead and that it works better?

As usual, thank you for testing the code and spotting mistakes for me :)

ANaaim, 2024-04-08:
Yep I tried and it worked so I will do a PR if needed.

davidpagnon, 2024-04-08:
Sure, that would be great!

ANaaim, 2024-04-09:
linked to #91

ANaaim, 2024-04-09:
Can be closed as PR was merged.

closed 2024-04-09 by ANaaim