perfanalytics/pose2sim #88: Error when running triangulation

hmejbri, 2024-04-08:
Why do I get this error 

`Project directory: /home/ubuntu/S00_P00_Participant/S00_P00_T00_StaticTrial
An error occurred: list indices must be integers or slices, not str`

when running `Pose2Sim.triangulation()`

davidpagnon, 2024-04-08:
Could you give me more context? Are you running the Demo or your own data? Did you edit your Config.toml file?
What is the full error message?

hmejbri, 2024-04-08:
I got pose videos and calib_checkerboard.toml file from another project and I generated the static and pose keypoints of the videos using openpose, I skipped the calibration and executed triangulation directly and I got that error. In the config.toml file i only changed calibration_type to 'calculate'

davidpagnon, 2024-04-08:
If you don't run calibration anymore, it should not make a difference whether you set calibration type to calculate or anything else.
Could you send me the full error message so that I can spot exactly where the code fails to execute?

hmejbri, 2024-04-08:
That's the full message

---------------------------------------------------------------------
Triangulation of 2D points for S00_P00_T00_StaticTrial, for all frames.
On Monday 08. April 2024, 12:12:20
---------------------------------------------------------------------

Project directory: /home/ubuntu/S00_P00_Participant/S00_P00_T00_StaticTrial
An error occurred: list indices must be integers or slices, not str

davidpagnon, 2024-04-08:
Oh really, is there nothing kind of like that?

-------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-1-34186a0c8775> in <module>
      3 choice = input('Which list index would you like to pick (0, 1, or 2)? ')
      4 
----> 5 print(favourite_three_franchises[choice])

TypeError: list indices must be integers or slices, not str

hmejbri, 2024-04-08:
Turns out I was using a try except that's why the full error was not showing.

`Traceback (most recent call last):
  File "/home/ubuntu/pfe_mocap/main.py", line 53, in <module>
    asyncio.run(main())
  File "/opt/conda/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/opt/conda/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/ubuntu/pfe_mocap/main.py", line 41, in main
    await pose_utils.triangulation()
  File "/home/ubuntu/pfe_mocap/Scripts/Pose_analysis/pose_utils.py", line 12, in triangulation
    Pose2Sim.triangulation()
  File "/opt/conda/lib/python3.10/site-packages/Pose2Sim/Pose2Sim.py", line 394, in triangulation
    triangulate_all(config_dict)
  File "/opt/conda/lib/python3.10/site-packages/Pose2Sim/triangulation.py", line 660, in triangulate_all
    nb_persons_to_detect = max([len(json.load(open(json_fname))['people']) for json_fname in json_tracked_files[0]])
  File "/opt/conda/lib/python3.10/site-packages/Pose2Sim/triangulation.py", line 660, in <listcomp>
    nb_persons_to_detect = max([len(json.load(open(json_fname))['people']) for json_fname in json_tracked_files[0]])
TypeError: list indices must be integers or slices, not str
`

davidpagnon, 2024-04-08:
This is strange, I don't see where you would get a string from this line...
Could you edit the line in question, add beforehand:

```python
print(repr([len(json.load(open(json_fname))['people']) for json_fname in json_tracked_files[0]]))
```

Then exit command prompt and run it again so that I can see the output?

hmejbri, 2024-04-08:
Traceback (most recent call last):
  File "/home/ubuntu/pfe_mocap/main.py", line 53, in <module> asyncio.run(main())
  File "/opt/conda/lib/python3.10/asyncio/runners.py", line 44, in run return loop.run_until_complete(main)
  File "/opt/conda/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete return future.result()
  File "/home/ubuntu/pfe_mocap/main.py", line 41, in main await pose_utils.triangulation()
  File "/home/ubuntu/pfe_mocap/Scripts/Pose_analysis/pose_utils.py", line 12, in triangulation Pose2Sim.triangulation()
  File "/opt/conda/lib/python3.10/site-packages/Pose2Sim/Pose2Sim.py", line 394, in triangulation triangulate_all(config_dict)
  File "/opt/conda/lib/python3.10/site-packages/Pose2Sim/triangulation.py", line 661, in triangulate_all print(repr(json.load(open(json_fname))['people']))

TypeError: list indices must be integers or slices, not str

davidpagnon, 2024-04-08:
There must be a problem in your json files. Can you print(json_tracked_files)
And open them with an editor to see if there is anything wrong with them?

hmejbri, 2024-04-08:
This is what the print shows 

[['./S00_P00_Participant/S00_P00_T00_StaticTrial/pose/cam_1_json/cam1.0.json', './S00_P00_Participant/S00_P00_T00_StaticTrial/pose/cam_1_json/cam1.1.json', './S00_P00_Participant/S00_P00_T00_StaticTrial/pose/cam_1_json/cam1.2.json', './S00_P00_Participant/S00_P00_T00_StaticTrial/pose/cam_1_json/cam1.3.json', ... ]]

And this is the content of one of the files 

[[[269.99053955078125, 387.82830810546875, 0.8961206674575806], [267.3406066894531, 429.7220458984375, 0.9161862134933472], [235.9087677001953, 432.2477111816406, 0.8875619173049927], [223.00326538085938, 495.0760192871094, 0.8772337436676025], ... ]]

Does this seem right ?

davidpagnon, 2024-04-09:
Okay, this is where the problem lies: how did you obtain your json files? They should look like that:

```json
{   "version":1.3,
    "people": [
        {
            "face_keypoints_2d": [
            ],
            "face_keypoints_3d": [
            ],
            "hand_left_keypoints_2d": [
            ],
            "hand_left_keypoints_3d": [
            ],
            "hand_right_keypoints_2d": [
            ],
            "hand_right_keypoints_3d": [
            ],
            "person_id": [
                -1
            ],
            "pose_keypoints_2d": [
                530.177,
                274.224,
                0.935098,
                533.18,
                268.365,
                0.871889,
                524.347,
                268.33,
                0.927471,
                0,
                0,
                0,
                497.886,
                265.285,
                0.84834,
                544.978,
                306.629,
                0.835899,
                459.615,
                309.575,
                0.872913,
                553.871,
                362.398,
                0.874023,
                433.285,
                365.611,
                0.924198,
                553.889,
                397.82,
                0.857626,
                486.094,
                385.952,
                0.906055,
                530.168,
                433.271,
                0.899863,
                477.319,
                436.043,
                0.82925,
                553.825,
                524.455,
                0.928564,
                477.352,
                527.369,
                0.934174,
                565.631,
                621.387,
                0.860776,
                462.635,
                609.772,
                0.826998,
                503.745,
                294.949,
                0.88991,
                527.267,
                224.125,
                0.922588,
                580.346,
                645.022,
                0.827753,
                594.997,
                639.118,
                0.813313,
                553.912,
                624.559,
                0.798435,
                474.338,
                627.385,
                0.696133,
                456.737,
                624.583,
                0.761559,
                468.435,
                618.6,
                0.672332
            ],
            "pose_keypoints_3d": [
            ]
        },
        {
            "face_keypoints_2d": [
            ],
            "face_keypoints_3d": [
            ],
            "hand_left_keypoints_2d": [
            ],
            "hand_left_keypoints_3d": [
            ],
            "hand_right_keypoints_2d": [
            ],
            "hand_right_keypoints_3d": [
            ],
            "person_id": [
                -1
            ],
            "pose_keypoints_2d": [
                336.022,
                250.714,
                0.883262,
                339.053,
                247.771,
                0.924258,
                ...],
            "pose_keypoints_3d": [
            ]
        },
}
```

hmejbri, 2024-04-09:
This is my script : 

`# generate json files containing the keypoints of each camera (static and pose)
async def create_keypoint_files(op, videos_path, keypoints_path):
    cam_num = 1

    for video in videos_path:
        # Flags
        parser = argparse.ArgumentParser()
        parser.add_argument("--video", default=video)
        parser.add_argument("--model_pose", default="BODY_25B")

        args = parser.parse_known_args()

        # Custom Params (refer to include/openpose/flags.hpp for more parameters)
        params = dict()
        params["model_folder"] = "/home/ubuntu/dw/openpose/models/"

        # Add others in path?
        for i in range(0, len(args[1])):
            curr_item = args[1][i]
            if i != len(args[1])-1: next_item = args[1][i+1]
            else: next_item = "1"
            if "--" in curr_item and "--" in next_item:
                key = curr_item.replace('-','')
                if key not in params:  params[key] = "1"
            elif "--" in curr_item and "--" not in next_item:
                key = curr_item.replace('-','')
                if key not in params: params[key] = next_item

        # Starting OpenPose
        opWrapper = op.WrapperPython()
        opWrapper.configure(params)
        opWrapper.start()

        # Process Video
        cap = cv2.VideoCapture(args[0].video)

        frame_num = 0

        while cap.isOpened():
            ret, frame = cap.read()
            if not ret:
                break

            datum = op.Datum()
            datum.cvInputData = frame
            opWrapper.emplaceAndPop(op.VectorDatum([datum]))

            # Display keypoints on the frame
            cv2.imshow(video, datum.cvOutputData)

            # Save keypoints to JSON file
            output_json_file = f"{keypoints_path}/cam_{cam_num}_json/cam{cam_num}.{frame_num}.json"
            with open(output_json_file, 'w') as f:
                if datum.poseKeypoints is not None:
                    json.dump(datum.poseKeypoints.tolist(), f)  # Convert keypoints to list before saving as JSON

            if cv2.waitKey(1) & 0xFF == ord('q'):
                break

            frame_num += 1

        cap.release()
        cv2.destroyAllWindows()
        
        cam_num += 1`

davidpagnon, 2024-04-09:
I guess you are using the OpenPose python API?
I have actually never tried it, but I'm surprised the output is different from the one from the OpenPoseDemo.exe one.

For it to work with Pose2Sim, you need to have json files write as above. Keep me updated!

davidpagnon, 2024-06-09:
This issue has been marked as stale. Feel free to reopen it or open a new one if needed.

closed 2024-06-09 by davidpagnon