perfanalytics/pose2sim #89: Should have to click one more time when doing calibration by pointing on an images

ANaaim, 2024-04-08:
Hi, 

When we are doing the extrinsic calibration by clicking on a image when the last point is clicked the data is directly send back in the code. Some misclick can happen so the user should have to click one more time or any other method no ?

best regards

davidpagnon, 2024-04-08:
Maybe one more click would make sense! It would be nice if returning the data would have to be triggered, but could be done either with an additional click, or by pressing enter, escape, or Y.
Would you have time to implement it?

ANaaim, 2024-04-08:
Would probably look like something like this : 
```python 
                    # Question to check if user ok 
                    user_input = input("Are you satisfied with this point? (yes/no): ")
                    if user_input.lower() == "yes":
                        # retrieve objp_confirmed and plot 3D
                        objp_confirmed = [objp[count]]
                        ax_3d.scatter(*objp[count], marker='o', color='g')
                        fig_3d.canvas.draw()
                    elif user_input.lower() == "no":
                        # Remove lastpoint from image
                        new_xydata = scat.get_offsets()[:-1]
                        scat.set_offsets(new_xydata)
                        plt.draw()
                        # Remove last point from imgp_confirmed
                        imgp_confirmed = imgp_confirmed[:-1]
                        if len(objp) != 0:
                            if count >= 0:
                                count -= 1
                            # Remove last point from objp_confirmed
                            objp_confirmed = objp_confirmed[:-1]
                            # remove from plot 
                            if len(ax_3d.collections) > len(objp):
                                ax_3d.collections[-1].remove()
                                fig_3d.canvas.draw()
```

davidpagnon, 2024-04-08:
It's not annoying, it is just time consuming 😁

You write "are you satisfied with this keypoint", does it mean that there would need to be a confirmation after each keypoint, or only after the last one? The second option would probably be better.

Would it require the user to type in the console? I'm afraid that they won't think of looking there, unless there is a pop-up window appearing.

I wonder if just clicking once more or typing a key would not be more straightforward?

ANaaim, 2024-04-08:
yes i was fast for generating this 
I will use a tkinter like this : 
[image]

ANaaim, 2024-04-08:
And just for the last point.

davidpagnon, 2024-04-08:
That looks great 👍

ANaaim, 2024-04-08:
Some problem with the tkinter, during it you can still click on the image... not very good... I am looking for a solution.

davidpagnon, 2024-04-09:
It would be better if the image were locked, but even if you cannot find a solution this is already an improvement as is

ANaaim, 2024-04-10:
Found a way to lock the image. Using a Boolean to follow if we are in the prompt asking. If yes I skip any element done in the in_click function.

davidpagnon, 2024-04-10:
Great, I'll let you update your pull request when you have time!

ANaaim, 2024-04-12:
It is already in the PR. It was a additional comment.

davidpagnon, 2025-05-15:
Implemented in the latest release :)

closed 2025-05-15 by davidpagnon