perfanalytics/pose2sim #93: Extrinsics is not getting calculated

hmejbri, 2024-04-12:
When I try to perform calibration while I have the .toml file in the calibration folder I should only calculate extrinsics but instead I get the intrinsics frames.

I had to take a look inside the calibration.py and I noticed this block of code :

```
# calculate extrinsics
if calculate_extrinsics:
    logging.info(f'\nCalculating extrinsic parameters...')
    
    # check that the number of cameras is consistent
    nb_cams_extrinsics = len(next(os.walk(os.path.join(calib_dir, 'extrinsics')))[1])
    if nb_cams_intrinsics != nb_cams_extrinsics:
        raise Exception(f'Error: The number of cameras is not consistent:\
                Found {nb_cams_intrinsics} cameras based on the number of intrinsic folders or on calibration file data,\
                and {nb_cams_extrinsics} cameras based on the number of extrinsic folders.')

    ret, C, S, D, K, R, T = calibrate_intrinsics(calib_dir, intrinsics_config_dict)

# calculate extrinsics
if calculate_extrinsics:
    logging.info(f'\nCalculating extrinsic parameters...')
    ret, C, S, D, K, R, T = calibrate_extrinsics(calib_dir, extrinsics_config_dict, C, S, K, D)
```

and I don't know why the intrinsics are being calculated once more, Correct me if i'm wrong but I think it should be like this 

```
# calculate extrinsics
if calculate_extrinsics:
    logging.info(f'\nCalculating extrinsic parameters...')
    
    # check that the number of cameras is consistent
    nb_cams_extrinsics = len(next(os.walk(os.path.join(calib_dir, 'extrinsics')))[1])
    if nb_cams_intrinsics != nb_cams_extrinsics:
        raise Exception(f'Error: The number of cameras is not consistent:\
                Found {nb_cams_intrinsics} cameras based on the number of intrinsic folders or on calibration file data,\
                and {nb_cams_extrinsics} cameras based on the number of extrinsic folders.')

    logging.info(f'\nCalculating extrinsic parameters...')
    ret, C, S, D, K, R, T = calibrate_extrinsics(calib_dir, extrinsics_config_dict, C, S, K, D)
```

davidpagnon, 2024-04-12:
Hi, before I check that, are you sure that you did not set `overwrite_intrinsics = true` in your Config.toml?
It should be set to false in order not to recalculate intrinsic parameters.

hmejbri, 2024-04-12:
Yup overwrite_intrinsics is set to false

davidpagnon, 2024-04-12:
Yes you're right, I must have overwritten something by mistake... I'll fix this and push the change as well as a bunch of others tonight or during the week-end!

hmejbri, 2024-04-15:
I just tested the latest version of the code and its still causing problems. This time it loads the extrinsics frame but after labeling the corners of the first camera by hand I get this error : 

Traceback (most recent call last):
  File "/home/ubuntu/pfe_mocap/main.py", line 52, in <module>
    asyncio.run(main())
  File "/opt/conda/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/opt/conda/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/ubuntu/pfe_mocap/main.py", line 36, in main
    await calibration_utils.calibrate_cams()
  File "/home/ubuntu/pfe_mocap/Scripts/Calibration/calibration_utils.py", line 59, in calibrate_cams
    await Pose2Sim.calibration()
  File "/opt/conda/lib/python3.10/site-packages/Pose2Sim/Pose2Sim.py", line 207, in calibration
    calibrate_cams_all(config_dict)
  File "/opt/conda/lib/python3.10/site-packages/Pose2Sim/calibration.py", line 1344, in calibrate_cams_all
    ret, C, S, D, K, R, T = calib_fun(*args_calib_fun)
  File "/opt/conda/lib/python3.10/site-packages/Pose2Sim/calibration.py", line 526, in calib_calc_fun
    ret, C, S, D, K, R, T = calibrate_extrinsics(calib_dir, extrinsics_config_dict, C, S, K, D)
  File "/opt/conda/lib/python3.10/site-packages/Pose2Sim/calibration.py", line 688, in calibrate_extrinsics
    if imgp == []:
ValueError: operands could not be broadcast together with shapes (28,1,2) (0,)

hmejbri, 2024-04-15:
I replaced imgp == [] with len(imgp) == 0 and it works now

davidpagnon, 2024-04-15:
Hi, thanks!
Sorry I had a few merge issues but I'll make the fixes tonight.

closed 2024-06-09 by davidpagnon