perfanalytics/pose2sim #96: BodyKinematics

lishuaifu, 2024-04-18:
Hello, I saw that you have consulted about Opensim’s BodyKinematics on Opensim. I want to use BodyKinematics to obtain the position of the center of mass of the body part throughout the experiment. How should I achieve this in python?

davidpagnon, 2024-04-18:
Hi, this script might help you: https://github.com/perfanalytics/pose2sim/blob/main/Pose2Sim/Utilities/bodykin_from_mot_osim.py

This is actually what I use in [Pose2Sim_Blender](https://github.com/davidpagnon/Pose2Sim_Blender).

lishuaifu, 2024-04-18:
Thank you very much, this problem is solved perfectly.I have another question. I can only get the flexion and extension angles using this. I want to change the model to calculate rotation and addition, but I can only calculate the addition at the moment. I want to calculate the rotation, but I don’t know how to modify osim.

davidpagnon, 2024-04-19:
I'm not sure I understand so tell me if I answer off topic!

Using this script, you get all angles defined in your OpenSim model: if your shoulder has 3 degrees of freedom, you'll get flexion, adduction, and internal rotation. If your elbow has 1 single degree of freedom, you get only flexion.

Basically, it all depends on your OpenSim model. You can find and download different models depending on your needs, and you can also edit the osim files. But I can't explain it like that, I would recommend you to try [their tutorials[(https://opensimconfluence.atlassian.net/wiki/spaces/OpenSim/pages/53088695/Examples+and+Tutorials) to understand better (introductory examples are enough).

closed 2024-05-15 by davidpagnon