perfanalytics/pose2sim #98: list index out of range

chizhan9527, 2024-04-19:
Hello David,
    I am trying to use your Pose2Sim program. I read the README.MD and I'm doing triangulation. But I meet an error below.

```python
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[2], line 1
----> 1 Pose2Sim.triangulation()

File D:\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:398, in triangulation(config)
    395 logging.info("---------------------------------------------------------------------")
    396 logging.info(f"\nProject directory: {project_dir}")
--> 398 triangulate_all(config_dict)
    400 end = time.time()
    401 elapsed = end-start

File D:\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\triangulation.py:1531, in triangulate_all(config)
   1526     logging.info(
   1527         "\nThe trc and c3d files have been renamed to match the order of the static sequences."
   1528     )
   1530 # Recap message
-> 1531 recap_triangulate(
   1532     config,
   1533     error_tot,
   1534     nb_cams_excluded_tot,
   1535     keypoints_names,
   1536     cam_excluded_count,
   1537     interp_frames,
   1538     non_interp_frames,
   1539     trc_paths,
   1540 )

File D:\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\triangulation.py:406, in recap_triangulate(config, error, nb_cams_excluded, keypoints_names, cam_excluded_count, interp_frames, non_interp_frames, trc_path)
    404 print(f"cam_names :{cam_names}")
    405 print(f"cam_excluded_count :{cam_excluded_count}")
--> 406 cam_names = cam_names[list(cam_excluded_count[0].keys())]
    407 error_threshold_triangulation = config.get("triangulation").get(
    408     "reproj_error_threshold_triangulation"
    409 )
    410 likelihood_threshold = config.get("triangulation").get(
    411     "likelihood_threshold_triangulation"
    412 )

IndexError: list index out of range
```
I dont know how to solve it. I print some information below.
```python
print(f"calib_dir :{calib_dir}")
print(f"calib_file :{calib_file}")
print(f"calib :{calib}")
print(f"cam_names :{cam_names}")
print(f"cam_excluded_count :{cam_excluded_count}")

calib_dir :D:\pose2sim_test\S00_Demo_Session\S00_Calibration
calib_file :D:\pose2sim_test\S00_Demo_Session\S00_Calibration\Calib_qualisys.toml
calib :{'cam_01': {'name': 'cam_01', 'size': [1088.0, 1920.0], 'matrix': [[1681.244873046875, 0.0, 532.97369384375], [0.0, 1681.075439453125, 948.137390140625], [0.0, 0.0, 1.0]], 'distortions': [-0.000721609375, 0.002187234375, 9.5e-06, 1.078125e-05], 'rotation': [1.6882754799999993, 1.0483220499999997, -0.41955852000000016], 'translation': [0.3211048899999996, 0.9563320600000009, 2.8907130499999996], 'fisheye': False}, 'cam_02': {'name': 'cam_02', 'size': [1088.0, 1920.0], 'matrix': [[1673.729614265625, 0.0, 534.494567875], [0.0, 1673.79724121875, 963.225891109375], [0.0, 0.0, 1.0]], 'distortions': [-0.000747609375, 0.00213728125, 1.51875e-05, 4.546875e-06], 'rotation': [1.34975875, 1.5963809099999993, -1.1983285799999999], 'translation': [-0.11152829000000014, 0.7766184800000001, 3.0675519599999994], 'fisheye': False}, 'cam_03': {'name': 'cam_03', 'size': [1088.0, 1920.0], 'matrix': [[1681.598388671875, 0.0, 513.20837403125], [0.0, 1681.509887703125, 955.005126953125], [0.0, 0.0, 1.0]], 'distortions': [-0.000729765625, 0.00215034375, -8.46875e-06, -8.078125e-06], 'rotation': [0.8109654899999995, -2.1972129299999996, 1.3760277799999996], 'translation': [-0.7934803899999995, 0.32283594000000126, 4.353514870000001], 'fisheye': False}, 'cam_04': {'name': 'cam_04', 'size': [1088.0, 1920.0], 'matrix': [[1675.234985359375, 0.0, 540.106201171875], [0.0, 1675.204223640625, 964.0302734375], [0.0, 0.0, 1.0]], 'distortions': [-0.000744265625, 0.002104171875, 4.328125e-06, 3.109375e-06], 'rotation': [1.4045571699999995, -1.3887412699999993, 0.42535743000000026], 'translation': [0.5030217200000007, 0.04894934000000107, 4.406564460000002], 'fisheye': False}, 'metadata': {'adjusted': False, 'error': 0.0}}
cam_names :['cam_01' 'cam_02' 'cam_03' 'cam_04' None]
cam_excluded_count :[]
```

need your help!

EmanuelWicki, 2024-04-19:
I have the same error for some of my files but not for all. Have you calibrated the extrinsics successfully?

davidpagnon, 2024-04-19:
Hi, you may have come across a bug that only happens if everything too well 😅
I can't check this right now but it seems like no cameras had to be excluded, so the list is empty. Your files will probably be triangulated okay though, since this is the recap message that comes at the very end. Can you have a look and tell me?

I'll try to fix it and push a new release this week-end.

chizhan9527, 2024-04-19:
> Hi, you may have come across a bug that only happens if everything too well 😅 I can't check this right now but it seems like no cameras had to be excluded, so the list is empty. Your files will [...]

Hi David，
 Thank you for your replly.I check my pose-3d folder, but it is empty, so my files may not be triangulated. Sorry about it.😢

davidpagnon, 2024-04-19:
Oh so that's not it  :(

Can you both tell me how you calibrate? (file conversion, scene, checkerboard...)

chizhan9527, 2024-04-19:
> Oh so that's not it :(
> 
> Can you both tell me how you calibrate? (file conversion, scene, checkerboard...)

I use checkboard to do intrinsics. Using scene to do extrinsics.
my 'Calib_qualisys.toml' file below
```shell
[cam_01]
name = "cam_01"
size = [ 1088.0, 1920.0]
matrix = [ [ 1681.244873046875, 0.0, 532.97369384375], [ 0.0, 1681.075439453125, 948.137390140625], [ 0.0, 0.0, 1.0]]
distortions = [ -0.000721609375, 0.002187234375, 9.5e-06, 1.078125e-05]
rotation = [ 1.6882754799999993, 1.0483220499999997, -0.41955852000000016]
translation = [ 0.3211048899999996, 0.9563320600000009, 2.8907130499999996]
fisheye = false

[cam_02]
name = "cam_02"
size = [ 1088.0, 1920.0]
matrix = [ [ 1673.729614265625, 0.0, 534.494567875], [ 0.0, 1673.79724121875, 963.225891109375], [ 0.0, 0.0, 1.0]]
distortions = [ -0.000747609375, 0.00213728125, 1.51875e-05, 4.546875e-06]
rotation = [ 1.34975875, 1.5963809099999993, -1.1983285799999999]
translation = [ -0.11152829000000014, 0.7766184800000001, 3.0675519599999994]
fisheye = false

[cam_03]
name = "cam_03"
size = [ 1088.0, 1920.0]
matrix = [ [ 1681.598388671875, 0.0, 513.20837403125], [ 0.0, 1681.509887703125, 955.005126953125], [ 0.0, 0.0, 1.0]]
distortions = [ -0.000729765625, 0.00215034375, -8.46875e-06, -8.078125e-06]
rotation = [ 0.8109654899999995, -2.1972129299999996, 1.3760277799999996]
translation = [ -0.7934803899999995, 0.32283594000000126, 4.353514870000001]
fisheye = false

[cam_04]
name = "cam_04"
size = [ 1088.0, 1920.0]
matrix = [ [ 1675.234985359375, 0.0, 540.106201171875], [ 0.0, 1675.204223640625, 964.0302734375], [ 0.0, 0.0, 1.0]]
distortions = [ -0.000744265625, 0.002104171875, 4.328125e-06, 3.109375e-06]
rotation = [ 1.4045571699999995, -1.3887412699999993, 0.42535743000000026]
translation = [ 0.5030217200000007, 0.04894934000000107, 4.406564460000002]
fisheye = false

[metadata]
adjusted = false
error = 0.0
```

davidpagnon, 2024-04-19:
can you do a `pip show pose2sim` and tell me which version you use? If it is below 0.8.1, can you upgrade it (`pip install pose2sim -U) and try again?

There are three things that I don't get: 
- your error comes at triangulation.py:1531, while this file has less than 1000 lines (at least in its latest version)
- the recap function comes at the very end of the script. So it should have produced a trc file, or else it should have thrown an error beforehand
- your calibration file is called calib_qualisys.toml, while this is what comes when you convert a qualisys file. If you calibrate with the scene, it should be called Calib_scene.toml

chizhan9527, 2024-04-19:
- My pose2sim version
[image]

- I added a few lines of code, but my vscode has automatic formatting and when I saved, the number of lines of code increased
- calib_qualisys.toml is my intrinsics. I have Calib_scene.toml and It's in my response to someone else on this issue

I use an old version S00_Demo_Session folder but new 'Config.toml' I got from the latest. Will it make a difference?

chizhan9527, 2024-04-19:
> can you do a `pip show pose2sim` and tell me which version you use? If it is below 0.8.1, can you upgrade it (`pip install pose2sim -U) and try again?
>
> There are three things that I don't get:
>
> * [...]

I'm sorry, I got it wrong. is it both intrinsics and extrinsics calibration file stored in Calib_scene.toml?

davidpagnon, 2024-04-19:
Okay thank you, this is definitely a bug that you just discovered! 
When you convert intrinsics, and then calibrate extrinsics without overwriting intrinsics, you end up with two calibration files and it takes the first by alphabetical order, which is not the one you want. I don't have time to fix it tonight but I'll try to do it over the week-end!
In the meantime, you can just delete your Calib_qualisys.toml and tell me if you get sensible results.

chizhan9527, 2024-04-19:
my Calib_scene.toml
```
[int_cam01_img]
name = "int_cam01_img"
size = [ 1920.0, 1080.0]
matrix = [ [ 1872.2393163205547, 0.0, 959.5], [ 0.0, 1878.3310918569832, 539.5], [ 0.0, 0.0, 1.0]]
distortions = [ -0.15437072597374374, 0.27810996659357345, 0.0027102690871292005, 0.00014185963176442318]
rotation = [ 0.5472294655086661, 2.3545378451558814, -1.7017148885469422]
translation = [ 0.013119616757832962, 0.3642080614987714, 5.835724605198875]
fisheye = false

[int_cam02_img]
name = "int_cam02_img"
size = [ 1920.0, 1080.0]
matrix = [ [ 1992.7165834355114, 0.0, 959.5], [ 0.0, 2004.4541434290397, 539.5], [ 0.0, 0.0, 1.0]]
distortions = [ -0.18812293549095657, 0.18463394456322152, -0.028893683766905643, -0.005285774234324994]
rotation = [ 1.8769714260521635, 0.41675500460510156, -0.211343551790896]
translation = [ -0.6916512539639558, 0.8382998244005676, 4.384508530401515]
fisheye = false

[int_cam03_img]
name = "int_cam03_img"
size = [ 1920.0, 1080.0]
matrix = [ [ 1989.338427238865, 0.0, 959.5], [ 0.0, 2015.7477030742887, 539.5], [ 0.0, 0.0, 1.0]]
distortions = [ -0.1487156275287396, -0.1905981760475995, -0.032762176101141086, 0.005337974298497598]
rotation = [ 1.883581002499085, -0.4695212036559561, 0.3441322980293029]
translation = [ -0.39058268523881123, 1.0519819388315186, 4.169159731100561]
fisheye = false

[int_cam04_img]
name = "int_cam04_img"
size = [ 1920.0, 1080.0]
matrix = [ [ 1851.1296191722372, 0.0, 959.5], [ 0.0, 1853.5271630204468, 539.5], [ 0.0, 0.0, 1.0]]
distortions = [ -0.12394139912610926, 0.020211498792164263, 0.0033857361683345545, -0.0042963059755158845]
rotation = [ 0.7133947983599229, -2.365013083575868, 1.7243973172880032]
translation = [ 0.6934328348466542, 0.6309465843602338, 5.699894096825773]
fisheye = false

[metadata]
adjusted = false
error = 0.0
```

davidpagnon, 2024-04-19:
Yes, both intrinsic and extrinsics are written in one single calibration file

chizhan9527, 2024-04-19:
oh no ! I delete Calib_qualisys.toml but it also occur a bug
[image]

chizhan9527, 2024-04-19:
```shell
---------------------------------------------------------------------
Triangulation of 2D points for S00_P00_T00_StaticTrial, for all frames.
On Saturday 20. April 2024, 00:08:10
---------------------------------------------------------------------

Project directory: D:\pose2sim_test\S00_Demo_Session\S00_P00_SingleParticipant\S00_P00_T00_StaticTrial
100%|██████████████████████████████████████████████████████████████████████████████████| 19/19 [00:00<00:00, 88.59it/s]
calib_dir :D:\pose2sim_test\S00_Demo_Session\S00_Calibration
calib_file :D:\pose2sim_test\S00_Demo_Session\S00_Calibration\Calib_scene.toml
calib :{'int_cam01_img': {'name': 'int_cam01_img', 'size': [1920.0, 1080.0], 'matrix': [[1872.2393163205547, 0.0, 959.5], [0.0, 1878.3310918569832, 539.5], [0.0, 0.0, 1.0]], 'distortions': [-0.15437072597374374, 0.27810996659357345, 0.0027102690871292005, 0.00014185963176442318], 'rotation': [0.5472294655086661, 2.3545378451558814, -1.7017148885469422], 'translation': [0.013119616757832962, 0.3642080614987714, 5.835724605198875], 'fisheye': False}, 'int_cam02_img': {'name': 'int_cam02_img', 'size': [1920.0, 1080.0], 'matrix': [[1992.7165834355114, 0.0, 959.5], [0.0, 2004.4541434290397, 539.5], [0.0, 0.0, 1.0]], 'distortions': [-0.18812293549095657, 0.18463394456322152, -0.028893683766905643, -0.005285774234324994], 'rotation': [1.8769714260521635, 0.41675500460510156, -0.211343551790896], 'translation': [-0.6916512539639558, 0.8382998244005676, 4.384508530401515], 'fisheye': False}, 'int_cam03_img': {'name': 'int_cam03_img', 'size': [1920.0, 1080.0], 'matrix': [[1989.338427238865, 0.0, 959.5], [0.0, 2015.7477030742887, 539.5], [0.0, 0.0, 1.0]], 'distortions': [-0.1487156275287396, -0.1905981760475995, -0.032762176101141086, 0.005337974298497598], 'rotation': [1.883581002499085, -0.4695212036559561, 0.3441322980293029], 'translation': [-0.39058268523881123, 1.0519819388315186, 4.169159731100561], 'fisheye': False}, 'int_cam04_img': {'name': 'int_cam04_img', 'size': [1920.0, 1080.0], 'matrix': [[1851.1296191722372, 0.0, 959.5], [0.0, 1853.5271630204468, 539.5], [0.0, 0.0, 1.0]], 'distortions': [-0.12394139912610926, 0.020211498792164263, 0.0033857361683345545, -0.0042963059755158845], 'rotation': [0.7133947983599229, -2.365013083575868, 1.7243973172880032], 'translation': [0.6934328348466542, 0.6309465843602338, 5.699894096825773], 'fisheye': False}, 'metadata': {'adjusted': False, 'error': 0.0}}
cam_names :['int_cam01_img' 'int_cam02_img' 'int_cam03_img' 'int_cam04_img' None]
cam_excluded_count :[]
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[2], line 1
----> 1 Pose2Sim.triangulation()

File D:\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:398, in triangulation(config)
    395 logging.info("---------------------------------------------------------------------")
    396 logging.info(f"\nProject directory: {project_dir}")
--> 398 triangulate_all(config_dict)
    400 end = time.time()
    401 elapsed = end-start

File D:\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\triangulation.py:1531, in triangulate_all(config)
   1526     logging.info(
   1527         "\nThe trc and c3d files have been renamed to match the order of the static sequences."
   1528     )
   1530 # Recap message
-> 1531 recap_triangulate(
   1532     config,
   1533     error_tot,
   1534     nb_cams_excluded_tot,
   1535     keypoints_names,
   1536     cam_excluded_count,
   1537     interp_frames,
   1538     non_interp_frames,
   1539     trc_paths,
   1540 )

File D:\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\triangulation.py:406, in recap_triangulate(config, error, nb_cams_excluded, keypoints_names, cam_excluded_count, interp_frames, non_interp_frames, trc_path)
    404 print(f"cam_names :{cam_names}")
    405 print(f"cam_excluded_count :{cam_excluded_count}")
--> 406 cam_names = cam_names[list(cam_excluded_count[0].keys())]
    407 error_threshold_triangulation = config.get("triangulation").get(
    408     "reproj_error_threshold_triangulation"
    409 )
    410 likelihood_threshold = config.get("triangulation").get(
    411     "likelihood_threshold_triangulation"
    412 )

IndexError: list index out of range
```

davidpagnon, 2024-04-19:
Sorry I need to go, I'll check this a bit later.
I'm wondering why you have a Config_qualisys.toml if you calibrated intrinsics with  checkerboard?

Can you delete all the calibration files and recalibrate with your checkerboard and your points on the scene?

chizhan9527, 2024-04-19:
ok.
I dont know why i have it , but  I will try to calibrate again

davidpagnon, 2024-04-19:
If you still have your images files for intrinsics and extrinsics it should not be too long :)

chizhan9527, 2024-04-19:
I do calibration again but still this error 
```shell
---------------------------------------------------------------------
Triangulation of 2D points for S00_P00_T00_StaticTrial, for all frames.
On Saturday 20. April 2024, 00:28:09
---------------------------------------------------------------------

Project directory: D:\pose2sim_test\S00_Demo_Session\S00_P00_SingleParticipant\S00_P00_T00_StaticTrial
100%|█████████████████████████████████████████████████████████████████████████████████| 19/19 [00:00<00:00, 111.43it/s]
calib_dir :D:\pose2sim_test\S00_Demo_Session\S00_Calibration
calib_file :D:\pose2sim_test\S00_Demo_Session\S00_Calibration\Calib_scene.toml
calib :{'int_cam01_img': {'name': 'int_cam01_img', 'size': [1920.0, 1080.0], 'matrix': [[1872.2393163205547, 0.0, 959.5], [0.0, 1878.3310918569832, 539.5], [0.0, 0.0, 1.0]], 'distortions': [-0.15437072597374374, 0.27810996659357345, 0.0027102690871292005, 0.00014185963176442318], 'rotation': [0.5255745275956876, 2.3438443182764632, -1.7147244890313007], 'translation': [0.020846285155764317, 0.36399477859901747, 5.785977112422314], 'fisheye': False}, 'int_cam02_img': {'name': 'int_cam02_img', 'size': [1920.0, 1080.0], 'matrix': [[1992.7165834355114, 0.0, 959.5], [0.0, 2004.4541434290397, 539.5], [0.0, 0.0, 1.0]], 'distortions': [-0.18812293549095657, 0.18463394456322152, -0.028893683766905643, -0.005285774234324994], 'rotation': [1.8786248351995523, 0.42523091534280005, -0.20760252426145778], 'translation': [-0.694777054775141, 0.8415708419968231, 4.412979674913439], 'fisheye': False}, 'int_cam03_img': {'name': 'int_cam03_img', 'size': [1920.0, 1080.0], 'matrix': [[1989.338427238865, 0.0, 959.5], [0.0, 2015.7477030742887, 539.5], [0.0, 0.0, 1.0]], 'distortions': [-0.1487156275287396, -0.1905981760475995, -0.032762176101141086, 0.005337974298497598], 'rotation': [1.8721158179042117, -0.45043612248866643, 0.35355193135060975], 'translation': [-0.39945587556688356, 1.0304593884169466, 4.112977619421741], 'fisheye': False}, 'int_cam04_img': {'name': 'int_cam04_img', 'size': [1920.0, 1080.0], 'matrix': [[1851.1296191722372, 0.0, 959.5], [0.0, 1853.5271630204468, 539.5], [0.0, 0.0, 1.0]], 'distortions': [-0.12394139912610926, 0.020211498792164263, 0.0033857361683345545, -0.0042963059755158845], 'rotation': [0.7131102058401283, -2.3630914188653445, 1.7236891673825105], 'translation': [0.6900484679358816, 0.6361491551506732, 5.712548584442903], 'fisheye': False}, 'metadata': {'adjusted': False, 'error': 0.0}}
cam_names :['int_cam01_img' 'int_cam02_img' 'int_cam03_img' 'int_cam04_img' None]
cam_excluded_count :[]
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[3], line 1
----> 1 Pose2Sim.triangulation()

File D:\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\Pose2Sim.py:398, in triangulation(config)
    395 logging.info("---------------------------------------------------------------------")
    396 logging.info(f"\nProject directory: {project_dir}")
--> 398 triangulate_all(config_dict)
    400 end = time.time()
    401 elapsed = end-start

File D:\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\triangulation.py:1531, in triangulate_all(config)
   1526     logging.info(
   1527         "\nThe trc and c3d files have been renamed to match the order of the static sequences."
   1528     )
   1530 # Recap message
-> 1531 recap_triangulate(
   1532     config,
   1533     error_tot,
   1534     nb_cams_excluded_tot,
   1535     keypoints_names,
   1536     cam_excluded_count,
   1537     interp_frames,
   1538     non_interp_frames,
   1539     trc_paths,
   1540 )

File D:\anaconda3\envs\Pose2Sim\lib\site-packages\Pose2Sim\triangulation.py:406, in recap_triangulate(config, error, nb_cams_excluded, keypoints_names, cam_excluded_count, interp_frames, non_interp_frames, trc_path)
    404 print(f"cam_names :{cam_names}")
    405 print(f"cam_excluded_count :{cam_excluded_count}")
--> 406 cam_names = cam_names[list(cam_excluded_count[0].keys())]
    407 error_threshold_triangulation = config.get("triangulation").get(
    408     "reproj_error_threshold_triangulation"
    409 )
    410 likelihood_threshold = config.get("triangulation").get(
    411     "likelihood_threshold_triangulation"
    412 )

IndexError: list index out of range
```

davidpagnon, 2024-04-19:
What are the files you have in your calibration folder?

chizhan9527, 2024-04-19:
> What are the files you have in your calibration folder?
[image]

davidpagnon, 2024-04-20:
I honestly don't know what the problem is. Could you send me a link to your session folder? (here or by email at dp2032@bath.ac.uk if you want)

chizhan9527, 2024-04-20:
> I honestly don't know what the problem is. Could you send me a link to your session folder? (here or by email at [dp2032@bath.ac.uk](mailto:dp2032@bath.ac.uk) if you want)

hi, I send an email to you

davidpagnon, 2024-04-20:
Okay, thank you for sending this to me. The problem is calibration, as I first expected. More specifically, it is extrinsic calibration. 

On the attached images, the points to click are not precise enough: it is hard to know where to click on the bottle, the corners of the board are not sharp, and your config file shows that you also click somewhere in the middle of the board, while there is no mark. There is no way to be precise this way. And indeed, in the end, your calibration error is always more than 10 cm, sometimes close to 20 cm. Subsequently, no point is successfully triangulated among the 15 frames of the capture.

One way to get more accurate points would be to put tape on the floor, or add a table or anything else in the scene and measure its coordinates. Don't worry, as soon as calibration is done you can remove all the calibration marks and objects until you move the cameras again.

I still got a successful triangulation by massively increasing the threshold beyond which a result is dismissed: in Config.toml, I increased `reproj_error_threshold_triangulation` from 15 to 150 px. Of course, this is not ideal.

P.S.: if the person does not move, the synchronization won't be precise since I use keypoint speeds. If you only capture an A-pose and the person does not move, I would not bother to synchronize. Moreover, if you use Qualisys cameras, are they not already synchronized?
For starters, I would also recommend not to run personAssociation either. You'll gradually improve the complexity once you manage to get results.

[image]

chizhan9527, 2024-04-21:
Hi David,

 Thank you for your long reply! This is very helpful to me. I will try to follow your advice. I dont use qualisys camera and I use ordinary camera. As a starter, I also have some questions.
-  What does personAssociation do?  Why shouldn't I do it?
-  If I do synchronization, how do I cancel it?

Thank you again!

davidpagnon, 2024-04-21:
- if you want to analyse multiple people in the same scene, personAssociation makes sure you triangulate the same person from each view. In single person analysis, it will choose the most prominent place in the scene. But here, the persons in the backgrounds are rather far and I don't suspect they will be chosen for triangulation. Ultimately you should do personAssociation, but when debugging you often want to keep things simple and to better pin where the error comes from.
- you can reset synchronization by setting reset_sync to true in Config.toml, and then run synchronization again. Don't forget to set it to false again if you want to synchronize later on.

EmanuelWicki, 2024-04-22:
Would it also be possible to scale the model without a static pose (normal video of person moving) ?

davidpagnon, 2024-04-22:
The way I generally do it is by extracting a few frames in the motion capture where the participant does not move much. 
In theory, it should also work if the person is moving, since we are just scaling limb sizes according to keypoint positions (and unlike with marker-based methods, these do not need to be adjusted when another person places the markers).

I have not tried it this way though, so if you do, I'm interested in knowing how it goes!

chizhan9527, 2024-04-29:
Thank you！
By re-calibrating the internal and external parameters, I succeeded

closed 2024-04-29 by chizhan9527