perfanalytics/pose2sim #99: z coordinates of triangulation are negative

EmanuelWicki, 2024-04-19:
Hi David,
I have all negative z-coordinates in my triangualted .trc file. The rest appears to be correct. Have you ever experienced something like this?

Best,
Emanuel

davidpagnon, 2024-04-19:
You may have switched the X and Y axis, which in a right-handers system would lead to a Z axis pointing downwards

EmanuelWicki, 2024-04-19:
I used the checkerboard, wouldt that automatically recognize the coordinate direction ?

davidpagnon, 2024-04-19:
You may have a mirrored version of the standard checkerboard I suppose. Where did you print it from? Can you show me a picture?

EmanuelWicki, 2024-04-19:
[image]

davidpagnon, 2024-04-19:
This may be the issue: 

https://github.com/perfanalytics/pose2sim?tab=readme-ov-file#calculate-from-scratch
> Make sure that the board:
>  - is filmed from different angles, covers a large part of the video frame, and is in focus.
>  - is flat, without reflections, surrounded by a white border, and is not rotationally invariant (Nrows ≠ Ncols, and Nrows odd if Ncols even).
> 
> A common error is to specify the external, instead of the internal number of corners. This may be one less than you would intuitively think.

Here, both the number of rows and of columns are odd, which makes the board not rotationally invariant.

Did you already move your cameras? If not, you can print another checkerboard, capture it, and then run the analysis again.
If you did, you can measure the several coordinates of your walking track and calibrate your extrinsics from the scene instead. That's actually what I would advise, as calibration is better if a larger space area is covered.

EmanuelWicki, 2024-04-22:
Thanks a lot. I will do it :)

EmanuelWicki, 2024-05-02:
Thank you very much, this issue is resolved

closed 2024-05-02 by EmanuelWicki