perfanalytics/pose2sim #100: All nan-slice encountered

sarkadava, 2024-04-24:
I am updating my pose2sim pipeline to the 0.8 version and am encoutering this error

```

Project directory: C:\Users\kadava\Documents\GitHub\demo_3Dtracking_pose2sim_updated\projectdata\Session_s1\P1\T1
C:\Users\kadava\Documents\GitHub\demo_3Dtracking_pose2sim_updated\projectdata\Session_s1
  0%|                                                                                          | 0/299 [00:00<?, ?it/s]
error  [nan]

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[6], line 92
     88             shutil.copy(toml_file, dest)
     90 #Pose2Sim.synchronization()
     91 #Pose2Sim.personAssociation()
---> 92 Pose2Sim.triangulation()
     93 Pose2Sim.filtering()
     94 Pose2Sim.markerAugmentation()

File ~\Documents\GitHub\demo_3Dtracking_pose2sim_updated\Pose2Sim\Pose2Sim.py:398, in triangulation(config)
    395 logging.info("---------------------------------------------------------------------")
    396 logging.info(f"\nProject directory: {project_dir}")
--> 398 triangulate_all(config_dict)
    400 end = time.time()
    401 elapsed = end-start 

File ~\Documents\GitHub\demo_3Dtracking_pose2sim_updated\Pose2Sim\triangulation.py:822, in triangulate_all(config)
    819 coords_2D_kpt = np.array( (x_files[n][:, keypoint_idx], y_files[n][:, keypoint_idx], likelihood_files[n][:, keypoint_idx]) )
    820 coords_2D_kpt_swapped = np.array(( x_files[n][:, keypoints_idx_swapped[keypoint_idx]], y_files[n][:, keypoints_idx_swapped[keypoint_idx]], likelihood_files[n][:, keypoints_idx_swapped[keypoint_idx]] ))
--> 822 Q_kpt, error_kpt, nb_cams_excluded_kpt, id_excluded_cams_kpt = triangulation_from_best_cameras(config, coords_2D_kpt, coords_2D_kpt_swapped, P, calib_params) # P has been modified if undistort_points=True
    824 Q[n].append(Q_kpt)
    825 error[n].append(error_kpt)

File ~\Documents\GitHub\demo_3Dtracking_pose2sim_updated\Pose2Sim\triangulation.py:524, in triangulation_from_best_cameras(config, coords_2D_kpt, coords_2D_kpt_swapped, projection_matrices, calib_params)
    522 error_min = np.nanmin(error)
    523 # print(error_min)
--> 524 best_cams = np.nanargmin(error)
    525 nb_cams_excluded = nb_cams_excluded_filt[best_cams]
    527 Q = Q_filt[best_cams][:-1]

File <__array_function__ internals>:180, in nanargmin(*args, **kwargs)

File ~\AppData\Local\anaconda3\envs\3Dtracking_vol2\lib\site-packages\numpy\lib\nanfunctions.py:553, in nanargmin(a, axis, out, keepdims)
    551     mask = np.all(mask, axis=axis)
    552     if np.any(mask):
--> 553         raise ValueError("All-NaN slice encountered")
    554 res = np.argmin(a, axis=axis, out=out, keepdims=keepdims)
    555 return res

ValueError: All-NaN slice encountered

```

As you see, I guess the error is caused by the printed error [nan] but I have no idea where does it come from? Is that due to calibration? Because my calib.toml is filled with values. I also tried to increase the repro error to 150 px but same error remains

Any help appreciated :)

sarkadava, 2024-04-24:
Problem solved, just need to always calculate extrinsics

closed 2024-04-24 by sarkadava

davidpagnon, 2024-04-24:
I like it when the problem solves itself without me having to take action :D
I provided the `calculate_extrinsics` in Config.toml so that people can calibrate the intrinsic parameters of their cameras before the experiment if needed. Did you set it to false by mistake, or did I miss some confusion configuration somewhere?

sarkadava, 2024-04-24:
@davidpagnon I thought that I can run both intrinsic and extrinsic just ones and when they are in calib.toml, I don't have to anymore.