perfanalytics/pose2sim #102: overcome partial occlusion in pose estimation

EmanuelWicki, 2024-05-02:
Hi David,

I am doing gait analysis with 2 cameras, one is dorsal and one sagittal (perpendicular to each other) and bc of that i have partial occlusion of the left side of the person (I cant move the cameras bc. i am working with existing videos). Keypoints are not so well detected. During triangulation, most of the data cnat be used because of that. Do you have a possible solution for that?

Best,
Emanuel

closed 2024-05-03 by EmanuelWicki

davidpagnon, 2024-05-04:
I'm glad you made it work!
Just in case, I would recommend to have the cameras at about 30 degrees on each side of the participant. Maybe the person can walk in a different direction if you cannot move the cameras?

EmanuelWicki, 2024-05-05:
Hey David, 
I did not yet make it work... the problem is that i have to work with a given dataset... and there i always have perpendicular camera meaning sagittal and dorsal... But I am trying to figure out the best way. Will Pose it here as soon as I have found a good solution